evcc-io/evcc #35: go-e Error

tmeidinger, 2020-04-14:
[main] INFO 2020/04/14 13:51:52 using config file /etc/evcc.yaml
[main] INFO 2020/04/14 13:51:52 listening at 0.0.0.0:7070
[go-e] WARN 2020/04/14 13:51:52 -- experimental --
[load] INFO 2020/04/14 13:51:52 main config: vehicle ✓ grid ✓ pv — charge —
[load] INFO 2020/04/14 13:51:52 main charge mode: off
[load] ERROR 2020/04/14 13:51:53 main charger error: invalid character '<' looking for beginning of value
[load] ERROR 2020/04/14 13:51:53 main charge controller error: invalid character '<' looking for beginning of value
[load] ERROR 2020/04/14 13:51:53 main charger error: invalid character '<' looking for beginning of value
panic: runtime error: index out of range [11] with length 0

Any ideas?

andig, 2020-04-14:
Ich konnte go-e leider nicht testen. Bitte mal mit —log trace starten und per pm unter cpuidle@gmx.de melden.

andig, 2020-04-14:
Problem is probably in https://github.com/andig/evcc/blob/master/charger/go-e.go#L134 as I don't check the error upfront and rely on slice to have 12 elements. Will fix but trace should show what actually gets returned by api.

andig, 2020-04-14:
Noch eine Idee: das < sieht aus als würde Deine go-e evtl. mit XML antworten- eigentlich hätte ich JSON erwartet?

andig, 2020-04-14:
Should be fixed with 9efe53e7273fbc1ee89d7d5a2683cc4be038fe60

PR #36 (2020-04-15): Fix go-eCharger api
Fix #35
closed 2020-04-15 by andig