evcc-io/evcc #50: MaxCurrent seems to be 10 time too high

Caibuk, 2020-04-25:
I have evcc 0.5 running on docker and a Wallbe eco 2.0 connected to 1 phase at the moment.
evcc seems to control the wallbox (connection ok).
But evcc seems to set the MaxCurrent value 10 times too high (e.g. 60 amps instead of 6).

evcc set to "pv":
[image]

result in wallbe interface:
[image]

As a result the wallbox runs at full power even though PV energy is too low.

andig, 2020-04-25:
Thank you @Caibuk but giving EVCC a try. It seems that there are two revisions of the Wallbe. An older, using the Phoenix charge controller and a newer, using a (rebranded) controller with different register layout.

Could you please let me know which model/revision your Wallbe has and how old it is? Do you potentially have any inside photos of the controller (**attention: life danger- only for qualified personell**).

If you're adventurous you could also try the `phoenix` charger type which should cover the older revisions.

Caibuk, 2020-04-25:
Hi Andreas,
thanks for your prompt feedback. 
I think it is the same controller like in your box. The rebranded Phoenix one.
The box was made around 2017.
The inside looks exactly like this one:
https://uploads.tapatalk-cdn.com/20180105/5242535a01292aae67d27ed3bdf59686.jpg
(sry, I have no clue how to embed a picture using my github on my smartphone's browser). 
I will give the Phoenix type charger a try. 
Regards
Markus

andig, 2020-04-25:
> The controllers *look* the same, but mine is from 2019...

You're welcome. Thanks for trying! 

Could you kindly also try to read the firmware version:

    docker run volkszaehler/mbmd read -a 192.168.0.8:502 -d 255 -t input -e string 105 2

and show the firmware version according to web interface? 

The controllers *look* the same, but mine is from 2019... You can follow the discussion at https://www.goingelectric.de/forum/viewtopic.php?p=1240962#p1240962 if there is any way to programmatically find out the difference.

Caibuk, 2020-04-25:
Here are the modbus readings:
```
./mbmd read -a 10.0.1.67:502 -d 255 -t input -e string 105 2
.095
```
```
./mbmd read -a 10.0.1.67:502 -d 255 -t holding -e string 304 6
00000000869
```
```
./mbmd read -a 10.0.1.67:502 -d 255 -t holding -e string 310 5
mSrat
```
Sadly, I could not find the firmware version on the web interface.

andig, 2020-04-25:
Could you post your log file with —log trace? That‘ll allow us to see whats going on on the modbus read/writes.

Caibuk, 2020-04-25:
Frankly, I don't know how?
I can run the binary on my synology:
```
sh-4.3# ./evcc -l trace
[load] ERROR 2020/04/25 17:42:25 main updating charge meter: All attempts fail:
#1: mbmd/sdm2301-10/Power outdated: 18s
#2: mbmd/sdm2301-10/Power outdated: 19s
#3: mbmd/sdm2301-10/Power outdated: 19s
```
But I can see only errors (my modbus meter is a bit "zickig"). No warnings, infos, traces ...

Caibuk, 2020-04-25:
Do you think I could/should update the controller? (seems to be possible: https://www.goingelectric.de/forum/viewtopic.php?f=102&t=53433)
I know you could not tell me, but what is your gut feeling?

This is a actual picture of my box:
[image]

andig, 2020-04-25:
> Frankly, I don't know how?

Make sure your evcc image is current (docker pull). Check your mbmd UI- is the meter really updated? Post your evcc config. And: feel free to email me (cpuidle@gmx.de).

Please also show

    ./mbmd read -a 10.0.1.67:502 -d 255 -t holding -e int 300 1

that should be the maximum current for Phoenix. It might never get updated if the meters don't work as EVCC cannot know how to react.

Caibuk, 2020-04-25:
Here is the mbmd output:
```
./mbmd read -a 10.0.1.67:502 -d 255 -t holding -e int 300 1
6
```
The rest via e-mail in German ...

andig, 2020-04-25:
Added 5fb5e22cca3e8d5b066a59c7090cba33849bac24 to document how to differentiate charger types.

andig, 2020-04-25:
> Here is the mbmd output:

Looks like 6A on a Phoenix charger! Wallbe would show 60 here. Waiting for your mail ;)

PR #56 (2020-04-26): Support Wallbe chargers with Phoenix EV-CC-AC1-M3-CBC-RCM-ETH controller
Fix #50, improves 5fb5e22cca3e8d5b066a59c7090cba33849bac24
closed 2020-04-26 by andig