evcc-io/evcc #52: Supporting scenario with house battery inverters

TheRealKerni, 2020-04-26:
If a house setup is also having a battery inverter (and battery connected), then it can/will block dynamic adjustments of the power to charge the car with as long as the inverters charging capacity is within the unused solar generated power.

The system basically works like this:
- The energy manager checks the power usage in the house, which is including the power the ev charger takes
- tells the battery inverter to draw the difference between the generated solar power and the used power
- if the difference is bigger than the capacity of the battery inverter, it will send the rest to the grid
- now this rest is only what the EVCC can detect right now to use for power adjustments for the ev charger

There are multiple options to handle this:
- Do nothing special as the end user still can use "Sofortladen" to overwrite all these limitations to tell the ev charger to use full available/possible power
- Do nothing special and the system simply has to wait until the house battery is charged sufficiently (this is for the energy manager to determine) until more power to charge the car with is getting available
- Consider the battery inverters current charging power as "available" power to add to the ev charging process
- Consider the battery inverters current charging power as "available" and add an option for the user to "allow" the battery inverter to always have a minimum power available

Are there any other options I didn't think of here?

andig, 2020-04-26:
I would probably ask what the preferred mode of operation is:

- if sufficient PV: should the car be charged before the battery? Seems yes, but it this always the case?
- if no sufficient PV: should the car really be charged from the house battery? If yes (i.e. if it needs charging), should this count as "deplete the battery as fast as possible" (i.e. pv replacement) or as "keep charging at minimum until better times come". I feel the latter makes more sense.

Long story short: if car has priority it might make sense to treat batter charge power als "available pv power" and thus a second power source for the charger. Effectively that would give the charger priority of charging home battery.

Does that make sense? Could also double-check how other implementations are doing that.

premultiply, 2020-04-27:
Transfering energy from (the small) house battery to (huge in relation) car battery is IMHO the most expensive way to charge a car. So this should be avoided.
The PV energy should always flow to at least one battery or to the grid. The car should always be charged directly from PV or from the grid. The house battery is only charged from PV if there is enough power.

andig, 2020-04-27:
Hence the design goal: car has priority over home battery.

That implies battery charge power is treated as "additional available pv power" and thus a second power source for the charger.

PR #83 (2020-04-30): Add battery meter support
Fix #52 to give EV charging priority over home battery charging.

The additional battery/ battery charger can be configured like this:

```yaml
meters:
- name: battery
  type: modbus
  model: sunny-island # generic name for sunspec-compatible inverter
  uri: 192.168.0.23:502
  id: 126

loadpoints:
- name: home
  meters:
    grid: ...
    battery: battery
```

**Breaking change**: `meters` has now become a hierarchy level in the `loadpoint` configuration.
closed 2020-04-30 by andig