evcc-io/evcc #54: Add public end user documentation

TheRealKerni, 2020-04-26:
We need a place for end users to find anything they need to use EVCC, if possible in English and German languages.

I would suggest to use Github pages and create a website for this use, e.g. using Hugo static site generator. The Github readme imho should stay development centric. What do you think?

andig, 2020-04-26:
The wiki would be easier to maintain.

TheRealKerni, 2020-04-26:
Would be absolutely a great start. The webpage rather a further goal once functionality and usability is reaching a more 1.0 status and the system is more attractive for non hardcore geeks 😁

andig, 2020-04-26:
Wiki is open for contributions ;)

andig, 2020-05-01:
It would also be great to have German documentation.

closed 2020-07-08 by stale[bot]