evcc-io/evcc #61: Support NRGKick with BLE

andig, 2020-04-27:
See https://www.goingelectric.de/forum/viewtopic.php?p=1241738#p1241738

premultiply, 2020-04-27:
Anbei ein Python-**Testcode** den ich eigentlich für OpenWB entwickelt habe.
Dran kann man aber die grundsätzliche Funktion und die notwendigen BLE-Characteristics gut erkennen.
Sicherlich sind die Fehlererkennung und auch die sonstige Codestruktur noch stark ausbaufähig.
Hier ging es erstmal um einen funktionsfähigen Prototypen.

```
#!/usr/bin/python3
# -*- coding: utf-8 -*-

# NRGkick BLE communication daemon

import sys
import time
import pygatt
from struct import *

def getCPStatusBySignal(e):
	return "NOT_CONNECTED" if e > 10.5 and e <= 13.5 else "CONNECTED" if e > 7.5 and e <= 10.5 else "CHARGE_ACTIVE" if e > 4.5 and e <= 7.5 else "CHARGE_WITH_VENTILATION" if e > 1.3 and e <= 4.5 else "UNDEFINED"

def storeValue(file, value):
	f = open("/var/www/html/openWB/ramdisk/" + file, "w")
	#print(file, value)
	f.write(str(value))
	f.close()

def readValue(file):
	try:
		f = open("/var/www/html/openWB/ramdisk/" + file, "r")
		value = f.read()
		#print(file, value)
		f.close()
	except:
		value = "0"
	return int(float(value))

def disconnected(self):
	print("Disconnected")

hci_device = "hci0"

MacAddress = "00:11:22:33:44:55"
pin = 1234

try:
	adapter = pygatt.GATTToolBackend(hci_device)
	adapter.start()

	while True:
		try:
			device = adapter.connect(MacAddress, timeout=2, address_type=pygatt.BLEAddressType.public, auto_reconnect=True)
			#device.register_disconnect_callback(disconnected)
		except pygatt.exceptions.NotConnectedError:
			#device.remove_disconnect_callback(disconnected)
			#device.disconnect()
			print("Connection to %s failed." % MacAddress)
			continue

		print("Connected to %s" % MacAddress)

		while True:
			time.sleep(2) #NRGkick measurement sampling interval is limited.

			current = readValue('llsoll')

			#WRITE

			try:
				appInfo = unpack('>BHBBBBBBBBxx', device.char_read("8f75bba0-c903-11e4-9fe8-0002a5d6b15d"))
			except:
				break
[... 95 lines cut ...]
			storeValue('lla2', MeasuresChargingCurrentPhase2) #ChargingCurrentPhase[1]
			storeValue('lla3', MeasuresChargingCurrentPhase3) #ChargingCurrentPhase[2]
			storeValue('llv1', MeasuresVoltagePhase1) #VoltagePhase[0]
			storeValue('llv2', MeasuresVoltagePhase2) #VoltagePhase[1]
			storeValue('llv3', MeasuresVoltagePhase3) #VoltagePhase[2]
			VAPhase1 = MeasuresVoltagePhase1 * MeasuresChargingCurrentPhase1 #PowerFactorPhase[0]
			storeValue('llpf1', round(MeasuresChargingPowerPhase1 * 1000 / VAPhase1, 3)) if VAPhase1 > 0.0 else storeValue('llpf1', 1.000)
			VAPhase2 = MeasuresVoltagePhase2 * MeasuresChargingCurrentPhase2 #PowerFactorPhase[1]
			storeValue('llpf2', round(MeasuresChargingPowerPhase2 * 1000 / VAPhase2, 3)) if VAPhase2 > 0.0 else storeValue('llpf2', 1.000)
			VAPhase3 = MeasuresVoltagePhase3 * MeasuresChargingCurrentPhase3 #PowerFactorPhase[2]
			storeValue('llpf3', round(MeasuresChargingPowerPhase3 * 1000 / VAPhase3, 3)) if VAPhase3 > 0.0 else storeValue('llpf3', 1.000)

			plugstat = 1 if 1.3 <= MeasuresCPSignal <= 10.5 else 0
			storeValue('plugstat', plugstat)

			chargestat = 1 if 1.3 <= MeasuresCPSignal <= 7.5 else 0
			storeValue('chargestat', chargestat)

except (KeyboardInterrupt, SystemExit):
	# handle process termination and CTRL+C
	sys.exit()

finally:
	adapter.stop()
```

andig, 2020-04-27:
Sieht gut aus- Go kann auch gatt: https://github.com/paypal/gatt <https://github.com/paypal/gatt>

Vielleicht hast Du ja Lust mal mit Go zu spielen oder willst den Code auch direkt ins Github mit rein posten?

Viele Grüße, Andreas

premultiply, 2020-04-27:
Ich habe wirklich Null Ahnung von Go und noch weniger Durchblick in die ordentliche Struktur von EVCC ;-)

andig, 2020-04-27:
Letzteres macht nix- in evcc kann ich es einbauen. Falls Du Lust auf Go hättest- schöne Sprache die mit genialem Design Spass macht. Vielleicht kann ich es Dir ja schmackhaft machen ;)

Viele Grüße, Andreas

TheRealKerni, 2020-04-27:
Also nur mal so: Ich hatte bis zu meinem ersten Pull Request hier vor gut ner Woche auch keine Ahnung von go 😄  Und wenn ich das hinbekomme, dann ... ihr wisst schon 😁

andig, 2020-05-03:
@premultiply könntest du mir darauf remote zugriff geben, also eine shell? Dann bitte unter cpuidle@gmx.de melden.

PR #120 (2020-05-12): Support NRGKick bluetooth
Fix #61
closed 2020-05-12 by andig