evcc-io/evcc #62: Allow grid meter to get inverted

goebelmeier, 2020-04-27:
My grid meter is currently providing positive values as long as power gets delivered to the grid and negative values if power is getting purchased from the grid. 

Please implement a config parameter to invert the reading. Currently i'm just dry-running evcc just with my meters. Charger and vehicle is configured as dummy script.

andig, 2020-04-27:
Would you expect this on the plugin or on the meter level?

goebelmeier, 2020-04-27:
From my point of view, it should be safe on the meter level

andig, 2020-05-02:
@goebelmeier as per https://github.com/andig/evcc/pull/90 the `scale` parameter is supported for HTTP and MQTT devices. Where else would we need this? Modbus? Websocket?

closed 2020-05-02 by andig