evcc-io/evcc #66: Support vzlogger or volkszaehler

schenlap, 2020-04-28:
vzlogger has a few interfaces that could be used:

- Push Server (updates very fast)

- HTTP (it depends on the set agg time, but this is rather high because it also used to writes the data to the database)

- MQTT (I do not know the update rate, probably the same as API)

MQTT should be supported already by evcc, it could be uses as a workaround. But not everybody is using mqtt.

volkszaehler:

- HTTP

andig, 2020-04-28:
For HTTP+JSON see #67. The purpose here should be- as I understand- to be able to combine Bezug and Einspeisung into a single meter?

andig, 2020-04-28:
Renamed title to cover the remaining issue of the use case laid out across #66, #67, #68

andig, 2020-04-28:
Websocket receiver should allow JSON decoding similar to #67

schenlap, 2020-04-28:
> The purpose here should be- as I understand- to be able to combine Bezug and Einspeisung into a single meter?

It is a design decision where to subtract the import and export power, I think it could be a new plugin that can handle calulations of meter values wich does not depend on specific meters. If so this topic would be for websockets, not dealing with import and export at the same time.

TheRealKerni, 2020-04-28:
I really like the idea of having a websocket API. That way evcc could in the long rather focus more on the main use case talking to the wallbox mainly with data coming from other providers in a constant stream.

andig, 2020-04-30:
@schenlap see #72 and https://github.com/andig/evcc#http-readwrite. You can now connect to volkszaehler: 

```yaml
type: http
uri: https://volkszaehler/api/data/<uuid>.json?from=now
jq: .data.tuples[0][1] # parse response json
```

andig, 2020-05-01:
The tricky part with the ws receiver is that it could contain any data and, hence, reference any device and reading. Websockets would need something like a selector, e.g. a jq expression which must match in order to recognize a value. Could be done similar to HTTP.

andig, 2020-05-03:
@schenlap what would you jq expression look like to check if the value received was for a particular channel and then select the actual value?

schenlap, 2020-05-03:
`.data.uuid`
`jq .data.tuples[0][1]`

andig, 2020-05-03:
You need to express your if then as jq query for this to work.

andig, 2020-05-18:
Another option of expression this is 

    .data | select(.uuid==\"xxxxxx-wwwww-vvvvvv-zzzzz-yyyyyyyyy\") .tuples[0][1]

I'll check how this could be implemented.

schenlap, 2020-05-18:
I like your solution better :-)

andig, 2020-05-19:
@schenlap magst Du mal testen? Mich würde v.a. interessieren ob es klappt mehrere Werte (Leistung, Energie) auszulesen.

PR #171 (2020-05-20): Add websocket handler
Fix #66
closed 2020-05-20 by andig