evcc-io/evcc #68: Support meter with import & export power at the same time

schenlap, 2020-04-28:
My smart meter (AMIS Zähler) returns import and export power at the same time. Both values are positive. The smartmeter is read with vzlogger and optical interface EN 13757 (M-Bus).
To calculate the real export power, the import power has to be substracted from the export power.

andig, 2020-04-28:
Relates to #66

TheRealKerni, 2020-04-28:
So basically the values are not balanced? (Saldierend)

andig, 2020-04-28:
Same situation really as with the SMA meter and the 2 included OBIS codes.

andig, 2020-05-02:
@schenlap your scenario is using vzlogger as meter? I could suggest something like this:

```yml
- name: grid
  type: default
  power:
    type: importexport # name tbd
    import:
      type: http
      uri: https://volkszaehler/api/data/bezug.json?from=now
      jq: .data.tuples[0][1] # parse response json
    export:
      type: http
      uri: https://volkszaehler/api/data/bezug.json?from=now
      jq: .data.tuples[0][1] # parse response json
```

If there are any more general but less complex ideas please let me know.

schenlap, 2020-05-02:
looks good to me.

schenlap, 2020-05-02:
or add an new meter type that hase two meters as input and define a math function for it. But it is not less compex

```
meters:
- name: vzimport
  type: default
  power:
    type: http
    uri: http://volkszaehler/api/data/<uuid>.json?from=now
    jq: .data.tuples[0][1] # parse response json
- name: vzexport
  type: default
  power:
    type: http
    uri: http://volkszaehler/api/data/<uuid>.json?from=now
    jq: .data.tuples[0][1] # parse response json
- name: grid
  type: math
  power1:
    meter: vzimport
  power2:
    meter: vzexport
  func: power2 - power1
```

TheRealKerni, 2020-05-02:
Looks like a good case for the template system: https://github.com/andig/evcc/issues/86

andig, 2020-05-02:
> or add an new meter type that hase two meters as input and define a math function for it. But it is not less compex
> 
> ```
>   func: power2 - power1
> ```

That would require a math parser which is additional implementation complexity. We could also make this a (less specific) `add` node which takes a list of inputs. To subtract, add `scale: -1` to the sub nodes.

PR #93 (2020-05-02): Add calc plugin
Fix #68
closed 2020-05-02 by andig