evcc-io/evcc #73: Add Kostal Smart Energy Manager (Wifi)

andig, 2020-04-29:
See https://www.tq-group.com/filedownloads/files/products/automation/software_tools/energy-management/TQ_EM_JSON-API.0104.pdf

fiolux, 2020-04-29:
KOSTAL Smart Energy Meter - Interface Description MODBUS:
https://www.kostal-solar-electric.com/en-gb/download/-/media/document-library-folder---kse/2019/05/09/13/57/ba_kostal_interface_ksem---201911.pdf

andig, 2020-04-29:
@fiolux you‘ve posted the modbus docs- see #74 for modbus

andig, 2020-04-29:
I would need an EMxxx for testing. Please pm me at cpuidle@gmx.de

andig, 2020-05-01:
After discussion it seems that all KOSTAL SEM modules seem to support ModBus TCP and the SunSpec protocol. Hence, they are fully supported by EVCC using the `modbus` meter type. 

Therefore closing this issue as won‘t fix. If still needed please let me know.

closed 2020-05-01 by andig