evcc-io/evcc #74: Add Kostal Smart Energy Meter (Modbus)

andig, 2020-04-29:
See https://www.tq-group.com/filedownloads/files/products/automation/software_tools/energy-management/TQ_EM_Modbus_Slave.0013.pdf. 

Not sure if this is needed or if the SMM can be read via the KOSTAL SunSpec model.

It would be great to get remote access for testing.

Caibuk, 2020-04-29:
@andig I have a Kostal Plenticore Plus in combination with a Kostal Smart Energy Meter. I think I could expose you the Modbus/TCP-Port of the meter and/or the Plenticore.

fiolux, 2020-04-29:
KOSTAL Smart Energy Meter - Interface Description MODBUS:
https://www.kostal-solar-electric.com/en-gb/download/-/media/document-library-folder---kse/2019/05/09/13/57/ba_kostal_interface_ksem---201911.pdf

andig, 2020-04-29:
> I think I could expose you the Modbus/TCP-Port of the meter and/or the Plenticore.

@Caibuk could you pm me at cpuidle@gmx.de?

andig, 2020-04-29:
Looks as if this is working now due to #58:

```yaml
meters:
- name: netz
  type: modbus
  model: ksem # unknown models will be interpreted as sunspec device
  uri: ...:502
  id: 1
  power: Power
  energy: Export
```

Can this be closed?

closed 2020-04-29 by andig