evcc-io/evcc #75: Add: Phoenix Contact charge controllers EVCC Basic (RTU-Modbus)

22kW, 2020-04-29:
Ich habe einen Phoenix Contact charge controller der EV-CC-AC1-M3 Serie in meiner Wallbox. Die Kommunikation geht über RS-485 Modbus. Hätte auch ein paar Python Beispiele bei Interesse:
https://www.phoenixcontact.com/online/portal/de/?uri=pxc-oc-itemdetail:pid=1622459

andig, 2020-04-29:
Sollte das nicht dieser hier sein: https://github.com/andig/evcc/blob/master/charger/phoenix.go? Evtl. reicht hier nur die Doku nicht?

22kW, 2020-04-29:
Kann leider kein go könnte nur meinen Python code beisteuern.

andig, 2020-04-29:
Schau doch einfach mal rein. Der bestehende hat nur 4 Modbus Register:

	phRegStatus     = 100 // Input
	phRegChargeTime = 102 // Input
	phRegMaxCurrent = 300 // Holding
	phRegEnable     = 400 // Coil

Wenn Dein Controller ähnlich ist sollte es sehr einfach sein.

22kW, 2020-04-29:
Also bei mir sieht es eher so aus:

setEvcMaxCurrent: 	evc.write_register(22000, current, numberOfDecimals=0, functioncode=6, signed=False)
getEvcStatus():	evStatusTmp = evc.read_string(24000, numberOfRegisters=1, functioncode=3)
setEvcCharging(enable):	evc.write_bit(20000, 1, functioncode=5)
enableEvcChargingRemotecontrol():	evc.write_register(4000, 3, numberOfDecimals=0, functioncode=6, signed=False)

closed 2020-04-29 by andig

andig, 2020-04-29:
@22kW ist gemerged aber nicht mehr im 0.6er Release. Versuch es bitte mal. Bei Fehlern bitte Log mit `log: trace` posten.

PR #76 (2020-04-29): Add Phoenix EV-CC-AC1-M
Fix #75. Depends on refactor of #58.