evcc-io/evcc #78: Regulation "PV Only" disables charger

schenlap, 2020-04-29:
I am manually simulating same situations.
My testsetup: Only gridmeter and charger (without meter). PV Only

```
[main] INFO 2020/04/29 11:37:24 evcc dev (HEAD)
[main] INFO 2020/04/29 11:37:24 using config file evcc_configurable.yaml
[main] INFO 2020/04/29 11:37:24 listening at 0.0.0.0:7070
[mqtt] INFO 2020/04/29 11:37:24 connecting evcc-530756363 at localhost:1883
[wifi] WARN 2020/04/29 11:37:24 -- experimental --
[load] INFO 2020/04/29 11:37:24 main config: vehicle ✓ grid ✓ pv — charge —
[load] INFO 2020/04/29 11:37:24 main charger: power — energy — timer —
...
[load] DEBUG 2020/04/29 11:39:52 main charger status: B
[load] DEBUG 2020/04/29 11:39:52 main grid power: 1700.0W
[load] DEBUG 2020/04/29 11:39:52 main charge power: 0.0W
[load] DEBUG 2020/04/29 11:39:52 main target power: -1700W = 0W charge - 1700W grid - 0W residual
[load] DEBUG 2020/04/29 11:39:52 main target charge current: 0A
[load] DEBUG 2020/04/29 11:39:52 main vehicle soc: 50.0%
[load] DEBUG 2020/04/29 11:40:02 main charger status: B
[load] DEBUG 2020/04/29 11:40:02 main grid power: -1700.0W
[load] DEBUG 2020/04/29 11:40:02 main charge power: 0.0W
[load] DEBUG 2020/04/29 11:40:02 main target power: 1700W = 0W charge - -1700W grid - 0W residual
[load] DEBUG 2020/04/29 11:40:02 main target charge current: 7A
[load] DEBUG 2020/04/29 11:40:02 main charger enable - contactor delay 1m30s
[load] DEBUG 2020/04/29 11:40:02 main vehicle soc: 50.0%
[load] DEBUG 2020/04/29 11:40:12 main charger status: C
[load] INFO 2020/04/29 11:40:42 main start charging ->
[load] DEBUG 2020/04/29 11:40:12 main grid power: -1700.0W
[load] DEBUG 2020/04/29 11:40:12 main charge power: 0.0W
[wrap] DEBUG 2020/04/29 11:40:12 main set charge power: 0W
[load] DEBUG 2020/04/29 11:40:12 main target power: 1700W = 0W charge - -1700W grid - 0W residual
[load] DEBUG 2020/04/29 11:40:12 main target charge current: 7A
[load] DEBUG 2020/04/29 11:40:12 main charger enable - contactor delay 1m20s
[load] DEBUG 2020/04/29 11:40:12 main vehicle soc: 50.0%
--> Reduce grid power, because charger would consume ~1600W
[load] DEBUG 2020/04/29 11:40:22 main charger status: C
[load] DEBUG 2020/04/29 11:40:22 main grid power: -100.0W
[load] DEBUG 2020/04/29 11:40:22 main charge power: 0.0W
[wrap] DEBUG 2020/04/29 11:40:22 main set charge power: 0W
[load] DEBUG 2020/04/29 11:40:22 main target power: 100W = 0W charge - -100W grid - 0W residual
[load] DEBUG 2020/04/29 11:40:22 main target charge current: 0A
[load] DEBUG 2020/04/29 11:40:22 main vehicle soc: 50.0%
```

Charging is startet, I reduce grid import power to 100W (value = -100) (because charger consumes 1600W), in the next cycle main target power is calculated wrong and charger gets disabled. main target power should include already set power for the second cycle.
Also if not reducing grid power the charger get disabled.

Attached is my configuration.
[evcc_configurable_short.txt](https://github.com/andig/evcc/files/4551524/evcc_configurable_short.txt)

Maybe it is only a simulation issue.

andig, 2020-04-29:
You have consigured a grid meter like this:

    cmd: /bin/sh -c "cat /tmp/evcc/grid"

What's in that file? If zero then your charge power will always be zero. Set `log: trace` to see all values going around.

schenlap, 2020-04-29:
I think you see it in the logs. At start 1700 (import power), changed it to -1700 (export now)
`[load] DEBUG 2020/04/29 11:40:02 main grid power: -1700.0W` and set it to -100 (export reduced due too charging power required)
`[load] DEBUG 2020/04/29 11:40:22 main grid power: -100.0W`
. I expected the controller to keep current power, not to shut off.

Also if I do not reduze to -100 and keep -1700 the charger gets disabled. I would expect that the target current should increase because there would be enough export power.

andig, 2020-04-29:
> main charge power: 0.0W

The loadpoint cannot know how that happens. Your problem is the line above.

andig, 2020-04-29:
> I expected the controller to keep current power, not to shut off.

Sorry, where do you see the shutoff? Please put comments into the log ;)

schenlap, 2020-04-29:
Sorry, this was maybe not a good example. I hava a new log I don't understand.
```
GRIP Power is -1700, Charger Status = B, Mode=PV Only
[load] DEBUG 2020/04/29 19:33:50 main charger status: B
[exec] TRACE 2020/04/29 19:33:50 /bin/sh -c cat /tmp/evcc/grid: -1700
[load] DEBUG 2020/04/29 19:33:50 main grid power: -1700.0W
[load] DEBUG 2020/04/29 19:33:50 main charge power: 0.0W
[load] DEBUG 2020/04/29 19:33:50 main target power: 1700W = 0W charge - -1700W grid - 0W residual
[load] DEBUG 2020/04/29 19:33:50 main target charge current: 7A
[exec] TRACE 2020/04/29 19:33:50 /bin/sh -c echo true > /tmp/evcc/enable: 
[load] INFO 2020/04/29 19:33:50 main charger enable

--> OK Charger gets enabled. This was expected as grid power is available

[load] DEBUG 2020/04/29 19:33:50 main vehicle soc: 50.0%
[iflx] TRACE 2020/04/29 19:33:51 writing 9 point(s)
-[srvr] TRACE 2020/04/29 19:33:52 GET   /api/health     1.290053ms
[srvr] TRACE 2020/04/29 19:33:55 GET    /api/health     237.953µs
[srvr] TRACE 2020/04/29 19:33:57 GET    /api/health     503.268µs
[srvr] TRACE 2020/04/29 19:34:00 GET    /api/health     1.242044ms
[exec] TRACE 2020/04/29 19:34:00 /bin/sh -c cat /tmp/evcc/charge_status: B
[load] DEBUG 2020/04/29 19:34:00 main charger status: B
[exec] TRACE 2020/04/29 19:34:00 /bin/sh -c cat /tmp/evcc/grid: -1700
[load] DEBUG 2020/04/29 19:34:00 main grid power: -1700.0W
[load] DEBUG 2020/04/29 19:34:00 main charge power: 0.0W

-> OK, charge power is assumed as 0 as charger status is B -> connected but not loading

[load] DEBUG 2020/04/29 19:34:00 main target power: 1700W = 0W charge - -1700W grid - 0W residual
[load] DEBUG 2020/04/29 19:34:00 main target charge current: 7A
[load] DEBUG 2020/04/29 19:34:00 main set charge current: 7A
[exec] TRACE 2020/04/29 19:34:00 /bin/sh -c cat /tmp/evcc/maxcurrent: 16
[load] DEBUG 2020/04/29 19:34:00 main vehicle soc: 50.0%
[srvr] TRACE 2020/04/29 19:34:02 GET    /api/health     302.324µs
[srvr] TRACE 2020/04/29 19:34:05 GET    /api/health     1.182421ms
[srvr] TRACE 2020/04/29 19:34:07 GET    /api/health     379.557µs
[srvr] TRACE 2020/04/29 19:34:10 GET    /api/health     416.205µs

-> I have set charger_status to C now -> loading. It is read correct on next line

[exec] TRACE 2020/04/29 19:34:10 /bin/sh -c cat /tmp/evcc/charge_status: C
[load] DEBUG 2020/04/29 19:34:10 main charger status: C
[load] INFO 2020/04/29 19:34:10 main start charging ->
[exec] TRACE 2020/04/29 19:34:10 /bin/sh -c cat /tmp/evcc/grid: -1700
[load] DEBUG 2020/04/29 19:34:10 main grid power: -1700.0W
[load] DEBUG 2020/04/29 19:34:10 main charge power: -1610.0W

-> OK, charge power is calculated correct.

[wrap] DEBUG 2020/04/29 19:34:10 main set charge power: -1610W
[load] DEBUG 2020/04/29 19:34:10 main target power: 90W = -1610W charge - -1700W grid - 0W residual

-> Wrong. This is not target power, it is more an additional target power. 1610W charge should be ok for charging if there is 1700W export power. In real world the export power would decrease which would it make even worse. It would be: main target power -1700 = -1610 --90 grid -0W. It could also happen that grid power does not change much because pv increase at the same time which should result in an increase of target power.

Setting an constant grid export power must result in an increasing charging power!

[load] DEBUG 2020/04/29 19:34:10 main target charge current: 0A

-> target current = 0 is wrong. It should stay at 7A or increase as there is 1700W export.

[load] DEBUG 2020/04/29 19:34:10 main set charge current: 6A <- begin to reduce becaus target power is 0
[exec] TRACE 2020/04/29 19:34:10 /bin/sh -c cat /tmp/evcc/maxcurrent: 16
[load] DEBUG 2020/04/29 19:34:10 main vehicle soc: 50.0%
[srvr] TRACE 2020/04/29 19:34:12 GET    /api/health     1.282063ms
[srvr] TRACE 2020/04/29 19:34:15 GET    /api/health     280.199µs
[srvr] TRACE 2020/04/29 19:34:17 GET    /api/health     336.284µs
[srvr] TRACE 2020/04/29 19:34:20 GET    /api/health     2.034352ms
[exec] TRACE 2020/04/29 19:34:20 /bin/sh -c cat /tmp/evcc/charge_status: C
[load] DEBUG 2020/04/29 19:34:20 main charger status: C
[exec] TRACE 2020/04/29 19:34:20 /bin/sh -c cat /tmp/evcc/grid: -1700
[load] DEBUG 2020/04/29 19:34:20 main grid power: -1700.0W
[load] DEBUG 2020/04/29 19:34:20 main charge power: -1610.0W
[wrap] DEBUG 2020/04/29 19:34:20 main set charge power: -1610W
[load] DEBUG 2020/04/29 19:34:20 main target power: 90W = -1610W charge - -1700W grid - 0W residual
[load] DEBUG 2020/04/29 19:34:20 main target charge current: 0A
[exec] TRACE 2020/04/29 19:34:20 /bin/sh -c echo false > /tmp/evcc/enable: 

-> disabled because of wrong target power

[load] INFO 2020/04/29 19:34:20 main charger disable
[load] DEBUG 2020/04/29 19:34:20 main vehicle soc: 50.0%
[iflx] TRACE 2020/04/29 19:34:21 writing 9 point(s)
[srvr] TRACE 2020/04/29 19:34:22 GET    /api/health     291.585µs
[srvr] TRACE 2020/04/29 19:34:25 GET    /api/health     1.244574ms
```
I hope it is clear now.

andig, 2020-05-01:
OK, lets see.

> GRID Power is -1700, Charger Status = B, Mode=PV Only
> --> OK Charger gets enabled. This was expected as grid power is available
> -> OK, charge power is assumed as 0 as charger status is B -> connected but not loading

ok

```
[load] DEBUG 2020/04/29 19:34:00 main target power: 1700W = 0W charge - -1700W grid - 0W residual
[load] DEBUG 2020/04/29 19:34:00 main target charge current: 7A
[load] DEBUG 2020/04/29 19:34:00 main set charge current: 7A
[exec] TRACE 2020/04/29 19:34:00 /bin/sh -c cat /tmp/evcc/maxcurrent: 16
```
Side note: this config is wrong. You're *setting* the current, this should write into the file, not `cat` it. The loadpoint thinks it has successfully set 7A now.
```
[load] DEBUG 2020/04/29 19:34:00 main vehicle soc: 50.0%
```
> -> I have set charger_status to C now -> loading. It is read correct on next line

ok
```
[exec] TRACE 2020/04/29 19:34:10 /bin/sh -c cat /tmp/evcc/charge_status: C
[load] DEBUG 2020/04/29 19:34:10 main charger status: C
[load] INFO 2020/04/29 19:34:10 main start charging ->
[exec] TRACE 2020/04/29 19:34:10 /bin/sh -c cat /tmp/evcc/grid: -1700
[load] DEBUG 2020/04/29 19:34:10 main grid power: -1700.0W
[load] DEBUG 2020/04/29 19:34:10 main charge power: -1610.0W
```
> -> OK, charge power is calculated correct.

Nope, this is the problem. Charge Power should be positive! I'm not sure why this happens at this point.

schenlap, 2020-05-01:
what do you think about this fix:

```
--- a/core/loadpoint.go
+++ b/core/loadpoint.go
@@ -212,9 +212,11 @@ func (lp *LoadPoint) evChargeCurrentHandler(m *wrapper.ChargeMeter) func(para ..
                }
                if current > 0 {
                        // limit available power to generation plus consumption/ minus delivery
-                       availablePower := math.Abs(lp.pvPower) + lp.availableBatteryPower() + lp.gridPower
-                       availableCurrent := int64(powerToCurrent(availablePower, lp.Voltage, lp.Phases))
-                       current = min(current, availableCurrent)
+                       availablePower := math.Abs(lp.pvPower) + lp.availableBatteryPower() - lp.gridPower
+                       totalPower := availablePower + lp.chargePower
+                       log.DEBUG.Printf("read *schenlap* %.1f %.1f %.1f %.1f", lp.pvPower,  lp.gridPower, availablePower, totalPower )
+                       totalCurrent := int64(powerToCurrent(totalPower, lp.Voltage, lp.Phases))
+                       current = min(current, totalCurrent)
                }
                m.SetChargeCurrent(current)
        }
```

schenlap, 2020-05-01:
pvPower must also be removed.

```
--- a/core/loadpoint.go
+++ b/core/loadpoint.go
@@ -212,13 +212,15 @@ func (lp *LoadPoint) evChargeCurrentHandler(m *wrapper.ChargeMeter) func(para ..
                }
                if current > 0 {
                        // limit available power to generation plus consumption/ minus delivery
-                       availablePower := math.Abs(lp.pvPower) + lp.availableBatteryPower() + lp.gridPower
-                       availableCurrent := int64(powerToCurrent(availablePower, lp.Voltage, lp.Phases))
-                       current = min(current, availableCurrent)
+                       availablePower := lp.availableBatteryPower() - lp.gridPower
+                       totalPower := availablePower + lp.chargePower
+                       totalCurrent := int64(powerToCurrent(totalPower, lp.Voltage, lp.Phases))
+                       current = min(current, totalCurrent)
                }
                m.SetChargeCurrent(current)
        }
 }
```

this is only correct if gridmeter is installed. If only pv meter is installen, it will not word. I don't really know how this different situations are handled intern, so I can not fix it

andig, 2020-05-01:
Could you describe what you perceive to be the problem instead of posting the solition? Maybe the Slack group is easier for this.

closed 2020-05-01 by andig