evcc-io/evcc #79: Add ABL Sursum emh Controller (EV Charger)

Roland-git, 2020-04-29:
It would be cool if you could integrate the ABL Charger.
Communication reference: 
https://manualsbrain.com/de/manuals/703790/?page=7
I could sent out an ABL emh1 (1-phase), to test.

andig, 2020-04-30:
@Roland-git looking at the docs I'm not quite sure how the ABL works. It does seem to have Modbus (as master or slave?) and yet speek some sort of TCP'ish protocol. It doesn't mention anything about how to connect. Could you elaborate further or is there example code I could look at?

andig, 2020-05-04:
ping @Roland-git

andig, 2020-05-12:
ping @Roland-git still need help ;)

andig, 2020-07-03:
Ping @Roland-git kein Interesse mehr?

andig, 2020-07-06:
Closing due to lack of interest

closed 2020-07-06 by andig

andig, 2022-07-28:
Siehe https://github.com/evcc-io/evcc/discussions/3844#discussioncomment-3268760

antonhagg, 2024-04-02:
Have there been any more discussions regarding this charger?

andig, 2024-04-02:
Aus verlinkter Diskussion, hast Du sicher schon gelesen:

> Gibts denn hier noch mehr Bedarf diese Uraltbox zu unterstützen so dass sich der Aufwand lohnen könnte?
> Ach die ganz alte ABL EVCC...
> PR welcome, bei Codingfragen kann ich gerne unterstützen.