evcc-io/evcc #82: Add E3DC

andig, 2020-04-30:
In #52 we're discussing support how home batteries. It would be great to get access to one for understanding the possibilities. E3DC would be most helpful as it supports Modbus TCP and could be integrated right away.

andig, 2020-04-30:
Closed by #83. Still to be validated, but that should cover E3DC, too.

goebelmeier, 2020-04-30:
I would be more than happy to validate. Could you give me a little introduction how to do it?

andig, 2020-04-30:
Should work as documented here: https://github.com/andig/evcc/pull/83. Would be great to get a report of the results!

goebelmeier, 2020-05-01:
Currently my config looks like:
`- name: e3dc
  type: modbus
  model: e3dc
  uri: 192.168.4.77:502
  id: 40070`
but as E3DC doesn't support sun-spec, im receiving an error while launching evcc:
`[main] INFO 2020/05/01 19:37:45 evcc dev (HEAD)
[main] INFO 2020/05/01 19:37:45 using config file /etc/evcc.yaml
[main] INFO 2020/05/01 19:37:45 listening at 0.0.0.0:7070
[mqtt] INFO 2020/05/01 19:37:45 connecting evcc-1741695561 at localhost:1883
[mqtt] TRACE 2020/05/01 19:37:45 localhost:1883 connected
[modb] TRACE 2020/05/01 19:37:45 modbus: send 00 01 00 00 00 06 86 03 9c 40 00 02
[modb] TRACE 2020/05/01 19:37:45 modbus: recv 00 01 00 00 00 07 86 03 04 e3 dc 01 02
[modb] TRACE 2020/05/01 19:37:45 modbus: send 00 02 00 00 00 06 86 03 c3 50 00 02
[modb] TRACE 2020/05/01 19:37:45 modbus: recv 00 02 00 00 00 03 86 83 02
[modb] TRACE 2020/05/01 19:37:45 modbus: send 00 03 00 00 00 06 86 03 00 00 00 02
[modb] TRACE 2020/05/01 19:37:45 modbus: recv 00 03 00 00 00 07 86 03 04 e3 dc 01 02
[modb] FATAL 2020/05/01 19:37:45 not a SunSpec device`

E3DC Modbus/TCP docs can be found googling or i could send it to you via e-mail if needed.

andig, 2020-05-01:
Die ID stimmt sicher nicht- das sieht aus wie ein typisches ModBus SunSpec Register. ID 1 geht nicht?

andig, 2020-05-01:
Probier bitte mal ID 5, analog https://www.symcon.de/forum/threads/39963-E3-DC-Hauskraftwerk-Modbus-Schnittstelle

goebelmeier, 2020-05-02:
Unfortunately not. 

```
[12:33:19] root@openHABianPi:~# grep -A5 e3dc -m1 /etc/evcc.yaml
- name: e3dc
  type: modbus
  model: e3dc
  uri: 192.168.4.77:502
  id: 5
- name: charge
[12:33:25] root@openHABianPi:~# ./go/bin/evcc --log trace
[main] INFO 2020/05/02 12:33:35 evcc dev (HEAD)
[main] INFO 2020/05/02 12:33:35 using config file /etc/evcc.yaml
[main] INFO 2020/05/02 12:33:35 listening at 0.0.0.0:7070
[mqtt] INFO 2020/05/02 12:33:36 connecting evcc-100969438 at localhost:1883
[mqtt] TRACE 2020/05/02 12:33:36 localhost:1883 connected
[mqtt] TRACE 2020/05/02 12:33:36 localhost:1883 subscribe openhab/out/MeterCurrentActivePowerPurchase/state
[modb] TRACE 2020/05/02 12:33:36 modbus: send 00 01 00 00 00 06 05 03 9c 40 00 02
[modb] TRACE 2020/05/02 12:33:36 modbus: recv 00 01 00 00 00 07 05 03 04 e3 dc 01 02
[modb] TRACE 2020/05/02 12:33:36 modbus: send 00 02 00 00 00 06 05 03 c3 50 00 02
[modb] TRACE 2020/05/02 12:33:36 modbus: recv 00 02 00 00 00 03 05 83 02
[modb] TRACE 2020/05/02 12:33:36 modbus: send 00 03 00 00 00 06 05 03 00 00 00 02
[modb] TRACE 2020/05/02 12:33:36 modbus: recv 00 03 00 00 00 07 05 03 04 e3 dc 01 02
[modb] FATAL 2020/05/02 12:33:36 not a SunSpec device```

andig, 2020-05-02:
Sorry- that setting is determined by your installation (id 1 is also very common). Feel free to email me for remote access.

amigatommy, 2020-05-02:
Protokoll lässt sich am E3DC-Gerät einstellen: E3DC oder SUN_SPEC ! Ebenso die ID, Standard ist ID 1.

andig, 2020-05-02:
@amigatommy bedeutet das, dass die E3DC nur entweder oder sprechen? Bei den meisten ModBus Geräten ist es eigentlich üblich, dass sie sowohl einen eigenen Standard als auch SunSpec unterstützen und diese eben auf unterschiedlichen Registern abbilden?

amigatommy, 2020-05-02:
@andig Ja, laut Beschreibung entweder E3DC oder SUN-SPEC, leider ;o(

andig, 2020-05-02:
Sieht so aus: https://www.loxwiki.eu/download/attachments/9339566/ModuBus-Dokumentation_2016-08-05.pdf?version=1&modificationDate=1471507657000&api=v2#page8. Warum einfach wenns auch kompliziert geht. 

@goebelmeier Könntest Du das umstellen? Sonst müsste der E3DC nachprogrammiert werden.

andig, 2020-05-03:
Mit Hilfe von #96 wäre es möglich, die Modbus Register auch manuell zu setzen. Hier eine Konfiguration für den E3DC:

```yaml
- name: e3dc-grid
  type: default
  power:
    type: modbus
    uri: e3dc.fritz.box:502
    id: 1
    register: # manual register configuration
      address: 40072
      length: 2
      type: holding
      decode: int32
- name: e3dc-battery
  type: default
  power:
    type: modbus
    uri: e3dc.fritz.box:502
    id: 1
    register: # manual register configuration
      address: 40070
      length: 2
      type: holding
      decode: int32
    scale: -1 # reverse direction
- name: e3dc-charge
  type: default
  power:
    type: modbus
    uri: e3dc.fritz.box:502
    id: 1
    register: # manual register configuration
      address: 40078
      length: 2
      type: holding
      decode: int32
```

Wäre Klasse wenn das jemand ausprobieren und bestätigen könnte.

andig, 2020-05-03:
@goebelmeier: hast Du auch eine e3dc Wallbox? Wäre spannend zu wissen ob das "nur" eine umgelabelte Wallbe ist- sie sieht nämlich so aus ;)

amigatommy, 2020-05-03:
Ich habe auch die E3DC Wallbox easy connect, es ist eine Wallbe ! Sieht man direkt, wenn man das Webinterface der Box aufruft ;o) Leider gibt es kein Passwort für die erweiterten Einstellungen, das Standardpasswort des Phoenix Contact Controllers funktioniert leider nicht :o( Aber man kann über ModBus auf die Box zugreifen.

andig, 2020-05-03:
@amigatommy hast Du auch einen Speicher? Könntest Du bestätigen, dass sowohl die Konfiguration für Speicher und Zähler oben als auch die Konfiguration der Wallbox als Wallbe bei Zugriff durch EVCC funktionieren? Das wäre genial :)

amigatommy, 2020-05-03:
Ja, habe auch einen Speicher, leider nutze ich aber EVCC nicht, mache alles über fhem, Steuerung der Wallbox erfolgt ja auch direkt durch das E3/DC Hauskraftwerk. Hab auch im Moment wenig Zeit zum testen :o(

goebelmeier, 2020-05-04:
@andig eine Wallbox habe ich leider nicht zur Verfügung, aber ich habe sie probehalber mal in meine Config aufgenommen. Für den Grid Meter hätte ich eher 40074 statt 40072 genommen, oder? Die Beschreibung von 40074 ist _Leistung am Netzübergabepunkt in Watt (negative Werte = Einspeisung)_ und von 40072 _Hausverbrauchs-Leistung in Watt_. Leider bekomme ich für die PV-Leistung noch keine Werte und ich sehe meinen Fehler noch nicht:

```
meters:
- name: e3dc-pv
  type: default
  power:
    type: modbus
    uri: 192.168.4.77:502
    id: 1
    register:
      address: 40068
      length: 2
      type: holding
      decode: int32
- name: e3dc-grid
  type: default
  power:
    type: modbus
    uri: 192.168.4.77:502
    id: 1
    register: # manual register configuration
      address: 40074
      length: 2
      type: holding
      decode: int32
- name: e3dc-battery
  type: default
  power:
    type: modbus
    uri: 192.168.4.77:502
    id: 1
    register: # manual register configuration
      address: 40070
      length: 2
      type: holding
      decode: int32
    scale: -1 # reverse direction
- name: e3dc-charge
  type: default
  power:
    type: modbus
    uri: 192.168.4.77:502
    id: 1
    register: # manual register configuration
      address: 40078
      length: 2
      type: holding
      decode: int32

loadpoints:
- name: main # name for logging
  vehicle: default
  charger: configurable # charger
  meters:
    grid: e3dc-grid # grid meter
    pv: e3dc-pv # pv meter
    battery: e3dc-battery
    charge: e3dc-charge # charge meter
  sensitivity: 1 # current raise/lower steps size (default 1A)
  guardduration: 10m # switch charger contactor not more often than this (default 10m)
  maxcurrent: 16 # maximum charge current (default 16A)
  phases: 3 # ev phases (default 3)
```

```
[17:19:33] root@openHABianPi:~# ./go/bin/evcc --log trace
[main] INFO 2020/05/04 17:20:32 evcc dev (HEAD)
[main] INFO 2020/05/04 17:20:32 using config file /etc/evcc.yaml
[main] INFO 2020/05/04 17:20:32 listening at 0.0.0.0:7070
[mqtt] INFO 2020/05/04 17:20:34 connecting evcc-1104223532 at localhost:1883
[mqtt] TRACE 2020/05/04 17:20:34 localhost:1883 connected
[load] INFO 2020/05/04 17:20:34 main loadpoint config: vehicle ✓ grid ✓ pv ✓ battery ✓ charge ✓
[load] INFO 2020/05/04 17:20:34 main charger config: power — energy — timer —
[load] INFO 2020/05/04 17:20:34 main charge mode: off
[exec] TRACE 2020/05/04 17:20:34 /bin/sh -c echo true: true
[load] INFO 2020/05/04 17:20:34 main charger enabled
[load] DEBUG 2020/05/04 17:20:34 main set charge current: 6A
[exec] TRACE 2020/05/04 17:20:34 /bin/sh -c echo 6: 6
[exec] TRACE 2020/05/04 17:20:34 /bin/sh -c echo F: F
[load] DEBUG 2020/05/04 17:20:34 main charger status: F
[modb] TRACE 2020/05/04 17:20:34 modbus: send 00 01 00 00 00 06 01 03 9c 8a 00 02
[modb] TRACE 2020/05/04 17:20:34 modbus: recv 00 01 00 00 00 07 01 03 04 ff ff 00 00
[load] DEBUG 2020/05/04 17:20:34 main grid power: -65536.0W
[modb] TRACE 2020/05/04 17:20:34 modbus: send 00 02 00 00 00 06 01 03 9c 84 00 02
[modb] TRACE 2020/05/04 17:20:34 modbus: recv 00 02 00 00 00 07 01 03 04 00 00 00 00
[load] DEBUG 2020/05/04 17:20:34 main pv power: 0.0W
[modb] TRACE 2020/05/04 17:20:34 modbus: send 00 03 00 00 00 06 01 03 9c 86 00 02
[modb] TRACE 2020/05/04 17:20:34 modbus: recv 00 03 00 00 00 07 01 03 04 00 00 05 f3
[load] DEBUG 2020/05/04 17:20:34 main battery power: -1523.0W
[modb] TRACE 2020/05/04 17:20:34 modbus: send 00 04 00 00 00 06 01 03 9c 8e 00 02
[modb] TRACE 2020/05/04 17:20:34 modbus: recv 00 04 00 00 00 07 01 03 04 00 00 00 00
[load] DEBUG 2020/05/04 17:20:34 main charge power: 0.0W
[modb] TRACE 2020/05/04 17:20:39 modbus: closing connection due to idle timeout: 5.000137006s
```

andig, 2020-05-04:
> eine Wallbox habe ich leider nicht zur Verfügung, aber ich habe sie probehalber mal in meine Config aufgenommen. Für den Grid Meter hätte ich eher 40074 statt 40072 genommen, oder? Die Beschreibung [...]

Denke auch, mein Fehler.

> Leider bekomme ich für die PV-Leistung noch keine Werte und ich sehe meinen Fehler noch nicht:

Die Config sieht korrekt aus. Scheint als würde Dein e3dc das einfach nicht liefern. Wäre aus meiner Sicht eine Frage an den Support?

andig, 2020-05-04:
Das sieht auch sehr merkwürdig aus:

> [modb] TRACE 2020/05/04 17:20:34 modbus: recv 00 01 00 00 00 07 01 03 04 ff ff 00 00
> [load] DEBUG 2020/05/04 17:20:34 main grid power: -65536.0W

Die `FF FF` "stinken".

Die Alternative wäre mal auf SunSpec umzustellen und schauen ob das besser funktioniert.

amigatommy, 2020-05-04:
@goebelmeier Beim auslesen mit fhem gab es auch Probleme, versuch mal die Registeradresse aus der E3DC Beschreibung -1.

goebelmeier, 2020-05-05:
SunSpec hätte den Nachteil, dass ich nur einen Bruchteil der Metriken bekomme, die ich aktuell per openHAB-Plugin (https://community.openhab.org/t/e3dc-with-the-new-modbus-binding/51345) beziehe. Und über openHAB stimmen die Werte auch. Ich sehe nur in dem Plugin nicht ob die noch irgendwelche Magie tun. Möchtest du mal per SSH auf meinen RPi und direkt gegen das E3DC spielen?

goebelmeier, 2020-05-05:
@amigatommy ja, an Off-by-one error dachte ich auch schon:

```
- name: e3dc-pv
  type: default
  power:
    type: modbus
    uri: 192.168.4.77:502
    id: 1
    register:
      address: 40067
      length: 2
      type: holding
      decode: int32
- name: e3dc-grid
  type: default
  power:
    type: modbus
    uri: 192.168.4.77:502
    id: 1
    register: # manual register configuration
      address: 40073
      length: 2
      type: holding
      decode: int32
- name: e3dc-battery
  type: default
  power:
    type: modbus
    uri: 192.168.4.77:502
    id: 1
    register: # manual register configuration
      address: 40069
      length: 2
      type: holding
      decode: int32
    scale: -1 # reverse direction
- name: e3dc-charge
  type: default
  power:
    type: modbus
    uri: 192.168.4.77:502
    id: 1
    register: # manual register configuration
      address: 40077
      length: 2
      type: holding
      decode: int32
```

das macht's aber leider nicht viel besser:
```
[modb] TRACE 2020/05/05 09:06:16 modbus: send 00 01 00 00 00 06 01 03 9c 89 00 02
[modb] TRACE 2020/05/05 09:06:16 modbus: recv 00 01 00 00 00 07 01 03 04 fc 2d ff ff
[load] DEBUG 2020/05/05 09:06:16 main grid power: -64094209.0W
[modb] TRACE 2020/05/05 09:06:16 modbus: send 00 02 00 00 00 06 01 03 9c 83 00 02
[modb] TRACE 2020/05/05 09:06:16 modbus: recv 00 02 00 00 00 07 01 03 04 05 cc 00 00
[load] DEBUG 2020/05/05 09:06:16 main pv power: 97255424.0W
[modb] TRACE 2020/05/05 09:06:16 modbus: send 00 03 00 00 00 06 01 03 9c 85 00 02
[modb] TRACE 2020/05/05 09:06:16 modbus: recv 00 03 00 00 00 07 01 03 04 00 00 00 00
[load] DEBUG 2020/05/05 09:06:16 main battery power: -0.0W
[modb] TRACE 2020/05/05 09:06:16 modbus: send 00 04 00 00 00 06 01 03 9c 8d 00 02
[modb] TRACE 2020/05/05 09:06:16 modbus: recv 00 04 00 00 00 07 01 03 04 00 00 00 00
[load] DEBUG 2020/05/05 09:06:16 main charge power: 0.0W
```

In https://www.loxforum.com/forum/hardware-zubeh%C3%B6r-sensorik/6019-n%C3%A4here-infos-zum-e3dc-stromspeicher?p=39964#post39964 steht noch:
_Negative Werte werden nicht als negativ angezeigt, sondern hex FFFF (65525 ->). Aus dem Grund die Formel für Batterie und Netzleistung. Vielleicht geht es ja auch anders._

Ist da irgendwie ein Trick vergraben?

andig, 2020-05-05:
@goebelmeier schick mir gerne PM für Raspi (cpuidle@gmx.de). Gibts zu OpenHab eine Doku gegen die ich checken könnte?

andig, 2020-05-05:
Also. Die Register ID muss immer -1, hier der Beweis:

    ❯ gor main.go -a localhost:5502 -d 1 -t holding read -e hex 40000 1         
    e3dc

Erkenntnis 2: es braucht einen neuen int32 Datentyp mit "geswappten" words:

    ❯ gor main.go -a localhost:5502 -d 1 -t holding read -e int32s 40073 2
    -14

    ❯ gor main.go -a localhost:5502 -d 1 -t holding read -e int32s 40069 2
    1245

    ❯ gor main.go -a localhost:5502 -d 1 -t holding read -e int32s 40077 2
    0

Dann stimmen sogar die negativen Zahlen. Ich reiche das Feature für MBMD+EVCC nach.

amigatommy, 2020-05-05:
@andig ich mache das mit den riesigen negativen Werten in fhem mit folgendem "Trick": z.B. Hausleistung h40071, unpack s> , und nur das Register h40071 also 16Bit abfragen, klappt einwandfrei 😊

andig, 2020-05-05:
Schöner ist natürlich trotzdem eine "richtige" Lösung die dann auch mit 32bit Werten ordentlich umgehen kann auch wenn das hier für Leistung gar nicht notwendig ist ;)

premultiply, 2020-05-05:
Taja... https://de.wikipedia.org/wiki/Byte-Reihenfolge ;)

closed 2020-05-05 by andig

amigatommy, 2020-05-05:
@andig 

> Schöner ist natürlich trotzdem eine "richtige" Lösung die dann auch mit 32bit Werten ordentlich umgehen kann auch wenn das hier für Leistung gar nicht notwendig ist ;)

Da hast Du natürlich Recht. Man müßte eigentlich beide Register getrennt auswerten, einmal Vorzeichen und zum anderen den Wert. Ich betrachte es mal so: ich lese nur den Wert aus, da er ja auch das Vorzeichen enthält (signed integer) und spare mir das andere Byte einfach ;o)

PR #100 (2020-05-05): Add swapped word order encodings
Fix #82