evcc-io/evcc #86: Add device templates for novice users

andig, 2020-05-01:
Examples of devices that are supported out of the box but not obviously so:
- SMA Home Manager and SMA Energy Meter (type „sma“)
- KOSTEM SEM and EMxx (modbus TCP)
- E3DC (modbus TCP)
- vzlogger (type „http“)
- any modbus RTU meter (SDM630 and any other) (modbus RTU oder TCP)

mfueser, 2020-05-23:
Hi!

I really would like to help out. I tried to get it somehow running the whole morning. 
I cannot figure out how to configure my Kostal Plenticore (with KSEM Energymeter installed) at all. I get it to show some readings, but those do not seem to be right compared to the values of the Plenticore Webinterface. It would help, to define which readings are necessary for operating with evcc.
I also tried adding my Zoe which does not seem to work, but I do not get logs about API-activity from evcc.
A test with modbus default and the API as specified here https://www.photovoltaikforum.com/core/attachment/81082-ba-kostal-interface-modbus-tcp-sunspec-pdf/ did not work too (readings were incorrect).

Any ideas on how to get started? I really appreciate the effort and would like to help, but I cannot get into it yet.

premultiply, 2020-05-23:
Maybe we can help you on the EVCC slack channel.

closed 2020-07-22 by andig