evcc-io/evcc #95: Make PV Mode robust against short power peaks / configurable pv mode

schenlap, 2020-05-03:
Currently evcc is reacting on every single measurement which may lead to a charge disable if grid power is read during the short spike (maybe only if charge current is at minimum, I don't know). A few things that I image that would be useful:

1. hold on time: grid power must be less than XX for YY sec/min to disable charger
2. holf off time: grid power must be greater than min charge power for xx sec/min to enable charger
3. Maybe also start / stop power limits could be configurable. Installations with small pv systems could configure to charge with e.g. 70% pv. Maybe setting the percentage value from ui would be useful. This is not comparable to "Minimum + PV" as this would starts immediately and not wait for sufficient pv power.

Caibuk, 2020-05-03:
I totally support these requests / proposals.
PV mode should be more robust against up and downs in export / import.
I'd rather use PV only as I don't want the car to charge after sun is down (we have grid high tariff between 17:00 and 20:30 o'clock).
Another thought:
- could the charging be suspended without switching contactors when there is not enough PV available? Like when I unlock the car, the car sets the power to 0 A for 20 or 30 seconds and continues to charge after that without switching the contactors. In Phoenix manual there is a modbus register 468 which maybe does something similar (function "Ladevorgang pausieren").

premultiply, 2020-05-03:
From Typ2 Mode 3 signaling there is no way to idle charge without opening the contactors.
The maximum current may only be set in a range from 6 to 80A per phase. Any value smaller than 6A results in stopping the charge and opening the contactor.

Caibuk, 2020-05-03:
Thanks for the clarification. Sadly, this is not the best behavior...

premultiply, 2020-05-03:
The only thing which can be done is implementing a minimum threshold power value (1320 W?) with a timeout after which the charge will be interrupted completly (5 Min?). This is how openWB behaves.

andig, 2020-05-03:
Great, thank you for all your comments! Lets start with what we currently have:

- `guardtime` will prevent switching on/off before `guardtime` is expired
- `sensitivity` controls by how much the charge current can be adjusted in each control cycle, that is each `interval` seconds (yes, I should add this to the readme).
- we also already open/close the contactor at minimum current only. I.e. if you switch `off` from max current that will happen in 2 steps: min current, then enable= false.

> - hold on time: grid power must be less than XX for YY sec/min to disable charger
> - hold off time: grid power must be greater than min charge power for xx sec/min to enable charger

Essentially, these two would replace `guardtime` with something more differentiated. Does it make sense to have both configurable? What are good default values?

> - Maybe also start / stop power limits could be configurable. Installations with small pv systems could configure to charge with e.g. 70% pv. Maybe setting the percentage value from ui would be useful. This is not comparable to "Minimum + PV" as this would starts immediately and not wait for sufficient pv power. The pv percentage setting should be a minimum pv value, charge power should only be increased if pv percentage is 100%. evcc should not regulate to the 70% value.

Sounds like an `enable threshold`. Which mode would that apply to: pv only? Would it be ok to look at grid export instead of pv production?

> - I'd rather use PV only as I don't want the car to charge after sun is down (we have grid high tariff between 17:00 and 20:30 o'clock).

Is that the same point as above?

> - The only thing which can be done is implementing a minimum threshold power value (1320 W?) with a timeout after which the charge will be interrupted completly (5 Min?). This is how openWB behaves.

Sounds like a `disable threshold`. Would that be in addition to the `guardtime`?

schenlap, 2020-05-03:
> guardtime will prevent switching on/off before guardtime is expired
Guardtime is startet when charger is enabled? After guardtime the charger can be disabled immediately when grid power increase (import)? If so it is something different.

> sensitivity controls by how much the charge current can be adjusted
Only integer are possible, no float, or? With a intervall of 10s it will be about 1 minute from max to min. For small pv systems where we are near 6A the time is shorter

> Essentially, these two would replace guardtime with something more differentiated. Does it make sense to have both configurable? What are good default values?
I see guardtime more as a safety feature. If somethng is configured wrong not to switch too often. 
For me one value for on and off would be ok. I would also keep guardtime. Implementing one for "enable" and one for "disable" is not too much more work and will be useable by some users, i think.
For me a default value of 5m would be ok (same as @premultiply suggested)

> Sounds like an enable threshold. Which mode would that apply to: pv only? Would it be ok to look at grid export instead of pv production? Yes sure, grid export is much better than pv production. Yes, only pv mode. The mode "sofortladen" and "minimum + pv" are immediate modes for me.

I'm excited to see how the others see it.

Caibuk, 2020-05-06:
I would highly appreciate this disable/enable delay time. In the attached screenshot one can see how fluctuations in PV generation lead to a couple ons and offs, which probably could have been prevented with a delay. Please note the (almost accurate) 10 min guardtime intervals during the "off" phases at the end.

[image]

andig, 2020-05-06:
Working on that, highest priority.

Caibuk, 2020-05-06:
Sorry, if you perceived my post as too pushy. Just wanted to show my interest in this feature. I didn't meant to bother you at all!
You are doing an AWESOME job!

andig, 2020-05-06:
Depends on #104 (which is about ready).

andig, 2020-05-06:
> I would highly appreciate this disable/enable delay time.

I feel this is the time _before_ an on/off signal should really kick in once it has been detected. This is _not_ a general minimum hold duration/ release duration, right? Shall we call it `enable delay`/ `disable delay`.

andig, 2020-05-06:
So... lets split the discussion into thresholds (power) and delays (time). We need need default values:

- `enable delay`: 0 // immediately
- `disable delay`: 1m // keep charging for 1min at min current
- `enable threshold`: 0 // start charging if grid export  exceeds this value
- `disable threshold`: 1kw // stop charging if grid import exceeds this value

Do these settings and default make sense for you?

/cc @schenlap @Caibuk @premultiply

goebelmeier, 2020-05-06:
Do i need the enable threshold? I need to start charging with at least 6A * 230V/400V * `phases`. But maybe there are good use cases to manually define the enable threshold.

schenlap, 2020-05-06:
default delays are ok.

threshold enable: start charging if grid export exceeds min charging power (min charging current * voltage * phases) e.g 6*230*1 = 1380
threshold disable: stop charging if grid import exceeds 20% of threshold enable.

People with small pv systems need to change values. Default should be configured to mostly use pv power in pv mode.

premultiply, 2020-05-06:
The default delay should be much larger. Zero should be avoided completly as this would make the charge go on and off on cloudy days or while using hot plates, oven etc.

I would prefer default delays in the range from 60 to 300 sec by default. Maybe 120 secs on and off.

Delay in this case means the grid power export/import is over/below [threshold] for [delay] seconds.

Caibuk, 2020-05-06:
> `delays:`
> 
> * `enable`: 0 // immediately

ack
> * `disable`: 1m // keep charging for 1min at min current even if grid import

I would propose around 5 min, but this is pure gut feeling 
> 
> `thresholds:`
> 
> * `enable`: 0 // start charging if grid export  exceeds this value

Isn't this way too low? I also think this should default to min charge power (or min A * # of phases * 230 V)
> * `disable`: 1kw // stop charging if grid import exceeds this value

My suggestion: 0 as default (as it is now); temporary lows in export could be handled by disable delay;
and as I understand it, this could be achieved right now with (negative) residual power? 
> 
> Having the delays would also remove need for `guardtime`.

For me it's like @schenlap put it previously. Guardtime is a safety measure (while in automatic pv mode) to prevent "bad" configs damage contactors. 
> 
> Do these settings and default make sense for you?
> 
> /cc @schenlap @Caibuk @premultiply

schenlap, 2020-05-07:
- `delay: disable` 5 min would be also a good value for me. I tought it 1 min would be ok but with my small pv system i have to increase it.

Setting `enable` to min charging power (6A * voltage * phases) and `disable` to 0 may be problematic as there would be no hysterheresis. It may disable charging after 5 min if there was a short positive peak in pv power ore the car charger takes a little bit more than set value. I do not know how accurate the car charger are. 
`delay: enable` 1min and a `disable threhold` of about 200W allowed import power may prevent this problem.

andig, 2020-05-07:
Ich schalte mal auf Deutsch um- ich glaube das passt für alle Beteiligten.

Konfiguration neu sortiert:

```yaml
enable: 
  delay: 1m
  threshold: 200 // W *additional* export power on top of minimum charge power

disable:
  delay: 5m
  threshold: 200 // W import power
```

Wenn ich Euch richtig verstehe, dann muss threshold für die Dauer von delay erfüllt sein. Ist das nicht der Fall wird der Timer zurück gesetzt.

Bzgl. guardduration: ist damit m.E. obsolet da es jetzt bei now/off eh schon umgangen wird.

premultiply, 2020-05-07:
Richtig.
Mit dieser Berechnungsgrundlage kann man den **Enable-Threshold** theoretisch auch auf 0 setzen (PV Erzeugung - Hausverbrauch - Prognostizierter max. Ladestrom), da das Fahrzeug MAXIMAL so viel ziehen darf. Bei tatsächlicher Abweichung muss man dann hier nachjustieren.

Bei Disable-Threshold könnte man überlegen diesen Wert zum besseren Verständnis vielleicht negativ zu machen. Also bei bis zu (-200) = 200 Watt Netzbezug wird der Timer noch nicht gestartet. Darüber dauerhaft hinaus wird nach Delay der Ladevorgang gestoppt. Fällt der Bezug unter diesen Wert wird der Timer wieder gestoppt.

andig, 2020-05-07:
Andersrum. Üblicherweise Import=positiv, Export=negativ, wenn müsste mal also den enable threshold negativ machen. Das fand ich allerdings verwirrend weil die Ladeleistung ja üblicherweise als positiv angesehen wird und ich gedanklich immer anfange die auszusummieren :O. Die Formulierung ist immer noch nicht glücklich :/

andig, 2020-05-07:
Import/Export konsistent:

```yaml
# site power: grid + battery
#   positive: import
#   negative: export

enable: 
  delay: 1m
  threshold: -200 # W remaining site power if charger where to start

disable:
  delay: 5m
  threshold: 200 # W maximum site power before charger stops
```

premultiply, 2020-05-07:
:/ dann vielleicht doch eher so wie bei OpenWB direkt den Überschuss als Threshold angeben?
Dann gibt man enable 1380W an bzw. disable -200W an. Dann passt das auch. Ich finde das **wesentlich** verständlicher.

andig, 2020-05-07:
Ich sehs auch grad- nur hab ich die Vorzeichen physikalisch korrekt, aber schwer verständlich. Letzter Versuch, ganz ohne Vorzeichen:

```yaml
enable: 
  delay: 1m
  threshold: 1380 # W minimum export

disable:
  delay: 5m
  threshold: 200 # W maximum import
```

schenlap, 2020-05-07:
wieso mind. 1m? nach dem enable läuft es mind. 5 minuten (`disable delay`) weiter, oder?

Caibuk, 2020-05-07:
>... Wäre es nicht doch schöner die Aktivierung nur am Netzpunkt, aber nicht am Lader auszurichten, also:
> 
> ```yaml
> enable: 
>   delay: 1m
>   threshold: -1000 # W minimum site power for charger to start
> ```

👍, damit wäre es auch analog zum disable passend.
Und vielleicht ist man auch besser vorbereitet, wenn man später "verschieden-phasige" Autos an einem Loadpoint laden/unterstützen will...

premultiply, 2020-05-07:
Vorzeichenlos ist Mist denn wenn mann gezielt immer einen Restüberschuss haben möchte macht keinen Sinn hier dann Negativwerte einzutragen wenn man vorher für Überschuss positive Werte verwendet hat.
Man kann für Default bei Disable auch Threshold 0 eintragen :)

Oder wenn die PV-Anlage zu klein ist um das Auto alleine damit überhaupt zu laden.

Betrachtet wird hier immer der Netzübergabepunkt und da haben Im- und Export nun mal unterschiedliche Vorzeichen. Wenn man das in einem Fall dreht ist die Verwirrung perfekt.

andig, 2020-05-08:
```yaml
enable: 
  delay: 1m
  threshold: -1380 # W

disable:
  delay: 5m
  threshold: 200 # W maximum import
**oder**
  threshold: 200 # W
```

> Vorzeichenlos ist Mist denn wenn man gezielt immer einen Restüberschuss haben möchte macht es keinen Sinn hier dann plötzlich Negativwerte einzutragen wenn man vorher für Überschuss positive Werte verwendet hat.

Mißverständnis- ich würde beide Werte "positiv" als import/export differenzieren. Aber auch mit Vorzeichen wäre möglich, siehe Beispiel oben

> Man kann für Default bei Disable auch Threshold 0 eintragen. :)

Für disable ist das korrekt: default sollte 0 sein. Für enable ist das allerdings Mist bzw. bedarf einer Sonderbehandlung: default 0 für enable *muss* ignoriert werden, anderenfalls würde bei 0W Einspeisung schon zugeschaltet, was ja Schwachsinn wäre ;) Ist mir beim Schreiben der Testcases aufgefallen...

Caibuk, 2020-05-08:
Könnte man für enable nicht die minimale Ladeleistung als (dynamischen) default-Wert verwenden (so wie es momentan wohl ist)? Nur wenn man etwas angibt wird es "hart" gesetzt.

andig, 2020-05-11:
@all: der PR wäre dank tatkräftiger Hilfe von @premultiply bereit zum testen

schenlap, 2020-05-12:
In der Simulation getestet und funktioniert! Anmerkung: disable delay kann überschritten werden weil zuerst der Strom runtergeregelt wird.

closed 2020-05-12 by andig

andig, 2020-05-13:
Hier noch ein Beispiel in dem kurze Störungen nicht zur Abschaltung führen:

[image]

PR #112 (2020-05-12): Improve PV on/off handling
Fix #95