evcc-io/evcc #98: Exclude "Sofortladen" and "Stop" from guardduration

fiolux, 2020-05-04:
Current state:
The guardduration ticks down since the last change to the WallBox.
If you click "Sofortladen" or "Stop" you have to wait for the change until the guardduration has expired.

Expected behavior:
If "Sofortladen" or "Stop" are clicked the command should be send to the WallBox immediately.

andig, 2020-05-04:
I would make one expection: we‘re first setting minimum charge current (=6A) before switching on/off. Currently there would be interval time between that. Is that ok? If we switch immediately, we cannot be sure that the EV has accepted the min current already. Opinions?

premultiply, 2020-05-05:
Pre-setting the charge current to minimum (6A) should only be done if a vehicle is currently charging (or drawing more than 6A from charger).

Apart from that some vehicles like Renault ZOE are **sometimes** a bit picky if the allowed charge current is to low when they should start charging. Setting a higher current in standby state (10 or better 16A) and immediatly decreasing it after the vehicle really started charging/draw more the allowed power will fix that.

Caibuk, 2020-05-06:
During my first tests I was puzzled as there was no reaction of wallbox/car after switching to off ("Sofortladen" was started only one or two minutes before).
I support the proposal of the thread opener considering @andig 's concern.

Another thing I'm wondering about: when evcc is set to pv only (and there is not enough PV generation) and I connect my car, then charging will start immediately (and stop after guardtime). Why does it start to charge in the first place?

premultiply, 2020-05-06:
> Why does it start to charge in the first place?

This MAY ALSO be a problem of the EVSE controller inside your wallbox.
There are devices like NRGkick that can only "pause" a charge but reset itself when car gets connected. So the only way for EVCC and others is to detect the running charge and interrupt it if it should not charge now. This may take a few seconds.

andig, 2020-05-06:
> Why does it start to charge in the first place?

See #105.

andig, 2020-05-06:
> Exclude "Sofortladen" and "Stop" from guardduration

We can do this- see #108. However, if anybody uses the API, this would also apply there. If you have a controller program going crazy then that might kill your wallbox?

PR #108 (2020-05-06): Apply now and off modes immediately
Fix #98
closed 2020-05-06 by andig