evcc-io/evcc #105: Set enable/disable in PV mode even if car disconnected

andig, 2020-05-06:
from #98:

> Another thing I'm wondering about: when evcc is set to pv only (and there is not enough PV generation) and I connect my car, then charging will start immediately (and stop after guardtime).

andig, 2020-05-06:
Root cause ist that I set MinCurrent when not connected to prevent that the car will connect and contactor close at full current. I think the needed behaviour is:

in PV mode:
- when connected: apply PV regulation enable/disable/current
- when disconnected apply PV regulation enable/disable at minimum current

premultiply, 2020-05-06:
The contactor does never close or open on current under normal operating conditions.
The contactor is driven by the EVSE mostly under vehicle control. Only if the vehicle signals "charging" the contactor has to be closed. But it will NOT immediatly draw current instead it will run several tests on mains (phase voltages, ground and neutral resistance,... )and if everything is ok it will slowly ramp up the current. The same happens in the opposite way when PWM is turned into DC only on CP (stop charging by EVSE).

So this handling from EVCC side is a bit useless.

andig, 2020-05-06:
> So this handling from EVCC side is a bit useless.

I've heard different before. Since this is conflicting input, please let's split these changes into two separate updates.

PR #106 (2020-05-06): Ensure charger target status is always updated even while car disconn…
…ected

Fix #105
closed 2020-05-06 by andig