evcc-io/evcc #111: Tesla OAuth Error

S3b4st1an-git, 2020-05-07:
Trying to setup evcc in a docker container on a synology 918+ for Tesla Model3, Wallbe eco2s and Solaredge PV (installation due), data going through mqtt/Home Assistant.

Pasted the client key and id from pastebin link in wiki but still getting
"https://owner-api.teslamotors.com/oauth/token": x509: certificate signed by unknown authority"

Didi anyone get a connection to the tesla mothership?

andig, 2020-05-08:
Moin @S3b4st1an-git das Tesla API erfordert zusätzlich noch Email und Passwort, siehe https://github.com/andig/evcc/blob/master/evcc.dist.yaml#L168. Wenn das nicht hilft würde es sich lohnen bei der verwendeten Library anzufragen: https://github.com/jsgoecke/tesla.

Danke dass Du EVCC ausprobierst!

S3b4st1an-git, 2020-05-08:
Moin.
Email und Passwd hatte ich eingetragen, das kenne ich schon von anderen Anwendungen, die die API von Tesla nutzen (logger). Nur die Client id war mir neu.
Werde mal weiter wühlen.
Vielen Dank für EVCC schonmal, sehr strukturiertes Projekt!

Viele Grüße 
Sebastian

andig, 2020-05-08:
Hier ist der Code- an welcher Stelle steigst Du aus- ist das eine der "cannot create Tesla Meldungen"?:

	cc := struct {
		Title                  string
		Capacity               int64
		ClientID, ClientSecret string
		Email, Password        string
		VIN                    string
		Cache                  time.Duration
	}{}
	util.DecodeOther(log, other, &cc)

	client, err := tesla.NewClient(&tesla.Auth{
		ClientID:     cc.ClientID,
		ClientSecret: cc.ClientSecret,
		Email:        cc.Email,
		Password:     cc.Password,
	})
	if err != nil {
		log.FATAL.Fatalf("cannot create tesla: %v", err)
	}

	vehicles, err := client.Vehicles()
	if err != nil {
		log.FATAL.Fatalf("cannot create tesla: %v", err)
	}

	v := &Tesla{
		embed: &embed{cc.Title, cc.Capacity},
	}

	for _, vehicle := range vehicles {
		if vehicle.Vin == cc.VIN {
			v.vehicle = vehicle.Vehicle
		}
	}

	if v.vehicle == nil {
		log.FATAL.Fatal("cannot create tesla: vin not found")
	}

andig, 2020-05-08:
Falls Du Hilfe beim Testen brauchst schick mir gerne PM (cpuidle@gmx.de).

closed 2020-05-08 by andig