evcc-io/evcc #117: Starting evcc with Mode = Off does not disable charger

schenlap, 2020-05-10:
Grid power = 1000W import, Mode is default Off. Charger is in **enabled state** during evcc start.
```
[main] INFO 2020/05/10 08:56:40 evcc dev (HEAD)
[main] INFO 2020/05/10 08:56:40 using config file /xxx/evcc/evcc.yaml
[main] INFO 2020/05/10 08:56:40 listening at 0.0.0.0:7070
[load] INFO 2020/05/10 08:56:40 main loadpoint config: vehicle — grid ✓ pv — battery — charge —
[load] INFO 2020/05/10 08:56:40 main charger config: power — energy — timer —
[load] INFO 2020/05/10 08:56:40 main charge mode: off
-> correct
[load] INFO 2020/05/10 08:56:40 main charger enabled
-> this is my init state.
[load] DEBUG 2020/05/10 08:56:40 main set charge current: 6A
-> charging will beginn / continue
[load] DEBUG 2020/05/10 08:56:40 > main
[load] DEBUG 2020/05/10 08:56:40 main charger status: B
[load] DEBUG 2020/05/10 08:56:40 main grid power: 1000.0W
[load] DEBUG 2020/05/10 08:56:40 main charge power: 0.0W
[load] DEBUG 2020/05/10 08:56:40 main charger disable - contactor delay 19s
[load] DEBUG 2020/05/10 08:56:40 < main
[load] DEBUG 2020/05/10 08:56:40 > main
[load] DEBUG 2020/05/10 08:56:40 main charger status: B
[load] DEBUG 2020/05/10 08:56:40 main grid power: 1000.0W
[load] DEBUG 2020/05/10 08:56:40 main charge power: 0.0W
[load] DEBUG 2020/05/10 08:56:40 main charger disable - contactor delay 19s
[load] DEBUG 2020/05/10 08:56:40 < main
[load] DEBUG 2020/05/10 08:56:50 > main
[load] DEBUG 2020/05/10 08:56:50 main charger status: B
[load] DEBUG 2020/05/10 08:56:50 main grid power: 1000.0W
[load] DEBUG 2020/05/10 08:56:50 main charge power: 0.0W
[load] DEBUG 2020/05/10 08:56:50 main charger disable - contactor delay 19s
[load] DEBUG 2020/05/10 08:56:50 < main
[load] DEBUG 2020/05/10 08:57:00 > main
[load] DEBUG 2020/05/10 08:57:00 main charger status: C
[load] INFO 2020/05/10 08:57:00 main start charging ->
[load] DEBUG 2020/05/10 08:57:00 main grid power: 1000.0W
[load] DEBUG 2020/05/10 08:57:00 main charge power: 1380.0W
[load] DEBUG 2020/05/10 08:57:00 main charger disable - contactor delay 19s
[load] DEBUG 2020/05/10 08:57:00 < main
[load] DEBUG 2020/05/10 08:57:10 > main
[load] DEBUG 2020/05/10 08:57:10 main charger status: C
[load] DEBUG 2020/05/10 08:57:10 main grid power: 1000.0W
[load] DEBUG 2020/05/10 08:57:10 main charge power: 1380.0W
[load] DEBUG 2020/05/10 08:57:10 main charger disable - contactor delay 19s
[load] DEBUG 2020/05/10 08:57:10 < main
[load] DEBUG 2020/05/10 08:57:20 > main
[load] DEBUG 2020/05/10 08:57:20 main charger status: C
[load] DEBUG 2020/05/10 08:57:20 main grid power: 1000.0W
[load] DEBUG 2020/05/10 08:57:20 main charge power: 1380.0W
[load] DEBUG 2020/05/10 08:57:20 main charger disable - contactor delay 19s
[load] DEBUG 2020/05/10 08:57:20 < main
```
I think enabled state of charger can occur in real world (e.g. restart of evcc during charging) and evcc should be able to handle this situation. Contactor delay in not decreased so it will never stop charger.

If I have misunderstood the charger state please close this issue.

premultiply, 2020-05-19:
Vorsicht: Der Mode heisst wohl inzwischen mindestens in der config-Datei "stop" nicht "off". Bei "off" kommt nur Mode "0". Bin ich gerade drüber gestolpert...

andig, 2020-05-19:
Ich versteh‘s nicht. Der Mode heisst off. Wo kommt Mode 0?

premultiply, 2020-05-19:
Wenn man in der evcc.yaml bei mode off einträgt und EVCC startet meldet er mode 0.
Wenn man dort stattdessen stop reinschreibt steht auch im Log mode stop.

andig, 2020-05-20:
> Wenn man in der evcc.yaml bei mode off einträgt und EVCC startet meldet er mode 0. Wenn man dort stattdessen stop reinschreibt steht auch im Log mode stop.

OK, neues Issue, wird gelöst.

Ist unabhängig davon dieser PR ein Schritt in die richtige Richtung für das beschriebene Problem?

schenlap, 2020-05-20:
getestet mit 0490db10fa23c520367488f22e279a3fc4bcb380 (current master)
funktioniert jetzt jedenfalls (auch mit mode=off). Danke!

andig, 2020-07-03:
Sollte behoben sein.

PR #199 (2020-07-16): Implement multiple loadpoints and new UI
Replace #181, fix #117.

This is a breaking change:
- loadpoint config is split into `site` and `loadpoints`
- grid/ pv and battery meters are moved to `site`
- push message templates are updated- more values available and missing values shown in logfile
- now requires InfluxDB 1.8/2.0 - InfluxDB 1.6/1.7 are no longer supported

Beispiele für geänderte Konfiguration sind in `evcc.dist.yaml` zu finden: https://github.com/andig/evcc/blob/master/evcc.dist.yaml
closed 2020-07-03 by andig