evcc-io/evcc #124: Keep settings and states in sync between EVSE and EVCC

premultiply, 2020-05-11:
Major settings like charge current and charge enabled should be kept in sync between internal EVCC and EVSE.

Due to any connection problem (shutdown or reboot of EVSE, bad wireless connection, EMP, manual settings on EVSE, internal EVSE logic etc.) settings can be desynced between EVCCs will and what the charger really does.
So EVCC should read the major settings from EVSE, compare and update them if they differ on a regular base where possible. Not all types of EVSE allow for reading the internal settings.

Appart from that it should make sure that the connection to EVSE is alive by implementing something like "heartbeat" or "ping" to monitor the connection #122

andig, 2020-05-11:
> Due to any connection problem (shutdown or reboot of EVSE, bad wireless connection, EMP, manual settings on EVSE, internal EVSE logic etc.) settings can be desynced between EVCCs will and what the charger really does.

Thats only partially true. EVCC keeps track if settings were successfully written to EVSE. However, EVCC does not currently re-read EVSE values from EVSE. This makes is susceptible to the following influences:
- manual changes by users
- new defaults upon EVSE reboot

One approach could be to re-write EVCC state to EVSE at a regular (configurable) interval. Even for reliable devices that would take care of forceful reboots or user interaction and prevent potential deadlocks in EVCC operations.

#122 is a different topic and will be addressed by pushing *all* warning or error messages to the UI.

premultiply, 2020-05-11:
Always Read-Compare-Write settings may also reduce EEPROM and Flash write cycles on some embedded devices.
So only write if values on device differ from EVCCs internal value.

andig, 2020-05-12:
> So only write if values on device differ from EVCCs internal value.

That's currently implemented- EVCC keeps state and only writes when it wants to request a state change.

premultiply, 2020-05-12:
Yes, but this is not what is meant here.

TheRealKerni, 2020-05-12:
How about reading the current value in each cycle and only write if it differs from the target value?

andig, 2020-05-12:
> How about reading the current value in each cycle and only write if it differs from the target value?

Sure, but would require changing the entire charger interface. Not highest priority.

PR #172 (2020-05-19): Synchronize enabled state with charger if state doesn't match
Fix #124 (partially), #169 and #157.

Der NRGKick scheint bzgl. Start des Ladevorgangs ein Eigenleben zu haben, die EVSE braucht eine Extraeinladung wenn sie schonmal geladen hat. Dieser PR sollte beide Probleme lösen und würde gleichzeitig auch bei einem Restart des Chargers aktiv werden.
closed 2020-05-19 by andig