evcc-io/evcc #145: Enhanced Push-Messaging

premultiply, 2020-05-13:
Ich würde vorschlagen die Push-Benachrichtigungen (Telegram, Pushover, SMS, E-Mail, ...) mit weiteren Metadaten zu füttern oder zumindest um weitere mögliche Datenfelder zu erweitern.

Hier mal eine lose Ideensammlung unabhängig vom aktuellen Implementierungsstand und der Umsetzungsschwierigkeit:

- [ ] Start-SOC
- [ ] Aktueller SOC
- [ ] Aktuelle Ladeleistung
- [ ] Durchschnittliche Ladeleistung
- [ ] Maximale Ladeleistung (Fahrzeug)
- [ ] Maximale freigegebene Ladeleistung (EVCC)
- [ ] Geladene Energie
- [ ] Energiekosten
- [ ] Vergangene Ladedauer
- [ ] Prognostizierte (Rest)Ladedauer
- [ ] Prognostiziertes Ladeende (Uhrzeit)
- [ ] Aktuelle Uhrzeit
- [ ] Ladekurve als Grafik
- [ ] Netzbezugsanteil

Darüber hinaus sollten zu folgenden Events entsprechde Mitteilungen verschickt werden können:

- [ ] Fahrzeug wurde verbunden
- [ ] Fahrzeug hat Ladung begonnen
- [ ] Fahrzeug hat Ladung beendet
- [ ] Fahrzeug wurde getrennt
- [ ] Ladung wurde freigegeben (durch EVCC während Fahrzeug verbunden)
- [ ] Ladung wurde gesperrt/pausiert (durch EVCC während Fahrzeug verbunden)
- [ ] EVCC-Fehlermeldung
- [ ] SOC-Wert geändert

andig, 2020-05-14:
>  Fahrzeug hat Ladung begonnen
>  Fahrzeug hat Ladung beendet

Die gibt es doch schon?

> SOC-Wert geändert

Mehrwert? Das dürfte zu ziemlich vielen Nachrichten führen.

premultiply, 2020-05-14:
Ich schrieb doch unabhängig vom aktuellen Implementierungsstand :)

Das sollen ja auch nur MÖGLICHE Triggerpunkte sein. Ist ja eine Ideensammlung. Manche Ideen sind halt gut, andere vielleicht nicht so gut :D

Der SOC ändert sich bei Renault z. B. nur alle 30 Minuten.

andig, 2020-05-14:
> Ich schrieb doch unabhängig vom aktuellen Implementierungsstand :)

Dann lass uns aber keine Sachen aufschreiben die es eh schon gibt- sonst verstehe ich es nicht ;)

premultiply, 2020-05-14:
War ja nur der Vollständigkeit halber.
Hab dir oben auch schon mal dafür zwei Haken gesetzt ;)

premultiply, 2020-07-20:
Schade. Hab mich eben gerade nochmal etwas darüber geärgert dass der Inhalt der Meldungen und die Trigger so spärlich sind.
So kann man z. B. nur raten warum eine Ladevorgang beendet wurde und man sieht auch nicht wieviel Energie insgesamt geladen wurde.

andig, 2020-07-20:
Das stimmt nur bedingt. Du kannst schon jetzt alle vorhandenen Variablen nutzen. Alle definierten Namen findest Du im Logfile, Kategorie „cache“- die kannst Du 1:1 weiter verwenden!

andig, 2020-07-21:
Sieht doch gut aus. Mir würden noch Gesamtladezeit+Energie fehlen, es ist aber nicht ganz klar wie die zum Event bestimmt werden sollen. Insbesondere: woher weiß ich, dass "gesamt" abgeschlossen ist? Eine Möglichkeit wäre soc=100, tw. habe ich aber Autos gesehen die nur auf 98% kommen. Wenn im PV Mode abgestöpselt wird bevor das Auto voll ist würde das hinsichtlich "gesamt" gleich mehrere Events auslösen was auch nicht ideal erscheint. Gerne Detaildiskussion bei Slack.

premultiply, 2020-07-21:
"Gesamt" bezieht sich auf die Periode zwischen an- und abstecken. Dazwischen wird alles im Status C summiert. Ob das Fahrzeug voll ist oder nicht spielt dabei eigentlich keine Rolle.
Ich bin in der Slack-Bar.

andig, 2020-07-22:
Ich glaube die wichtigen Dinge sind drin, Praxistest tbd. Ich schließe das Thema mal fürs Erste.

closed 2020-07-22 by andig