evcc-io/evcc #147: Nutzung von Discovergy Zählern

ttjaden, 2020-05-14:
Hallo! Ich wollte EVCC ab nächster Woche mal testen. 

**Setting:**
-- go-e Charger 
-- Renault ZOE (Modell 2020)
-- PV-Eigenversorgungsanlage mit Smartmetern von Discovergy
-- Synology-NAS vorhanden

**Fragen:**
-- Kann EVCC schon mit der Discovergy API umgehen (für die Abfrage von Einspeisung / Bezug). Hier gibt bei bei OpenWB bereits ein Modul
-- Empfielst du ein Docker auf der Synology-NAS oder besser eine andere Umsetzung?

andig, 2020-05-14:
Danke für Deine Frage! Deine Komponenten sollten perfekt passen:

> Kann EVCC schon mit der Discovergy API umgehen (für die Abfrage von Einspeisung / Bezug). 

Ja. Du kannst einen "default" Zähler mit "http" Plugin konfigurieren, siehe README. Die Konfiguration kannst Du dann auch einzeln mittels

    evcc -c deine-konfig.yaml meter

testen. Detailfragen gerne hier, wäre Klasse wenn Du das Ergebnis als Beispiel im Wiki ablegen könntest.

> Empfielst du ein Docker auf der Synology-NAS oder besser eine andere Umsetzung?

Was Dir am besten passt- alles ist möglich!

ttjaden, 2020-05-14:
Ich bin fast am Ziel. Leider wusste ich nicht, wie ich es über "type: html" machen sollte, da man nicht alle Infos in der URL einbauen kann. Daher musste ich auf "type: script" zurückgreifen.
```
# meters
meters:
- name: grid
  type: default
  power: # power reading
    type: script # use script
    cmd: /bin/sh -c "curl --connect-timeout 3 -s -u username:password "https://api.discovergy.com/public/v1/last_reading?meterId=id-of-my-meter" | jq .values.power"
    timeout: 2s 
```
Der Test mittels evcc meter ergibt:
```
./evcc meter
Power: -4446330W
```
Ich muss den Output der Abfrage noch durch 1000 teilen. Jedoch geht das leider nicht über die Möglichkeit "scale", welche man bei "type: http" hat. Weißt du hier rat?

ttjaden, 2020-05-14:
Weitere Sache, die ich noch nicht verstanden habe:
Wenn ich folgenden Bereich in der Konfiguration aktiviere...
```
  enable: # pv mode enable behavior
    delay: 1m # threshold must be exceeded for this long
    threshold: 0 # minimum export power (W). If zero, export must exceeds minimum charge power to enable
  disable: # pv mode disable behavior
    delay: 5m # threshold must be exceeded for this long
    threshold: 200 # maximum import power (W)
```
Startet evcc nicht mehr mit der Fehlermeldung
```
[main] FATAL 2020/05/14 16:35:38 config: failed parsing config file /home/tjarko/Downloads/Software-Nutzung/evcc_0.7_linux_64-bit/evcc.yaml: 1 error(s) decoding:

* 'LoadPoints[0]' has invalid keys: disable, enable
```

ttjaden, 2020-05-14:
> Ich muss den Output der Abfrage noch durch 1000 teilen. Jedoch geht das leider nicht über die Möglichkeit "scale", welche man bei "type: http" hat. Weißt du hier rat?

Habe es nun in der shell umgesetzt. Ich vermute eher, dass das nicht so elegant ist, aber es funktioniert:
```
    cmd: /bin/sh -c "output=$(curl --connect-timeout 3 -s -u t.tjaden@gmail.com:STk9sQgA "https://api.discovergy.com/public/v1/last_reading?meterId=6e3a260a875b4cbcb20c1346c7409a02") ; power=$(echo $output | jq .values.power) ; watt=$(( power / 1000 )) ; echo $watt"
```
[image]

andig, 2020-05-14:
> Habe es nun in der shell umgesetzt. Ich vermute eher, dass das nicht so elegant ist, aber es funktioniert:

Das Plugin heisst http, nicht html. Was fehlte Dir denn zur Umsetzung? Shell ist nicht notwendig...

PS.: Deinen User habe ich mal rauseditiert...

andig, 2020-05-14:
> Wenn ich folgenden Bereich in der Konfiguration aktiviere...

Du hast 0.7 aber eine Konfiguration aus master. Dafür müsstest Du dann den aktuellen Entwicklungsstand nehmen.

> Shell ist nicht notwendig...

OK, ein Baustein (#149) fehlte noch. Jetzt kannst Du Folgendes machen:

```yaml
- name: discovergy
  type: default
  power:
    type: http
    auth:
      type: basic
      user: <xxx>
      password: <yyy>
    uri: https://api.discovergy.com/public/v1/last_reading?meterId=<id>
    jq: .values.power
    scale: 0.001
```

Wenn Du helfen möchtest könntest Du das Beispiel gerne ins Wiki übertragen?

ttjaden, 2020-05-14:
Danke für den Support! Nun klappt's. Den Wiki-Eintrag werde ich dann gerne einpflegen!

andig, 2020-05-14:
Dann mache ich hier mal zu- war mir eine Freude!

closed 2020-05-14 by andig