evcc-io/evcc #150: Netzschutzmaßnahmen und Steuerung durch Netzbetreiber

premultiply, 2020-05-15:
Ähnlich wie openWB könnten ggf. auch EVCC ein paar Basisnetz(schutz)funktionen ganz gut stehen.

- [ ] Frequenzabhängiger Lastabwurf (d.h. Unterbrechnung) des Ladevorgangs bei Netzüberlastung (< 49,200 Hz)

- [ ] Schieflastbegrenzung auf <= 20A  (max. Phasenstromdifferenz) am Netzübergabepunkt

Nach §14a EnWG:
- [ ] Abregelung durch VNB-Steuergerät (temporäre Limitierung auf x A)
- [ ] Freigabe/Sperrung durch VNB-Steuergerät ("Wärmepumpenstrom", abschaltbare Last)

Vlg. z. B. https://assets.ctfassets.net/xytfb1vrn7of/6TIfm0d9BAtSlWvTSY3zis/4dd36bd0311ac927d19c00b5c5490bda/2020_03_NetzeBW_TMA_Netzdienliche_Steuerung.pdf

andig, 2020-05-15:
> Frequenzabhängiger Lastabwurf (d.h. Unterbrechnung) des Ladevorgangs bei Netzüberlastung (< 49,200 Hz)

Bräuchte eine Erfassung der Frequenz? Letzteres wäre einfach machbar! 

>  Schieflastbegrenzung auf <= 20A (max. Phasenstromdifferenz) am Netzübergabepunkt

Das würde ich unter Support für mehrere Ladepunkte und entsprechendem Lastmanagement mit aufgehen.

> Abregelung durch VNB-Steuergerät (temporäre Limitierung auf x A)

Abregelung inge heute schon- deaktivierung auf `off` mittels API. Ggf. braucht es. da noch eine Art Verriegelung? Begrenzung des Stromes ist gut machbar, API fehlt noch.

> Freigabe/Sperrung durch VNB-Steuergerät ("Wärmepumpenstrom", abschaltbare Last)

Das fiele unter "HEMS"- im Moment haben wir nur die EVs als Lasten in der Betrachtung.

Generell wäre meine Frage: wer braucht die Features und möchte sie einsetzen? Welcher zusätzliche Nutzen erschließt sich?

premultiply, 2020-05-15:
Frequenzmessung kommt von einem (beliebigen) Zähler. Entweder direkt zentral vom Gridmeter oder vom Zähler der Wallbox. Die Frequenz ist überall gleich so lange Spannung anliegt.

Schieflastbegrenzung hat nicht unbedingt was mit mehreren Ladepunkten zu tun. Hier geht es rein um die Betrachtung am HÜP. Auch durch größere andere einphasige Verbraucher (Herd, Durchlauferhitzer, ...) oder Erzeuger (PV-WR, ...) im Haus kann die Schieftlastgrenze überschritten werden. Ist eine Schönheitssache.

Für die Steuerung durch den VNB braucht man grundsätzlich und letztendlich immer irgenwo einen Schaltkontakteingang und keine IT-Schnittstelle (API). Auch wenn es intern darüber gelöst ist.

andig, 2020-05-15:
> Für die Steuerung durch den VNB braucht man grundsätzlich und letztendlich immer irgenwo einen Schaltkontakteingang und keine IT-Schnittstelle (API, ...) - auch wenn es intern darüber gelöst ist.

Meinst Du einen Kontakt im Signaleingang oder den in der Wallbox? Fall Ersteres wäre das damit off-topic da nicht durch Software zu lösen?

premultiply, 2020-05-15:
Der VNB stellt ja sowas tolles wie eine Zeitschaltuhr oder einen Rundsteuerempfänger bereit. Der schließt oder öffnet einen Kontakt.

Da sich EVCC aber nicht um solche Hardware kümmert braucht es letztendlich nur ein API-Equivalent dazu so dass es mit einem geeigneten Hardwaremodul angesteuert werden kann.
Das könnte dann wiederum am Ende ein GPIO eines Raspberrys oder sowas in der Art sein.

tmeidinger, 2020-06-18:
Könnte man den Punkt erweitern um Ladefreigabe- oder Ladesperrzeiten (ggf. pro Wochentag) für das Laden mit Netzbezug?
Beispiel:
Das Auto wird um 17:00 angesteckt, der PV-Überschuss geht ins Auto. Die Sonne geht unter, das Laden wird angehalten. Ab z.B. 22:00 gibt es günstigen Nachtstrom und das Laden wird wieder gestartet.

andig, 2020-06-18:
Das wäre eine Art „automatischer Modus nach Uhrzeit“, Du müsstest damit ja auch auf Min+PV umstellen. Die Frage bei all diesen Anforderungen ist weniger ob das ginge, sondern wie das UI und der Algorithmus dahinter aussehen sollen?

Caibuk, 2020-06-18:
Das wäre was! Habe auch das Optimierungsproblem PV / Akku / Zweitarif...
Müsste man sich eine Strategie und flexible Umsetzung ausdenken.

premultiply, 2020-08-17:
Der fiese Bot... ;)

andig, 2020-12-16:
OCPP ist mittlerweile rudimentär umgesetzt, harrt aber einer echten Anwendung da kein bekannter VNB heute steuert. Zu den restlichen Features scheint der echte Bedarf zu fehlen- ich mache daher erstmal zu.

christoph-dunker, 2021-05-28:
> Der VNB stellt ja sowas tolles wie eine Zeitschaltuhr oder einen Rundsteuerempfänger bereit. Der schließt oder öffnet einen Kontakt.
>
> Da sich EVCC aber nicht um solche Hardware kümmert braucht es [...]

Mein Netzbetreiber EWE Netz bietet hierfür einen Rundsteuerempfänger an, der über einen potentialfreien Kontakt entweder ein Schütz schaltet oder ein EVU-Signal einer Steuereinheit der Wallbox(en). Ich gedenke über die gpio  des raspberry zu gehen.
Wichtig ist, ist dass komplett sämtliche Ladevorgänge blockiert werden, solange bis das Rundsteuersignal wieder anliegt.

Ich bin Softwarearchitekt und verdiene mein Geld in der Lagerautomatisierung. Auch wenn ich eigentlich im Bereich Java heimisch bin, würde ich mir zutrauen an diesem Issue mitzuarbeiten. Wir sollten das Abschalten über Netzbetreiber dann aber ggf absplitten in ein separates issue und dort "sauber" abarbeiten. 

Ich bräuchte Hinweise, wie ich starten könnte und über welchen Weg eine solche Funktionalität möglich wäre. "Doof" gedacht bräuchte ich ein zyklisches Abfragen eines definierten gpio paares und würde bei Ausbleiben einer 1 / Anliegen einer 0 den maximalen verfügbaren Ladestrom von 11kw auf 0kw begrenzen (16A -> 0A). Ggf. kann man sich auch als Listener an den gpio hängen und bei steigender / fallender Flanke benachrichtigt werden!?

Als Erweiterung könnte ich mir auch noch vorstellen, das HT/NT (Haupttarif / Nebentarif) über gpio abzufragen, hierfür kann auch ein Relais geschaltet werden. Damit kämen wir vermutlich in einen Lösungsbereich wie OCPP / HEMS.

Wir können dann auch gerne wieder auf Englisch wechseln ;)

premultiply, 2021-05-28:
Ich denke auch dass man sowas in der Art eines „Simple HEMS“ implementieren könnte.
Die SEMP-Implementierung sollte passende Ansatzpunkte geben. Dort gibt es aber bisher nur Freigabe ja/nein. Eine Leistungsbegrenzung ist dort leider noch nicht implementiert.

andig, 2021-05-29:
Genau. Schau Dir mal die Ocpp und Semp Implementierungen an. Em Ende kommt da ein softer oder harter Steuerungsimpuls für den Ladepunkt raus. Soft wird nur im PV Modus berücksichtigt, hart greift immer (Netzschaltmassnahme), regelt aktuell allerdings auf 0.

andig, 2021-05-29:
Echtes Lastmanagement mit Powershaping für alle Modi ist gerade in der Entstehung…

closed 2021-06-25 by stale[bot]

michirosenbuehl, 2023-03-10:
Wie ist denn hier der aktuelle Stand?
Wird es diese Funktionen bald geben?

Danke für eine kurze Info!

andig, 2023-03-10:
@michirosenbuehl nix geplant. Wofür brauchst Du die denn? Wie soll die Integration mit Deinem Netzbetreiber erfolgen?

michirosenbuehl, 2023-03-10:
@andig es geht doch darum, dass durch Überlastung des Hausanschlusses (z.B. Strom größer Hausanschlusssicherungen) keine Sicherung fliegt. Es geht also um ein Loadbalancing.

Am ehesten passiert das in einem Mehrfamilienhaus.

Beispiel:
 - sechs Wohneinheiten
 - Wallboxladeinfrastruktur mit vier Wallboxen à 11kW (maximalstrom ca. 64A)
 - PV-Anlage optional

Laden jetzt alle ihr Auto abends mit Netzstrom und auch nur ein Teil (3 von 6 Wohneinheiten) bezieht Haushaltsüblichen Strom - z.B. Kochen / Wsserkocher kommt man sehr schnell an die Grenze des Hausanschlusses, wenn man die Phasen einzeln betrachtet.
Es wäre also sehr hilfreich für solche Fälle, dass die Wallboxen kurzfristig zurückgeregelt werden.

Von der Anwendung ist es vergleichbar mit der Überschussladung - jedoch mit fester Grenze.

Der Netzbetreiber hat mit der Lösung nichts zu tun, außer dass er ggf. dem Elektriker ein entsprechendes Loadbalancing vorschreibt.

andig, 2023-03-10:
Dann willst du https://github.com/evcc-io/evcc/pull/3453

koaschten, 2024-02-11:
Mit der Novelle des EnWG §14a gibt es ja jetzt seit dem 01.01.2024 die "Dimmung". Also die Regelung auf 4,3kW, wenn man denn ab dem 1.1. eine Wärmepumpe oder Wallbox installiert, oder einfach mitspielen will. Pflicht ist das ja für Bestandsanlagen erst ab 2029.

Ist die Idee der Möglichkeit das Steuersignal des VNB in EVCC zu integrieren noch "da"? Und soll das immer via evcc API von extern kommen z.b. via Modbus Befehl von einer SPS am Schaltkontakt des Rundsteuer Empfängers oder des iMSys.

Die BNetzA hat da einiges zu geschrieben unter: https://www.bundesnetzagentur.de/DE/Fachthemen/ElektrizitaetundGas/Aktuelles_enwg/14a/start.html

wie sich zum Beispiel Go-E das vorstellt haben sie unter https://go-e.com/fileadmin/Support/Anleitungen/Deutsch/information-zur-paragraph-14aenwg-konformiaet-und-installation.pdf beschrieben.

Steve und Julian haben dazu heute gerade ein Video gemacht.
https://youtu.be/H61rf5qMwok

Interessant ist das halt in ein paar Szenarien.
- Das gilt pro angemeldeter Verbrauchs-Einrichtung (VE) -> wenn man zwei anmeldet darf man auch 2x 4,2kW aus dem Netz ziehen, auch wenn nur eine VE es verbraucht.
- es darf mit einer PV supplementiert werden, also 4,2kW + PV Erzeugung
- der "normale" Hausverbrauch kommt noch oben drauf. Und wird nicht limitiert.

andig, 2024-02-12:
Go-E erzählt nur die Hälfte:

> • Potentialfreier Eingang an Steuerbox oder Rundsteuerempfänger des Netzbetreibers
• Steuerung über Modbus TCP mittels SPS
• Schützschaltung mit Verbindung zu Steuerbox oder Rundsteuerempfänger des Netzbetreibers
• OCPP Schnittstelle des Netzbetreibers (z. B. Stromnetz Hamburg)

Die kommenten Steuerboxen werden vstl. entweder Relaisaufgang oder EEbus haben. Wir werden das also per EEbus umsetzen wenn und falls die entsprechenden Use Cases in der Library vorhanden sind.

christoph-dunker, 2024-02-13:
Bei unserem Netzbetreiber EWE Netz werden Wallboxen seit 01.01.2024 nur noch genehmigt, wenn ein ein Kabel zum Zähler gelegt wird, an dem die Steuerbox angeschlossen werden kann. Wir sind froh, dass wir bei unseren kunden go-e verbaut haben. Ich frage mich wann der erste Hersteller das EMS implementiert hat, dass man seinen PV-Strom trotzdem nutzen kann.

 

Zuhause nutze ich natürlich evcc. Aber so einfach, dass man es wie den charge controller auf die Hutschiene nagelt und per App einstellt ist es leider noch nicht 😉

 

Grüße,

 

Christoph

premultiply, 2024-02-13:
Reine Marketinggeschwätz, da auch go-e keine Spezifikation hat - da es noch keine gibt.
Ob ausgerechnet go-e mit seiner WiFi-only-Strategie da irgendeine Chance haben wird ist zudem mehr als fraglich.

Sowohl jegliche Forderung eines Netzbetreibers als auch irgendwelche Versprechungen von egal welchem Wallbox-Anbieter sind völlig an den Haaren herbeigezogen bis es eine öffentliche Spezifikation und irgendwelche Geräte gibt die den spezifizierten Kram auch wirklich umsetzen.

Da gibt es mehr als genug Fragezeichen auf allen Ebenen.

Die Gefahr ist maximal groß dass es Deutschland hier wieder schafft sich am Ende getrieben von Lobbyinteressen auf eine möglichst komplizierte, nicht offene Schnittstellenspezifikation "zu verständigen" die weder jemand ansehen noch sinnvoll und zum Nutzen aller Beteiligten implementieren kann.

christoph-dunker, 2024-02-13:
Als Installateur hat man aktuell leider keine Wahl, wenn man in unserem Netzgebiet eine Wallbox in Betrieb nehmen möchte.Die Steuerbox mit einem Schließer für einen potentialfreien Kontakt an der Wallbox oder ein Schütz ist Pflicht. Von daher bin ich froh, dass ich mit Go e auf 4,2 kW reduzieren kann und nicht komplett drosseln muss.Dass das alles viel zu schnell ging und jeder seine eigene Suppe kocht sehe ich auch so...

koaschten, 2024-02-13:
@premultiply  die go-e kann das sehr wohl schon umsetzen. Wie von @christoph-dunker und mir beschrieben

Siehe https://go-e.com/de-de/produkte/go-e-charger-gemini 
"jeweils + 2 Datenkabel für netzdienliche Steuerung"

Und wie in https://go-e.com/fileadmin/Support/Anleitungen/Deutsch/information-zur-paragraph-14aenwg-konformiaet-und-installation.pdf beschrieben ist, kann darüber mit einer entsprechenden Konfiguration für den Digitalen Eingang eine Strombegrenzung 6A/3phasig oder 18A/1phasig konfiguriert werden und dann mit Technikerpasswort den Digitalen Eingang vor Veränderung schützen.

DerAndereAndi, 2024-02-13:
> die weder jemand ansehen noch sinnvoll und zum Nutzen aller Beteiligten implementieren kann.

Einspruch.

premultiply, 2024-02-13:
Das alles hat wenig bis nichts mit den Vorhaben und Regelungen rund um §14a EnWG zu tun.

Es geht nicht darum Verbraucher wie zu Urgroßmutters Rundsteuerempfängerzeiten auf Befehl des Netzbetreibers stumpf zu drosseln sondern deren _netzwirksame_ Leistung fallweise auf minimal 4,2 kW pro Einheit zu reduzieren.

Das ist ein gewaltiger Unterschied wenn Eigenerzeugung mit im Spiel ist und lässt sich nicht mal eben mit ein paar Kontakten und Schützen sinngemäß umsetzen.

Oder um es noch deutlicher zu sagen: Das ist genau dass was der Gesetzgeber im Jahr 2024 und folgende eigentlich nicht mehr sehen will.

Netze sollen ausgebaut und nicht deren Nutzung eingeschränkt werden. Wo dies dennoch temporär Not tut soll dies so minimalinvasiv wie möglich, nachweislich und mit entsprechender Entschädigung an den Letztverbraucher erfolgen.

premultiply, 2024-02-13:
> > ...nicht offene Schnittstellenspezifikation "zu verständigen" die weder jemand ansehen noch sinnvoll und zum Nutzen aller Beteiligten implementieren kann.
> 
> Einspruch. @premultiply manchmal ist es hilfreich den Ball etwas flacher zu halten mit solchen Aussagen. Das ist einfach nur falsch.

Gerne. Aber dann mal bitte Butter bei die Fisch'.
Was genau ist daran falsch?

DerAndereAndi, 2024-02-13:
Die Spezifikation ist öffentlich einsehbar. Account auf eebus.org anlegen und herunterladen.

Sehr viele beteiligte Marktteilnehmer arbeiten bereits an der Unterstützung und Anwendung. Es gab ebenfalls letztes Jahr ein Treffen bei welchem die Umsetzungen einiger Geräte miteinander getestet wurden: https://www.eebus.org/eebusummit-2023-wie-die-digitale-vernetzung-die-energiewende-vorantreibt-und-§14a-enwg-in-der-praxis-umsetzbar-ist/

EEBUS als Standard ist der einzige momentan vom BSI und der Bundesnetzagentur bereits als mit §14a kompatibel bezeichnet. Es steht allen frei, bis Oktober 2024 weitere Vorschläge für alternative Protokolle zu machen.

Es gibt einen MIT lizenzierten Open Source Stack, von mir, mit welchem die Umsetzung mit EEBUS ebenfalls ohne Probleme möglich ist.

Also zusammenfassend:
- "nicht offene Schnittstellenspezifikation". Nein.
- "die weder jemand ansehen". Nein.
- "noch sinnvoll und zum Nutzen aller Beteiligten implementieren kann". Nein.

premultiply, 2024-02-13:
> Es gibt einen MIT lizenzierten Open Source Stack, von mir, mit welchem die Umsetzung mit EEBUS ebenfalls ohne Probleme möglich ist.

Das ist auch so ziemlich der realistisch einzige Weg der eine Umsetzung (wahrscheinlich sogar ganz grundsätzlich für alle Marktakteure überhaupt) ermöglicht.

rnndchsrl, 2024-02-16:
Verfolge das hier mal, da auch brandneu in EVCC unterwegs, mit WB, PV und Speicher.
Im Zusammenspiel mit EnWG 14a ist es ja auch interessant, das gem. aktuellem EEG §9 bei einer vorhandenen SteuVE die Auslesbarkeit der IST-Einspeisung und die Steuerung der Einspeiseleistung auch bei PV < 25kWp verlangt.

Darauf hat mich mein VNB bei der Anmeldung der PV hingewiesen.

> 1) Vorbehaltlich abweichender Vorgaben in einer aufgrund des § 95 Nummer 2 erlassenen Verordnung müssen die Betreiber von Anlagen und KWK-Anlagen mit einer installierten Leistung von mehr als 25 Kilowatt und die Betreiber von Anlagen, die hinter einem Netzanschluss mit mindestens einer steuerbaren Verbrauchseinrichtung nach § 14a des Energiewirtschaftsgesetzes betrieben werden, sicherstellen, dass bei ihren Anlagen und KWK-Anlagen spätestens zusammen mit dem intelligenten Messsystem technische Einrichtungen eingebaut werden, die notwendig sind, damit über ein Smart-Meter-Gateway nach § 2 Satz 1 Nummer 19 des Messstellenbetriebsgesetzes Netzbetreiber oder andere Berechtigte jederzeit entsprechend den Vorgaben in Schutzprofilen und in Technischen Richtlinien nach dem Messstellenbetriebsgesetz
> 
> 
> 
> 1.die Ist-Einspeisung abrufen können und
> 
> 2.die Einspeiseleistung stufenweise oder, sobald die technische Möglichkeit besteht, stufenlos ferngesteuert regeln können.

andig, 2024-02-17:
> Auch dazu gibt es noch keine spezifizierten Schnittstellen

@premultiply das ist so nicht ganz korrekt. Es ist bekannt, dass die Netzbetreiber auf EEBus setzen, siehe VDE FNN Impuls dazu.

> und wenn dann wäre das ein Fall für die PV-Systemhersteller.

Auch da stimme ich nicht zu. Hinter der SMGW kommt die steuerbare Ressource (SR), entweder eine PV, WP etc oder eben ein HEMS. Das wäre dann ein Fall für evcc.

premultiply, 2024-02-17:
Aber nur falls sämtliche HEMS-Logik und weitere Schnittstellen dann in evcc implementiert werden.

Warum sollte sich evcc auch noch z.B. eben um die Kontrolle und Abregelung der PV-Anlage kümmern wenn diese das in annähernder Perfektion autark tut oder tun könnte?

EEBus ist an sich ja nur das Protokoll. An sich aber erstmal ohne reale Funktion.
Relevant für reale Steuerungsmöglichkeiteb sind hier die Spezifikationen der Use Cases.

andig, 2024-02-17:
evcc würde sich um Fahrzeuge kümmern. PV ist aktuell out of scope.

> Relevant für reale Steuerungsmöglichkeiteb sind hier die Spezifikationen der Use Cases.

Das ist Teil der EEbus Spec.