evcc-io/evcc #151: Anpassung: PV-Regelung bei 1-phasiger Ladung an 3-phasigem loadpoint

ttjaden, 2020-05-15:
**Settings:**
```
charger: go-e
phases: 3
```
Da unser Zoe erste nächste Woche da ist, hat heute ein Freund die Wallbox mit einem Hyundai Ioniq eingeweiht. Der lädt einphasig, maximal 32A, also ca 7,3 kW.

**Beobachtung:**
- Die Steuerung des Ladestroms durch EVCC geht davon aus, dass dreiphasig geladen wird und berechnet daraufhin den Strom, der maximal fließen soll. Da jedoch nur einphasig geladen wird, fließt viel PV Strom ins Netz, siehe
[image]
--> grün die Bereich, wo mit "Sofort laden" der maximale Strom und damit die 7,2 kW Ladeleistung erreicht wurden.
--> nach den "Sofort laden" Zeiten fällt die Ladung im PV-Modus langsam ab. Ist das agiler einstellbar mit dem Setting
`sensitivity: 1 # current raise/lower step size (default 10A)`
--> Ist hier die Einheit Ampere/min gemeint?

**Hauptfrage:**
Wie bekommt man es hin, dass auch einphasige Autos an einem dreihpasigem loadpoint im PV-Modus bei genügend Überschuss mit entsprechend mehr Leistung geladen werden?

premultiply, 2020-05-15:
Derzeit wohl nur statisch über die genannte Konfigurationsoption.

Daneben gibt es übrigens auch 2-phasig ladende Fahrzeuge z. B. von VW.

Womöglich könnte man nach dem Ladebeginn mittels geeignetem Zähler/Wallbox die Phasenwirkleistung auslesen und bei Werten >0 dadurch den Strommultiplikator dynamisch anpassen. Grundvoraussetzung ist natürlich dass das jeweilige Zählgerät diese Funktion bietet.
Beim NRGkick z. B. wäre das kein Problem.

andig, 2020-05-15:
@ttjaden was ist Dein Anwendungsfall- unterschiedlich-phasige Fahrzeuge an einem Ladepunkt?

Woher kommt das schicke Diagramm- sieht nach Discovergy aus?

andig, 2020-05-15:
> Ist hier die Einheit Ampere/min gemeint?

Nein, das sind Ampere pro Regelschritt, also Schrittweite.

ttjaden, 2020-05-15:
> > Ist hier die Einheit Ampere/min gemeint?
> 
> Nein, das sind Ampere pro Regelschritt, also Schrittweite.

Alles klar, danke dir! Dann werde ich den Wert von 1 wieder etwas erhöhen. Wenn der ZOE nächste Woche da ist, werde ich mal einen wechselnd bewölkten Tag abwarten und dann das ganze nochmal testen und loggen.
Anschließend trage ich dann endlich auch mal zum WIKI bei für 
-- ein Beispiel-Template (Discovergy Smartmeter + go-e Charger + Zoe + PV-Laden)
-- Nutzung EVCC auf Synology-NAS
-- Nutzung EVCC auf Ubuntu-Server mit nginx Reverse Proxy und optionalem Login

andig, 2020-05-15:
Super! Besser wäre es aber die Module einzeln zu beschreiben- als Kombi gibts sonst zuviele Varianten.

ttjaden, 2020-05-15:
Es ist kein richtiger Anwendungsfall. Wir haben ja nur den Zoe. Wäre
aber natürlich trotzdem klasse, wenn man auch Gast-Autos, die nur
1-phasig laden, richtig mit Überschussstrom versorgen kann. Der Ansatz
über das Auslesen der Phasenströme, wie ihr auf github schon beschrieben
habt, ist aber ein guter Ansatz!

Das Diagramm kommt von Discovergy, ja. Wir haben die Smartmeter
offiziell von Commetering, weil ich die sehr sympathisch als
Messstellenbetreiber finde.

closed 2020-05-15 by ttjaden

TheRealKerni, 2020-05-15:
Der Fall könnte doch auch auftreten, wenn die Ladung gegen das Ende der Kapazität geht und dann das Auto auch bei 3 phasigem Anschluss umschaltet auf 1 phasig um mit niedriger Leistung voll zu laden.

premultiply, 2020-05-15:
Zumindest alle mir bekannten Fahrzeuge sind in ihrer Phasenkonfiguration entweder ohnehin starr oder sie prüfen andernfalls vor Ladebeginn die Netzverhältnisse, konfigurieren dann die Netzanschaltung der Ladesysteme und Netzfilter über Relais und behalten diese Konfiguration dann aber auch für den kompletten Ladevorgang bis zur Abschaltung bei.