evcc-io/evcc #152: Phasennutzungserkennung

premultiply, 2020-05-15:
Bei geeigneten LP-Zählern soll über die Phasenwirkleistung (>0) die Anzahl der vom Fahrzeug verwendeten Phasen automatisch während des Ladevorgangs ermittelt werden und der Berechnungsmultiplikator somit dynamisch angepasst werden.

Darüber hinaus wäre es ggf. auch hilfreich den verwendeten Strommultiplilator in der GUI und in der API zu veröffentlichen.

_Eine alternative Methode (aber deutlich fehlerbehafteter) für rein summierender Wirkleistungsmessung wäre anhand der bekannten vorgegebenen Phasenstromlimitierung, einer angenommenen oder gemessenen Netzspannung und dem Summenwirkleistungsmesswert den zugehörigen Multiplikator zu ermitteln. Hier ist aber das Problem, dass ein Fahrzeug auch deutlich weniger Leistung ziehen kann/darf als durch die Ladestromvorgabe erlaubt._

andig, 2020-05-15:
Ab welchem Strom/ welcher Leistung würde man eine Phase als genutzt betrachten?

premultiply, 2020-05-15:
Mehr als 0

Nur in geschlossenen Stromkreisen fließt ein Strom.

TheRealKerni, 2020-05-15:
Es gibt auch EVs, die können wohl bei einem 3 phasigen Anschluss einzelne Lader "abschalten", z.b. wenn es gegen Kapazitätsende der Batterie geht.

Hier ein Beispiel einer solchen Ladung mit einem Datensatz vom Mobile Charger Connect:

```json
{
  "L1": {"Ampere": 7.9000000000000004, "Power": 1907, "Volts ": 240.40000000000001},
  "L2": {"Ampere": 0.5, "Power": 0, "Volts": 244.80000000000001},
  "L3": {"Ampere": 0.5, "Power": 1, "Volts": 243.30000000000001}
}
```

premultiply, 2020-05-15:
Oha. Da müssen wir dann wohl doch eine etwas höhere Schwelle als 0.0 für Phaseninaktivität angeben.
Vielleicht < 1.0 A?

andig, 2020-07-15:
/keep

andig, 2020-07-16:
@premultiply der PR ist auf aktuellen Master aktualisiert- Du könntest testen ;)

premultiply, 2020-07-16:
Ab Sonntag wieder am Gerät 😉
Was sollte man denn sehen oder feststellen können?

andig, 2020-07-16:
Sobald ein Ladevorgang läuft (ohne fließt ja kein Strom) solltest Du- falls Strommessung irgendwo unterstützt wird (Charger? Zähler?) im Log einen Eintrag sehen der dann auch verwendet würde. Analog würdest Du unter `/api/state` sehen wieviele `phases` erkannt wurden.

andig, 2020-07-18:
...und die PV Berechung sollte entsprechend oassen. Wenn charger mit 3p konfiguriert aber mit 1p geladen sollte der nächste Ladevorgang bei Minimalleistung für 1p starten statt auf 3p zu warten.

PR #163 (2020-07-19): Implement phase detection
Fix #152
closed 2020-07-19 by andig