evcc-io/evcc #157: NRGkick startet Ladung nach verbinden des Fahrzeugs

premultiply, 2020-05-15:
Da der NRGkick intern den chargingPause-Status zurücksetzt wenn ein Fahrzeug neu verbunden wird startet automatisch immer die Ladung und (Bug ab hier:) wird auch nicht gleich wieder beendet obwohl EVCC z. B. OFF oder im PV-Modus ist aber Nachts überhaupt keine PV-Energie da ist.
Mögliche Lösung: #124

premultiply, 2020-05-15:
Korrektur: Es läuft das Contactor-Delay ab (bei mir 10m). Erst danach wird der Ladevorgang dann korrekt pausiert.
In diesem Fall sollte aber schnellstmöglich chargingPause auf 1 gesetzt werden ohne das Delay abzuwarten.

andig, 2020-05-16:
@premultiply das sieht erstmal so korrekt aus. Wenn der Charger eigenständig startet dann sollte er egtl. Auch erst nach delay wieder abgeschaltet werden (auch wenn das nicht schön ist). Was mich aber wundert ist, Was sich noch einbauen liesse wäre eine Sonderprüfung die schaut ob charging active obwohl enabled= false. Hättest Du mal einen Logsauschnitt der zeigt was da passiert, also ob evcc das pberhaupt mit bekommt und wie?

premultiply, 2020-05-18:
Hier der passende Logausschnitt. EVCC ist nachts im PV-Modus. Fahrzeug wird verbunden und fängt einfach mit 6A an zu laden ohne dass EVCC etwas dagegen tut. EVCC merkt sogar dass geladen wird und verschickt Push per Telegram. Eigentlich sollte es allerspätestens an dieser Stelle den Ladevorgang unterbinden, besser jedoch direkt nach dem anstecken.

```
[nrgbt] TRACE 2020/05/18 23:23:25 read fd005380-b065-11e4-9ce2-0002a5d6b15d 000000000000000001d7138800201d07016a0498
[nrgbt] TRACE 2020/05/18 23:23:25 read power: {TotalPower:0 L1:0 L2:0 L3:0 PeakPower:471 Frequency:5000 Temperature:32 RemainingDistance:7431 Costs:362 CPSignal:4}
[load] DEBUG 2020/05/18 23:23:25 main charge power: 0.0W
[nrgbt] TRACE 2020/05/18 23:23:28 read fd005380-b065-11e4-9ce2-0002a5d6b15d 000000000000000001d7138800201d07016a0498
[nrgbt] TRACE 2020/05/18 23:23:28 read power: {TotalPower:0 L1:0 L2:0 L3:0 PeakPower:471 Frequency:5000 Temperature:32 RemainingDistance:7431 Costs:362 CPSignal:4}
[load] DEBUG 2020/05/18 23:23:28 main charger status: A
[load] DEBUG 2020/05/18 23:23:28 main target power: -803W = 0W charge - 803W grid - 0W residual
[load] DEBUG 2020/05/18 23:23:28 main target charge current: 0A
[load] DEBUG 2020/05/18 23:23:30 main grid power: 803.8W
[nrgbt] TRACE 2020/05/18 23:23:30 read fd005380-b065-11e4-9ce2-0002a5d6b15d 000000000000000001d7138a00201d07016a035b
[nrgbt] TRACE 2020/05/18 23:23:30 read power: {TotalPower:0 L1:0 L2:0 L3:0 PeakPower:471 Frequency:5002 Temperature:32 RemainingDistance:7431 Costs:362 CPSignal:3}
[load] DEBUG 2020/05/18 23:23:30 main charge power: 0.0W
[nrgbt] TRACE 2020/05/18 23:23:33 read fd005380-b065-11e4-9ce2-0002a5d6b15d 000000000000000001d7138a00201d07016a035b
[nrgbt] TRACE 2020/05/18 23:23:33 read power: {TotalPower:0 L1:0 L2:0 L3:0 PeakPower:471 Frequency:5002 Temperature:32 RemainingDistance:7431 Costs:362 CPSignal:3}
[load] DEBUG 2020/05/18 23:23:33 main charger status: B
[load] INFO 2020/05/18 23:23:33 main car connected (B)
[load] DEBUG 2020/05/18 23:23:33 main target power: -804W = 0W charge - 804W grid - 0W residual
[load] DEBUG 2020/05/18 23:23:33 main target charge current: 0A
[load] DEBUG 2020/05/18 23:23:35 main grid power: 805.4W
[nrgbt] TRACE 2020/05/18 23:23:35 read fd005380-b065-11e4-9ce2-0002a5d6b15d 000000000000000001d7138a00201d07016a035b
[nrgbt] TRACE 2020/05/18 23:23:35 read power: {TotalPower:0 L1:0 L2:0 L3:0 PeakPower:471 Frequency:5002 Temperature:32 RemainingDistance:7431 Costs:362 CPSignal:3}
[load] DEBUG 2020/05/18 23:23:35 main charge power: 0.0W
[nrgbt] TRACE 2020/05/18 23:23:38 read fd005380-b065-11e4-9ce2-0002a5d6b15d 000000000000000001d7138a00201d07016a0365
[nrgbt] TRACE 2020/05/18 23:23:38 read power: {TotalPower:0 L1:0 L2:0 L3:0 PeakPower:471 Frequency:5002 Temperature:32 RemainingDistance:7431 Costs:362 CPSignal:3}
[load] DEBUG 2020/05/18 23:23:38 main charger status: B
[load] DEBUG 2020/05/18 23:23:38 main target power: -805W = 0W charge - 805W grid - 0W residual
[load] DEBUG 2020/05/18 23:23:38 main target charge current: 0A
[load] DEBUG 2020/05/18 23:23:40 main grid power: 807.4W
[nrgbt] TRACE 2020/05/18 23:23:40 read fd005380-b065-11e4-9ce2-0002a5d6b15d 000000000000000001d7138a00201d07016a0365
[nrgbt] TRACE 2020/05/18 23:23:40 read power: {TotalPower:0 L1:0 L2:0 L3:0 PeakPower:471 Frequency:5002 Temperature:32 RemainingDistance:7431 Costs:362 CPSignal:3}
[load] DEBUG 2020/05/18 23:23:40 main charge power: 0.0W
[nrgbt] TRACE 2020/05/18 23:23:43 read fd005380-b065-11e4-9ce2-0002a5d6b15d 000000000000000001d7138a0020000000000242
[nrgbt] TRACE 2020/05/18 23:23:43 read power: {TotalPower:0 L1:0 L2:0 L3:0 PeakPower:471 Frequency:5002 Temperature:32 RemainingDistance:0 Costs:0 CPSignal:2}
[load] DEBUG 2020/05/18 23:23:43 main charger status: C
[load] INFO 2020/05/18 23:23:43 main start charging ->
[nrgbt] TRACE 2020/05/18 23:23:45 read 0379e580-ad1b-11e4-8bdd-0002a5d6b15d 00194b350000589d0000329a000000b64e1d00
[nrgbt] TRACE 2020/05/18 23:23:45 read energy: {TotalEnergy:1657653 EnergyLastCharge:22685 Energy2ndLastCharge:12954 Energy3rdLastCharge:182 ChargingEnergyLimit:19997 Pad:0}
[wrap] DEBUG 2020/05/18 23:23:45 main charge start energy: 1658kWh
[push] TRACE 2020/05/18 23:23:45 telegram: sending to 266162151
[load] DEBUG 2020/05/18 23:23:45 main target power: -807W = 0W charge - 807W grid - 0W residual
[load] DEBUG 2020/05/18 23:23:45 main target charge current: 0A
[nrgbt] TRACE 2020/05/18 23:23:47 read 0379e580-ad1b-11e4-8bdd-0002a5d6b15d 00194b350000589d0000329a000000b64e1d00
[nrgbt] TRACE 2020/05/18 23:23:47 read energy: {TotalEnergy:1657653 EnergyLastCharge:22685 Energy2ndLastCharge:12954 Energy3rdLastCharge:182 ChargingEnergyLimit:19997 Pad:0}
[load] DEBUG 2020/05/18 23:23:47 main grid power: 1912.3W
[nrgbt] TRACE 2020/05/18 23:23:49 read fd005380-b065-11e4-9ce2-0002a5d6b15d 006d006d0000000001d71389002000000000023d
[nrgbt] TRACE 2020/05/18 23:23:49 read power: {TotalPower:109 L1:109 L2:0 L3:0 PeakPower:471 Frequency:5001 Temperature:32 RemainingDistance:0 Costs:0 CPSignal:2}
[load] DEBUG 2020/05/18 23:23:49 main charge power: 1090.0W
[nrgbt] TRACE 2020/05/18 23:23:51 read fd005380-b065-11e4-9ce2-0002a5d6b15d 006d006d0000000001d713890020000000000243
[nrgbt] TRACE 2020/05/18 23:23:51 read power: {TotalPower:109 L1:109 L2:0 L3:0 PeakPower:471 Frequency:5001 Temperature:32 RemainingDistance:0 Costs:0 CPSignal:2}
[load] DEBUG 2020/05/18 23:23:51 main charger status: C
[... cut ...]
[nrgbt] TRACE 2020/05/18 23:24:10 read fd005380-b065-11e4-9ce2-0002a5d6b15d 006d006d0000000001d713880020000200000242
[nrgbt] TRACE 2020/05/18 23:24:10 read power: {TotalPower:109 L1:109 L2:0 L3:0 PeakPower:471 Frequency:5000 Temperature:32 RemainingDistance:2 Costs:0 CPSignal:2}
[load] DEBUG 2020/05/18 23:24:10 main charger status: C
[load] DEBUG 2020/05/18 23:24:10 main target power: -808W = 1090W charge - 1898W grid - 0W residual
[load] DEBUG 2020/05/18 23:24:10 main target charge current: 0A
.
.
.
```

premultiply, 2020-05-18:
Wenn man in diesem Zustand EVCC neustartet wird es übrigens auch nicht besser. Der Ladezustand des NRGkicks wird sogar explizit auf enabled (ChargingPause:false) gesetzt:

```
pi@evcc:~/evcc $ sudo ~/go/bin/go run main.go
[main] INFO 2020/05/18 23:38:35 evcc dev (HEAD)
[main] INFO 2020/05/18 23:38:35 using config file /home/pi/evcc/evcc.yaml
[main] INFO 2020/05/18 23:38:35 listening at 0.0.0.0:7070
[nrgbt] WARN 2020/05/18 23:38:36 -- experimental --
[load] INFO 2020/05/18 23:38:36 main loadpoint config: vehicle — grid ✓ pv — battery — charge —
[load] INFO 2020/05/18 23:38:36 main charger config: power ✓ energy — timer —
[load] INFO 2020/05/18 23:38:36 main charge mode: pv
[nrgbt] TRACE 2020/05/18 23:38:38 read 8f75bba0-c903-11e4-9fe8-0002a5d6b15d 06001d1001005f0101200400be
[nrgbt] TRACE 2020/05/18 23:38:38 read info: {Current:6 KWhPer100:29 AmountPerKWh:16 FIEnabled:1 ErrorCode:0 Efficiency:95 ChargingActive:true PauseCharging:true ChargingCurrentMax:32 BLETransmissionPower:4 Pad:0}
[load] INFO 2020/05/18 23:38:38 main charger disabled
[load] DEBUG 2020/05/18 23:38:38 main set charge current: 6A
[nrgbt] TRACE 2020/05/18 23:38:40 read 8f75bba0-c903-11e4-9fe8-0002a5d6b15d 06001d1001005f0101200400be
[nrgbt] TRACE 2020/05/18 23:38:40 write settings: {PIN:xxxx Current:6 ChargingEnergyLimit:19997 KWhPer100:29 AmountPerKWh:16 Pad:0 Efficiency:95 PauseCharging:false BLETransmissionPower:4 PadTail:0}
[nrgbt] TRACE 2020/05/18 23:38:40 write 14b3afc0-ad1b-11e4-baab-0002a5d6b15d 028b064e1d001d1000005f00040000000000
[load] DEBUG 2020/05/18 23:38:42 main grid power: 1714.5W
[nrgbt] TRACE 2020/05/18 23:38:45 read fd005380-b065-11e4-9ce2-0002a5d6b15d 006a006a0000000001d713890023005900040242
[nrgbt] TRACE 2020/05/18 23:38:45 read power: {TotalPower:106 L1:106 L2:0 L3:0 PeakPower:471 Frequency:5001 Temperature:35 RemainingDistance:89 Costs:4 CPSignal:2}
[load] DEBUG 2020/05/18 23:38:45 main charge power: 1060.0W
[nrgbt] TRACE 2020/05/18 23:38:47 read fd005380-b065-11e4-9ce2-0002a5d6b15d 006a006a0000000001d713890023005900040242
[nrgbt] TRACE 2020/05/18 23:38:47 read power: {TotalPower:106 L1:106 L2:0 L3:0 PeakPower:471 Frequency:5001 Temperature:35 RemainingDistance:89 Costs:4 CPSignal:2}
[load] DEBUG 2020/05/18 23:38:47 main charger status: C
[load] INFO 2020/05/18 23:38:47 main start charging ->
[nrgbt] TRACE 2020/05/18 23:38:49 read 0379e580-ad1b-11e4-8bdd-0002a5d6b15d 00194c470000589d0000329a000000b64e1d00
[nrgbt] TRACE 2020/05/18 23:38:49 read energy: {TotalEnergy:1657927 EnergyLastCharge:22685 Energy2ndLastCharge:12954 Energy3rdLastCharge:182 ChargingEnergyLimit:19997 Pad:0}
[wrap] DEBUG 2020/05/18 23:38:49 main charge start energy: 1658kWh
[load] DEBUG 2020/05/18 23:38:49 main target power: -654W = 1060W charge - 1714W grid - 0W residual
[push] TRACE 2020/05/18 23:38:49 telegram: sending to 266162151
[load] DEBUG 2020/05/18 23:38:49 main target charge current: 0A
[nrgbt] TRACE 2020/05/18 23:38:51 read 0379e580-ad1b-11e4-8bdd-0002a5d6b15d 00194c480000589d0000329a000000b64e1d00
[nrgbt] TRACE 2020/05/18 23:38:51 read energy: {TotalEnergy:1657928 EnergyLastCharge:22685 Energy2ndLastCharge:12954 Energy3rdLastCharge:182 ChargingEnergyLimit:19997 Pad:0}
[load] DEBUG 2020/05/18 23:38:51 main grid power: 1709.9W
[nrgbt] TRACE 2020/05/18 23:38:53 read fd005380-b065-11e4-9ce2-0002a5d6b15d 006a006a0000000001d713890023005a0004023c
[nrgbt] TRACE 2020/05/18 23:38:53 read power: {TotalPower:106 L1:106 L2:0 L3:0 PeakPower:471 Frequency:5001 Temperature:35 RemainingDistance:90 Costs:4 CPSignal:2}
[load] DEBUG 2020/05/18 23:38:53 main charge power: 1060.0W
[nrgbt] TRACE 2020/05/18 23:38:55 read fd005380-b065-11e4-9ce2-0002a5d6b15d 006a006a0000000001d713890023005a0004023c
[nrgbt] TRACE 2020/05/18 23:38:55 read power: {TotalPower:106 L1:106 L2:0 L3:0 PeakPower:471 Frequency:5001 Temperature:35 RemainingDistance:90 Costs:4 CPSignal:2}
[load] DEBUG 2020/05/18 23:38:55 main charger status: C
[load] DEBUG 2020/05/18 23:38:55 main target power: -650W = 1060W charge - 1710W grid - 0W residual
[load] DEBUG 2020/05/18 23:38:55 main target charge current: 0A
[nrgbt] TRACE 2020/05/18 23:38:57 read 0379e580-ad1b-11e4-8bdd-0002a5d6b15d 00194c4a0000589d0000329a000000b64e1d00
[nrgbt] TRACE 2020/05/18 23:38:57 read energy: {TotalEnergy:1657930 EnergyLastCharge:22685 Energy2ndLastCharge:12954 Energy3rdLastCharge:182 ChargingEnergyLimit:19997 Pad:0}
[load] DEBUG 2020/05/18 23:39:02 main grid power: 1674.9W
[nrgbt] TRACE 2020/05/18 23:39:03 read fd005380-b065-11e4-9ce2-0002a5d6b15d 006a006d0000000001d713890023005b00040242
[nrgbt] TRACE 2020/05/18 23:39:03 read power: {TotalPower:106 L1:109 L2:0 L3:0 PeakPower:471 Frequency:5001 Temperature:35 RemainingDistance:91 Costs:4 CPSignal:2}
[load] DEBUG 2020/05/18 23:39:03 main charge power: 1060.0W
[nrgbt] TRACE 2020/05/18 23:39:05 read fd005380-b065-11e4-9ce2-0002a5d6b15d 006a006a0000000001d713890023005b00040242
[nrgbt] TRACE 2020/05/18 23:39:05 read power: {TotalPower:106 L1:106 L2:0 L3:0 PeakPower:471 Frequency:5001 Temperature:35 RemainingDistance:91 Costs:4 CPSignal:2}
[load] DEBUG 2020/05/18 23:39:05 main charger status: C
[load] DEBUG 2020/05/18 23:39:05 main target power: -615W = 1060W charge - 1675W grid - 0W residual
[load] DEBUG 2020/05/18 23:39:05 main target charge current: 0A
[nrgbt] TRACE 2020/05/18 23:39:07 read 0379e580-ad1b-11e4-8bdd-0002a5d6b15d 00194c4d0000589d0000329a000000b64e1d00
[... cut ...]
[load] DEBUG 2020/05/18 23:39:24 main target power: -557W = 1060W charge - 1617W grid - 0W residual
[load] DEBUG 2020/05/18 23:39:24 main target charge current: 0A
[nrgbt] TRACE 2020/05/18 23:39:26 read 0379e580-ad1b-11e4-8bdd-0002a5d6b15d 00194c530000589d0000329a000000b64e1d00
[nrgbt] TRACE 2020/05/18 23:39:26 read energy: {TotalEnergy:1657939 EnergyLastCharge:22685 Energy2ndLastCharge:12954 Energy3rdLastCharge:182 ChargingEnergyLimit:19997 Pad:0}
[load] DEBUG 2020/05/18 23:39:26 main grid power: 1619.8W
^Csignal: interrupt
pi@evcc:~/evcc $
```

andig, 2020-05-19:
Magst Du mal schauen ob die #172 das Problem löst?

premultiply, 2020-05-19:
Leider noch nicht ganz, weil der NRGkick zickt und zu diesem Zeitpunkt trotz Status ChargingPause:true einfach trotzdem lädt. Ich glaub man muss es ihm explizit nach dem/jedem CP-Statuswechsel nochmal reinschreiben wie man es jetzt haben will. AAAARG!

```
[load] DEBUG 2020/05/19 11:28:48 main charger status: B
[nrgbt] TRACE 2020/05/19 11:28:50 read 8f75bba0-c903-11e4-9fe8-0002a5d6b15d 06001d1000005f0001200400c0
[nrgbt] TRACE 2020/05/19 11:28:50 info: {Current:6 KWhPer100:29 AmountPerKWh:16 FIEnabled:0 ErrorCode:0 Efficiency:95 ChargingActive:false PauseCharging:true ChargingCurrentMax:32 BLETransmissionPower:4 Pad:0}
[load] DEBUG 2020/05/19 11:28:50 main grid power: -4464.8W
[nrgbt] TRACE 2020/05/19 11:28:52 read fd005380-b065-11e4-9ce2-0002a5d6b15d 000000000000000001d7138800340013000004a6
[nrgbt] TRACE 2020/05/19 11:28:52 power: {TotalPower:0 L1:0 L2:0 L3:0 PeakPower:471 Frequency:5000 Temperature:52 RemainingDistance:19 Costs:0 CPSignal:4}
[load] DEBUG 2020/05/19 11:28:52 main charge power: 0.0W
[nrgbt] TRACE 2020/05/19 11:28:54 read fd005380-b065-11e4-9ce2-0002a5d6b15d 000000000000000001d7138800340013000004a6
[nrgbt] TRACE 2020/05/19 11:28:54 power: {TotalPower:0 L1:0 L2:0 L3:0 PeakPower:471 Frequency:5000 Temperature:52 RemainingDistance:19 Costs:0 CPSignal:4}
[load] DEBUG 2020/05/19 11:28:54 main charger status: A
[load] INFO 2020/05/19 11:28:54 main car disconnected
[load] DEBUG 2020/05/19 11:28:54 main grid power: -4489.4W
[nrgbt] TRACE 2020/05/19 11:28:56 read fd005380-b065-11e4-9ce2-0002a5d6b15d 000000000000000001d7138800340013000004a3
[nrgbt] TRACE 2020/05/19 11:28:56 power: {TotalPower:0 L1:0 L2:0 L3:0 PeakPower:471 Frequency:5000 Temperature:52 RemainingDistance:19 Costs:0 CPSignal:4}
[load] DEBUG 2020/05/19 11:28:56 main charge power: 0.0W
[nrgbt] TRACE 2020/05/19 11:28:58 read fd005380-b065-11e4-9ce2-0002a5d6b15d 000000000000000001d7138800340013000004a3
[nrgbt] TRACE 2020/05/19 11:28:58 power: {TotalPower:0 L1:0 L2:0 L3:0 PeakPower:471 Frequency:5000 Temperature:52 RemainingDistance:19 Costs:0 CPSignal:4}
[load] DEBUG 2020/05/19 11:28:58 main charger status: A
[load] DEBUG 2020/05/19 11:28:58 main grid power: -4491.0W
[nrgbt] TRACE 2020/05/19 11:29:00 read fd005380-b065-11e4-9ce2-0002a5d6b15d 000000000000000001d7138900340013000004a6
[nrgbt] TRACE 2020/05/19 11:29:00 power: {TotalPower:0 L1:0 L2:0 L3:0 PeakPower:471 Frequency:5001 Temperature:52 RemainingDistance:19 Costs:0 CPSignal:4}
[load] DEBUG 2020/05/19 11:29:00 main charge power: 0.0W
[nrgbt] TRACE 2020/05/19 11:29:03 read fd005380-b065-11e4-9ce2-0002a5d6b15d 000000000000000001d7138900340013000004a6
[nrgbt] TRACE 2020/05/19 11:29:03 power: {TotalPower:0 L1:0 L2:0 L3:0 PeakPower:471 Frequency:5001 Temperature:52 RemainingDistance:19 Costs:0 CPSignal:4}
[load] DEBUG 2020/05/19 11:29:03 main charger status: A
[load] DEBUG 2020/05/19 11:29:03 main grid power: -4452.8W
[nrgbt] TRACE 2020/05/19 11:29:05 read fd005380-b065-11e4-9ce2-0002a5d6b15d 000000000000000001d7138900340013000004a6
[nrgbt] TRACE 2020/05/19 11:29:05 power: {TotalPower:0 L1:0 L2:0 L3:0 PeakPower:471 Frequency:5001 Temperature:52 RemainingDistance:19 Costs:0 CPSignal:4}
[load] DEBUG 2020/05/19 11:29:05 main charge power: 0.0W
[nrgbt] TRACE 2020/05/19 11:29:07 read fd005380-b065-11e4-9ce2-0002a5d6b15d 000000000000000001d71388003400130000036c
[nrgbt] TRACE 2020/05/19 11:29:07 power: {TotalPower:0 L1:0 L2:0 L3:0 PeakPower:471 Frequency:5000 Temperature:52 RemainingDistance:19 Costs:0 CPSignal:3}
[load] DEBUG 2020/05/19 11:29:07 main charger status: B
[load] INFO 2020/05/19 11:29:07 main car connected (B)
[nrgbt] TRACE 2020/05/19 11:29:09 read 8f75bba0-c903-11e4-9fe8-0002a5d6b15d 06001d1000005f0001200400c0
[nrgbt] TRACE 2020/05/19 11:29:09 info: {Current:6 KWhPer100:29 AmountPerKWh:16 FIEnabled:0 ErrorCode:0 Efficiency:95 ChargingActive:false PauseCharging:true ChargingCurrentMax:32 BLETransmissionPower:4 Pad:0}
[load] DEBUG 2020/05/19 11:29:09 main grid power: -4505.1W
[nrgbt] TRACE 2020/05/19 11:29:11 read fd005380-b065-11e4-9ce2-0002a5d6b15d 000000000000000001d71388003400130000036c
[nrgbt] TRACE 2020/05/19 11:29:11 power: {TotalPower:0 L1:0 L2:0 L3:0 PeakPower:471 Frequency:5000 Temperature:52 RemainingDistance:19 Costs:0 CPSignal:3}
[load] DEBUG 2020/05/19 11:29:11 main charge power: 0.0W
[nrgbt] TRACE 2020/05/19 11:29:13 read fd005380-b065-11e4-9ce2-0002a5d6b15d 000000000000000001d71388003400130000036c
[nrgbt] TRACE 2020/05/19 11:29:13 power: {TotalPower:0 L1:0 L2:0 L3:0 PeakPower:471 Frequency:5000 Temperature:52 RemainingDistance:19 Costs:0 CPSignal:3}
[load] DEBUG 2020/05/19 11:29:13 main charger status: B
[nrgbt] TRACE 2020/05/19 11:29:15 read 8f75bba0-c903-11e4-9fe8-0002a5d6b15d 06001d1001005f0101200400c0
[nrgbt] TRACE 2020/05/19 11:29:15 info: {Current:6 KWhPer100:29 AmountPerKWh:16 FIEnabled:1 ErrorCode:0 Efficiency:95 ChargingActive:true PauseCharging:true ChargingCurrentMax:32 BLETransmissionPower:4 Pad:0}
[load] DEBUG 2020/05/19 11:29:15 main grid power: -4502.2W
[nrgbt] TRACE 2020/05/19 11:29:18 read fd005380-b065-11e4-9ce2-0002a5d6b15d 000000000000000001d713880034000000000244
[nrgbt] TRACE 2020/05/19 11:29:18 power: {TotalPower:0 L1:0 L2:0 L3:0 PeakPower:471 Frequency:5000 Temperature:52 RemainingDistance:0 Costs:0 CPSignal:2}
[load] DEBUG 2020/05/19 11:29:18 main charge power: 0.0W
[nrgbt] TRACE 2020/05/19 11:29:20 read fd005380-b065-11e4-9ce2-0002a5d6b15d 004300470000000001d71389003400000000024a
[... cut ...]
[nrgbt] TRACE 2020/05/19 11:29:41 info: {Current:6 KWhPer100:29 AmountPerKWh:16 FIEnabled:1 ErrorCode:0 Efficiency:95 ChargingActive:true PauseCharging:true ChargingCurrentMax:32 BLETransmissionPower:4 Pad:0}
[nrgbt] TRACE 2020/05/19 11:29:43 read 0379e580-ad1b-11e4-8bdd-0002a5d6b15d 00194d1e0000003d000001a50000589d4e1d00
[nrgbt] TRACE 2020/05/19 11:29:43 energy: {TotalEnergy:1658142 EnergyLastCharge:61 Energy2ndLastCharge:421 Energy3rdLastCharge:22685 ChargingEnergyLimit:19997 Pad:0}
[load] DEBUG 2020/05/19 11:29:43 main grid power: -3400.6W

```

premultiply, 2020-05-19:
Ich mach nochmal gerade einen neuen Test. Nicht dass sich das Ding irgendwie verschluckt hat...

premultiply, 2020-05-19:
So, nach EVCC auf "stop", Strom aus - aber Fahrzeug nicht abgesteckt - funktioniert es nun so wie erwartet:

```
[nrgbt] TRACE 2020/05/19 11:55:24 read fd005380-b065-11e4-9ce2-0002a5d6b15d 000000000000000001d71388003a00fe000c0363
[nrgbt] TRACE 2020/05/19 11:55:24 power: {TotalPower:0 L1:0 L2:0 L3:0 PeakPower:471 Frequency:5000 Temperature:58 RemainingDistance:254 Costs:12 CPSignal:3}
[load] DEBUG 2020/05/19 11:55:24 main charger status: B
[nrgbt] TRACE 2020/05/19 11:55:26 read 8f75bba0-c903-11e4-9fe8-0002a5d6b15d 06001d1000005f0001200400c1
[nrgbt] TRACE 2020/05/19 11:55:26 info: {Current:6 KWhPer100:29 AmountPerKWh:16 FIEnabled:0 ErrorCode:0 Efficiency:95 ChargingActive:false PauseCharging:true ChargingCurrentMax:32 BLETransmissionPower:4 Pad:0}
[load] DEBUG 2020/05/19 11:55:26 main grid power: -4759.6W
[nrgbt] TRACE 2020/05/19 11:55:28 read fd005380-b065-11e4-9ce2-0002a5d6b15d 000000000000000001d71388003a00fe000c0378
[nrgbt] TRACE 2020/05/19 11:55:28 power: {TotalPower:0 L1:0 L2:0 L3:0 PeakPower:471 Frequency:5000 Temperature:58 RemainingDistance:254 Costs:12 CPSignal:3}
[load] DEBUG 2020/05/19 11:55:28 main charge power: 0.0W
[nrgbt] TRACE 2020/05/19 11:57:58 read fd005380-b065-11e4-9ce2-0002a5d6b15d 0000000000000000000013890039000000000373
[nrgbt] TRACE 2020/05/19 11:57:58 power: {TotalPower:0 L1:0 L2:0 L3:0 PeakPower:0 Frequency:5001 Temperature:57 RemainingDistance:0 Costs:0 CPSignal:3}
[load] DEBUG 2020/05/19 11:57:58 main charger status: B
[nrgbt] TRACE 2020/05/19 11:58:00 read 8f75bba0-c903-11e4-9fe8-0002a5d6b15d 06001d1000005f0000200400c1
[nrgbt] TRACE 2020/05/19 11:58:00 info: {Current:6 KWhPer100:29 AmountPerKWh:16 FIEnabled:0 ErrorCode:0 Efficiency:95 ChargingActive:false PauseCharging:false ChargingCurrentMax:32 BLETransmissionPower:4 Pad:0}
[load] ERROR 2020/05/19 11:58:00 main sync enabled
[nrgbt] TRACE 2020/05/19 11:58:03 read 8f75bba0-c903-11e4-9fe8-0002a5d6b15d 06001d1000005f0000200400c1
[nrgbt] TRACE 2020/05/19 11:58:03 info: {Current:6 KWhPer100:29 AmountPerKWh:16 FIEnabled:0 ErrorCode:0 Efficiency:95 ChargingActive:false PauseCharging:false ChargingCurrentMax:32 BLETransmissionPower:4 Pad:0}
[nrgbt] TRACE 2020/05/19 11:58:03 write 14b3afc0-ad1b-11e4-baab-0002a5d6b15d 028b004e1d001d1000005f01040000000000
[load] DEBUG 2020/05/19 11:58:05 main grid power: -4817.7W
[nrgbt] TRACE 2020/05/19 11:58:07 read fd005380-b065-11e4-9ce2-0002a5d6b15d 00000000000000000000138a0039000000000374
[nrgbt] TRACE 2020/05/19 11:58:07 power: {TotalPower:0 L1:0 L2:0 L3:0 PeakPower:0 Frequency:5002 Temperature:57 RemainingDistance:0 Costs:0 CPSignal:3}
[load] DEBUG 2020/05/19 11:58:07 main charge power: 0.0W
[nrgbt] TRACE 2020/05/19 11:58:09 read fd005380-b065-11e4-9ce2-0002a5d6b15d 00000000000000000000138a0039000000000374
[nrgbt] TRACE 2020/05/19 11:58:09 power: {TotalPower:0 L1:0 L2:0 L3:0 PeakPower:0 Frequency:5002 Temperature:57 RemainingDistance:0 Costs:0 CPSignal:3}
[load] DEBUG 2020/05/19 11:58:09 main charger status: B
[nrgbt] TRACE 2020/05/19 11:58:11 read 8f75bba0-c903-11e4-9fe8-0002a5d6b15d 06001d1000005f0001200400c1
[nrgbt] TRACE 2020/05/19 11:58:11 info: {Current:6 KWhPer100:29 AmountPerKWh:16 FIEnabled:0 ErrorCode:0 Efficiency:95 ChargingActive:false PauseCharging:true ChargingCurrentMax:32 BLETransmissionPower:4 Pad:0}
[load] DEBUG 2020/05/19 11:58:11 main grid power: -4815.3W
[nrgbt] TRACE 2020/05/19 11:58:13 read fd005380-b065-11e4-9ce2-0002a5d6b15d 000000000000000000001389003900000000036c
[nrgbt] TRACE 2020/05/19 11:58:13 power: {TotalPower:0 L1:0 L2:0 L3:0 PeakPower:0 Frequency:5001 Temperature:57 RemainingDistance:0 Costs:0 CPSignal:3}
[load] DEBUG 2020/05/19 11:58:13 main charge power: 0.0W
[nrgbt] TRACE 2020/05/19 11:58:15 read fd005380-b065-11e4-9ce2-0002a5d6b15d 000000000000000000001389003900000000036c
[nrgbt] TRACE 2020/05/19 11:58:15 power: {TotalPower:0 L1:0 L2:0 L3:0 PeakPower:0 Frequency:5001 Temperature:57 RemainingDistance:0 Costs:0 CPSignal:3}
[load] DEBUG 2020/05/19 11:58:15 main charger status: B
[nrgbt] TRACE 2020/05/19 11:58:18 read 8f75bba0-c903-11e4-9fe8-0002a5d6b15d 06001d1000005f0001200400c1
[nrgbt] TRACE 2020/05/19 11:58:18 info: {Current:6 KWhPer100:29 AmountPerKWh:16 FIEnabled:0 ErrorCode:0 Efficiency:95 ChargingActive:false PauseCharging:true ChargingCurrentMax:32 BLETransmissionPower:4 Pad:0}
[load] DEBUG 2020/05/19 11:58:18 main grid power: -4810.7W
[nrgbt] TRACE 2020/05/19 11:58:20 read fd005380-b065-11e4-9ce2-0002a5d6b15d 000000000000000000001389003900000000023f
[nrgbt] TRACE 2020/05/19 11:58:20 power: {TotalPower:0 L1:0 L2:0 L3:0 PeakPower:0 Frequency:5001 Temperature:57 RemainingDistance:0 Costs:0 CPSignal:2}
[load] DEBUG 2020/05/19 11:58:20 main charge power: 0.0W
[nrgbt] TRACE 2020/05/19 11:58:22 read fd005380-b065-11e4-9ce2-0002a5d6b15d 0000000000000000000013890039000000000364
[nrgbt] TRACE 2020/05/19 11:58:22 power: {TotalPower:0 L1:0 L2:0 L3:0 PeakPower:0 Frequency:5001 Temperature:57 RemainingDistance:0 Costs:0 CPSignal:3}
[load] DEBUG 2020/05/19 11:58:22 main charger status: B
[nrgbt] TRACE 2020/05/19 11:58:24 read 8f75bba0-c903-11e4-9fe8-0002a5d6b15d 06001d1000005f0001200400c1
[nrgbt] TRACE 2020/05/19 11:58:24 info: {Current:6 KWhPer100:29 AmountPerKWh:16 FIEnabled:0 ErrorCode:0 Efficiency:95 ChargingActive:false PauseCharging:true ChargingCurrentMax:32 BLETransmissionPower:4 Pad:0}
[load] DEBUG 2020/05/19 11:58:24 main grid power: -4812.4W
[... cut ...]
[load] DEBUG 2020/05/19 11:59:03 main grid power: -4803.8W
[nrgbt] TRACE 2020/05/19 11:59:05 read fd005380-b065-11e4-9ce2-0002a5d6b15d 0000000000000000000013890039000000000364
[nrgbt] TRACE 2020/05/19 11:59:05 power: {TotalPower:0 L1:0 L2:0 L3:0 PeakPower:0 Frequency:5001 Temperature:57 RemainingDistance:0 Costs:0 CPSignal:3}
[load] DEBUG 2020/05/19 11:59:05 main charge power: 0.0W
^Csignal: interrupt
pi@evcc:~/evcc $
```

Geht ganz kurz auf Laden und wird dann pausiert. Besser geht es da glaub ich auch erstmal nicht.
Merke: Reboot tut gut.

andig, 2020-05-19:
Danke fürs unermüdliche Testen!

closed 2020-05-19 by andig