evcc-io/evcc #169: evcc startet keinen zweiten Ladevorgang mit evse-wifi

schenlap, 2020-05-18:
**Describe the bug**
Nach einem erfolgreichen Ladevorgang mit Sofortladen mit evse-wifi und dann abstecken - fahren - anstecken wird der Ladevorgang nicht neu gestartet. Es muß off und dann wieder Sofortladen gedrückt werden für einen neuen Ladevorgang

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Sofortladen, Fahrzeug beginnt zu laden `"vehicleState":3,"evseState":true`
2. Warte bis Fahrzeug geladen `"vehicleState":2,"evseState":false`
3. Abstecken und fahren `"vehicleState":1,"evseState":false`
4. Anstecken `"vehicleState":2,"evseState":false`
5. Ladegerät bleibt in Zustand 2 (B) und geht nicht mehr in 3 (C)
6. Auch Stopp und wieder Sofortladen starten keinen Ladevorgang

**Expected behavior**
Ladegerät geht in 3 (C)

**EVCC version:**
Show output of `evcc -v`
Commit: d974028fce450c2caed5b4b8051d5fcef36b5770

**Additional context**
LogFile gekürzt um nur die relevanten Daten anzuzeigen, alle Zeilen von http, volkszaehler und soc entfernt. Redundante Ausgaben von evse wifi entfernt. Ich habe TotalEnergy, ChargedEnergy  und  Currents mitcompiliert (Kommentarzeichen entfernt)
Mode = Sofortladen.
```
[load] DEBUG 2020/05/18 20:41:43 main grid power: 1192.8W
[load] DEBUG 2020/05/18 20:41:43 main pv power: 7.0W
[load] DEBUG 2020/05/18 20:41:43 main charge power: 1199.8W
{"type":"parameters","list":[{"vehicleState":3,"evseState":true,"maxCurrent":16,"actualCurrent":16,"actualPower":0,"duration":710159,"alwaysActive":false,"lastActionUser":"API","lastActionUID":"API","energy":0.59,"mileage":3.9,"meterReading":0.59,"currentP1":0,"currentP2":0,"currentP3":0}]}
[load] DEBUG 2020/05/18 20:41:44 main charger status: C
[load] DEBUG 2020/05/18 20:41:53 main grid power: 297.9W
[load] DEBUG 2020/05/18 20:41:53 main pv power: 7.0W
[load] DEBUG 2020/05/18 20:41:53 main charge power: 1199.8W
{"type":"parameters","list":[{"vehicleState":3,"evseState":true,"maxCurrent":16,"actualCurrent":16,"actualPower":0,"duration":720192,"alwaysActive":false,"lastActionUser":"API","lastActionUID":"API","energy":0.59,"mileage":3.9,"meterReading":0.59,"currentP1":0,"currentP2":0,"currentP3":0}]}
[load] DEBUG 2020/05/18 20:41:54 main charger status: C
[load] DEBUG 2020/05/18 20:42:03 main grid power: 297.9W
[load] DEBUG 2020/05/18 20:42:03 main pv power: 7.0W
[load] DEBUG 2020/05/18 20:42:03 main charge power: 304.9W
{"type":"parameters","list":[{"vehicleState":2,"evseState":false,"maxCurrent":16,"actualCurrent":16,"actualPower":0,"duration":0,"alwaysActive":false,"lastActionUser":"API","lastActionUID":"API","energy":0,"mileage":0,"meterReading":0,"currentP1":0,"currentP2":0,"currentP3":0}]}
[load] DEBUG 2020/05/18 20:42:04 main charger status: B
[load] INFO 2020/05/18 20:42:04 main stop charging <-
{"type":"parameters","list":[{"vehicleState":2,"evseState":false,"maxCurrent":16,"actualCurrent":16,"actualPower":0,"duration":0,"alwaysActive":false,"lastActionUser":"API","lastActionUID":"API","energy":0,"mileage":0,"meterReading":0,"currentP1":0,"currentP2":0,"currentP3":0}]}
[load] DEBUG 2020/05/18 20:42:13 main grid power: 297.9W
[load] DEBUG 2020/05/18 20:42:13 main pv power: 7.0W
[load] DEBUG 2020/05/18 20:42:13 main charge power: 0.0W
{"type":"parameters","list":[{"vehicleState":2,"evseState":false,"maxCurrent":16,"actualCurrent":16,"actualPower":0,"duration":0,"alwaysActive":false,"lastActionUser":"API","lastActionUID":"API","energy":0,"mileage":0,"meterReading":0,"currentP1":0,"currentP2":0,"currentP3":0}]}
[load] DEBUG 2020/05/18 20:42:14 main charger status: B
...
{"type":"parameters","list":[{"vehicleState":2,"evseState":false,"maxCurrent":16,"actualCurrent":16,"actualPower":0,"duration":0,"alwaysActive":false,"lastActionUser":"API","lastActionUID":"API","energy":0,"mileage":0,"meterReading":0,"currentP1":0,"currentP2":0,"currentP3":0}]}
[load] DEBUG 2020/05/18 20:43:14 main charger status: B
{"type":"parameters","list":[{"vehicleState":2,"evseState":false,"maxCurrent":16,"actualCurrent":16,"actualPower":0,"duration":0,"alwaysActive":false,"lastActionUser":"API","lastActionUID":"API","energy":0,"mileage":0,"meterReading":0,"currentP1":0,"currentP2":0,"currentP3":0}]}
[load] DEBUG 2020/05/18 20:43:23 main grid power: 383.6W
[load] DEBUG 2020/05/18 20:43:23 main pv power: 7.0W
[load] DEBUG 2020/05/18 20:43:23 main charge power: 0.0W
{"type":"parameters","list":[{"vehicleState":1,"evseState":false,"maxCurrent":16,"actualCurrent":16,"actualPower":0,"duration":0,"alwaysActive":false,"lastActionUser":"API","lastActionUID":"API","energy":0,"mileage":0,"meterReading":0,"currentP1":0,"currentP2":0,"currentP3":0}]}
---> Fahrzeug voll geladen
[load] DEBUG 2020/05/18 20:43:24 main charger status: A
[load] INFO 2020/05/18 20:43:24 main car disconnected
[load] DEBUG 2020/05/18 20:43:24 main set charge current: 6A
S0_set current to given value
[load] DEBUG 2020/05/18 20:43:33 main grid power: 383.6W
[load] DEBUG 2020/05/18 20:43:33 main pv power: 7.0W
[load] DEBUG 2020/05/18 20:43:33 main charge power: 0.0W
{"type":"parameters","list":[{"vehicleState":1,"evseState":false,"maxCurrent":16,"actualCurrent":6,"actualPower":0,"duration":0,"alwaysActive":false,"lastActionUser":"API","lastActionUID":"API","energy":0,"mileage":0,"meterReading":0,"currentP1":0,"currentP2":0,"currentP3":0}]}
[load] DEBUG 2020/05/18 20:43:33 main charger status: A
[load] DEBUG 2020/05/18 20:43:43 main grid power: 383.6W
[load] DEBUG 2020/05/18 20:43:43 main pv power: 7.0W
[load] DEBUG 2020/05/18 20:43:43 main charge power: 0.0W
{"type":"parameters","list":[{"vehicleState":1,"evseState":false,"maxCurrent":16,"actualCurrent":6,"actualPower":0,"duration":0,"alwaysActive":false,"lastActionUser":"API","lastActionUID":"API","energy":0,"mileage":0,"meterReading":0,"currentP1":0,"currentP2":0,"currentP3":0}]}
[load] DEBUG 2020/05/18 20:43:44 main charger status: A
[load] DEBUG 2020/05/18 20:43:53 main grid power: 428.3W
[load] DEBUG 2020/05/18 20:43:53 main pv power: 7.0W
[... 25 lines cut ...]
[load] DEBUG 2020/05/18 20:50:13 main charge power: 0.0W
{"type":"parameters","list":[{"vehicleState":2,"evseState":false,"maxCurrent":16,"actualCurrent":16,"actualPower":0,"duration":0,"alwaysActive":false,"lastActionUser":"API","lastActionUID":"API","energy":0,"mileage":0,"meterReading":0,"currentP1":0,"currentP2":0,"currentP3":0}]}
[load] DEBUG 2020/05/18 20:50:14 main charger status: B
{"type":"parameters","list":[{"vehicleState":2,"evseState":false,"maxCurrent":16,"actualCurrent":16,"actualPower":0,"duration":0,"alwaysActive":false,"lastActionUser":"API","lastActionUID":"API","energy":0,"mileage":0,"meterReading":0,"currentP1":0,"currentP2":0,"currentP3":0}]}
---> Ladung startet nicht.
```
auch ein Stopp und wieder Sofortladen starten keinen Ladevorgang. evcc muss neu gestartet werden.

```
[load] INFO 2020/05/18 21:15:21 main set charge mode: now
[load] DEBUG 2020/05/18 21:15:21 main grid power: 365.2W
[load] DEBUG 2020/05/18 21:15:21 main pv power: 5.0W
[load] DEBUG 2020/05/18 21:15:21 main charge power: 0.0W
[wifi] TRACE 2020/05/18 21:15:21 http://10.0.0.208/getParameters
{"type":"parameters","list":[{"vehicleState":2,"evseState":false,"maxCurrent":16,"actualCurrent":6,"actualPower":0,"duration":0,"alwaysActive":false,"lastActionUser":"API","lastActionUID":"API","energy":0,"mileage":0,"meterReading":0,"currentP1":0,"currentP2":0,"currentP3":0}]}
[load] DEBUG 2020/05/18 21:15:21 main charger status: B
[load] DEBUG 2020/05/18 21:15:21 main set charge current: 16A
[wifi] TRACE 2020/05/18 21:15:21 http://10.0.0.208/setCurrent?current=16
S0_set current to given value
[wifi] TRACE 2020/05/18 21:15:21 http://10.0.0.208/getParameters
{"type":"parameters","list":[{"vehicleState":2,"evseState":false,"maxCurrent":16,"actualCurrent":16,"actualPower":0,"duration":0,"alwaysActive":false,"lastActionUser":"API","lastActionUID":"API","energy":0,"mileage":0,"meterReading":0,"currentP1":0,"currentP2":0,"currentP3":0}]}
[wifi] TRACE 2020/05/18 21:15:21 http://10.0.0.208/getParameters
{"type":"parameters","list":
-> evcc bleibt in diesem Zusand
```

andig, 2020-05-18:
Ich kann nur raten: die EVSE fühlt sich nur einmal "active" fällt danach wieder in den Dornröschenschlaf, will also nach jedem Ladevorgang wieder neu wachgeküsst werden. Du kannst mir gerne noch das komplette Log schicken, aber das bekommt mir plausibel. Hier https://github.com/andig/evcc/pull/170 wäre ein Fix.

andig, 2020-05-18:
Eine andere Lösung wäre #124, #170 ist einfacher und weniger invasiv.

andig, 2020-05-19:
@schenlap bitte probier mal die #172. Lässt sich zwar noch nicht mergen, sollte der EVSE aber einen kleinen "Kick" geben damit sie auch einen zweiten Ladevorgang startet.

closed 2020-05-19 by andig

schenlap, 2020-05-19:
#172 behebt das Problem. Danke dir!
Ist schon gemerged, ich wollte aber dennoch Rückmeldung geben

andig, 2020-05-19:
Klasse, danke!