evcc-io/evcc #175: Docker help on Synology NAS

detunedguitar, 2020-05-20:
Hi, I tried multiple times to install evcc on Docker as specified in the documentation. The container won‘t start up after configuration and <uneven cannot see why it shuts down immediately with code 0. I tried nearly everything. Is there a hint how to get it running? 

This may also be helpful to others, so I raised an issue here.

Thx in advance!

andig, 2020-05-20:
Could you please use the issue template? Specifically, what is the output of

    evcc -v

Please also show the docker logs.

detunedguitar, 2020-05-20:
Thanks for the fast response. Think I still have to learn a lot - and doing so by reading a lot of basic tutorials.

Output of evcc -v is:
evcc version ffffa2c (ffffa2c)

The Docker GUI log says just the container has started, nothing more.

andig, 2020-05-20:
Please disable the pushover config- seems you‘ve not configured it correctly.

detunedguitar, 2020-05-20:
Was also my guess ... Seems that its working, although I have to try some config as my 11kW Eco 2.0s only shows up with 1p and the meters deliver no values (Kostal Plenticore). 

One more thing: When ssh'ing the 2nd run command, it stucks for a very long time - no response. I can then manually turn on the container in DSM. Is this suspicious, that the config could be wrong?

Thank you!

andig, 2020-05-21:
It is hard to  as your issue is lacking any relevant information. Please provide steps to reproduce, console commands, log files.

detunedguitar, 2020-05-22:
I am reading some basic tutorials to no bother you with basic stuff. Right now, a basic setup runs, so I'll come back to that later. Setup I could provide information or a sampe file later:

Kostal Plenticore WR (+ KSEM)
Wallbe Eco 2.0s
BMW i3

Thank you very much for the support so far!

andig, 2020-05-22:
You‘re welcome to use the Readme, the wiki, check evcc.dist.yaml or join the slack user group (linked in the wiki). As your issue seems to be resolved I‘m closing here.

Just to be clear: there is no need to use Docker if you don‘t like it!

closed 2020-05-22 by andig