evcc-io/evcc #177: Websockets erzeugt unmenge an Warnungen

schenlap, 2020-05-21:
**Describe the bug**
Teilweise werden im Browser sekündlich Warnungen erzeugt:
`[ws] WARN 2020/05/21 09:13:51 invalid: {"version":"0.3","data":{"uuid":"aaaaaaaaaaa-xxxx-yyyy-zzzz-bbbbbbbbbbb","tuples":[[1590045231021,355,1],[1590045231163,355,1]]}}`
Die Warnungen kommen von [provider/socket.go#L111](https://github.com/andig/evcc/blob/0ee51d90fd5d39c2be0fbfe7628c1a452d36ca92/provider/socket.go#L111)
Die Warnungen können z.B am Handy die gesamte UI Seite überdecken und so nicht mehr mehr bedienbar machen. Sobald mehrere websockets abgerufen werden, macht jedes querry welches eine andere uuid sucht die Warnung. Ich habe noch nicht verstanden warum diese Warnungen nicht immer im Browser erscheinen. Es scheint aber damit zusammenzuhängen wenn evcc auf einem anderen Gerät läuft als der Browser, nur dann werden sie angezeigt. Das ist aber noch nicht ganz sicher.

**To Reproduce**
Steps to reproduce the behavior:
1. 2 websockets hinzufügen in evcc.yaml
2. evcc starten mit -l trace -> da sieht man die Warnungen immer
3. <evccIP:7070> von einem anderen Gerät aufrufen
4. Warnungen im Browser werden sekündlich angezeigt

**Expected behavior**
jq querry darf an diesem Punkt scheitern, es soll aber keine Warnung angezeigt werden.

**EVCC details**
```
$ ./evcc_amd -v
evcc version dev (HEAD)
```

BTW: ich habe mir die Datenkommunikation mit wireshark angeschaut. Wenn zwei websockets eingetragen werden (mit selber uri), dann kommen auch doppelt so viele Daten von vzlogger zu evcc. Websockets werden also tatsächlich 2 mal gesendet. Das wäre nicht notwendig, stört aber auch nicht.

andig, 2020-05-21:
Die Abfrage sollte sicher stellen, dass exakt ein Wert gefunden wird damit nicht aus versehen mit falschen Daten gerechnet wird.

Wie sieht Dein Query aus?

schenlap, 2020-05-21:
Ja, das ist mir klar. Es soll aber keine Warnung ausgegeben werden wenn eine andere uuid überprüft wurde. Query ist wie im readme: `.data | select(.uuid=="<uuid>") .tuples[0][1]`

andig, 2020-05-21:
Fixed by https://github.com/andig/evcc/commit/2d64086086189f9fca4dd0acd8e3a18bb6de0dc3. Der Check war Quatsch, danke für die Meldung.

closed 2020-05-21 by andig