evcc-io/evcc #178: Help wanted: New UI for EVCC

andig, 2020-05-21:
To further improve EVCC we need a new UI that allows to support multiple loadpoints.

If anyone has experience in UI design and Bootstrap help would be highly appreciated. Further discussion on Slack: https://join.slack.com/t/evccgroup/shared_invite/zt-dwrjvn3p-45tR~StFP82aoNDTU1Ofsg

TheRealKerni, 2020-05-21:
Working on it. Additional support is more than welcome. Using Sketch for now, so collaboration is possible.

andig, 2020-05-21:
Added a first intermediate UI to support further refactoring while @TheRealKerni is working on the final UI.

andig, 2020-05-24:
Here's the current iteration:

[image]

andig, 2020-05-24:
...and showing to loadpoints:

[image]

andig, 2020-05-24:
...and in responsive mode:

[image]

premultiply, 2020-05-24:
Sehr minimalistisch.

Es fehlen noch folgende Anzeigen aus meiner Sicht:
- Unter Home „Verbrauch“
- Beim Ladepunkt der vorgegebene max. Ladestrom sowie die Anzahl der verwendeten Phasen
- Beim Ladepunkt eine Unterscheidung zwischen „verbunden“ und „lädt“.

andig, 2020-05-29:
> Unter Home „Verbrauch“

Hausverbrauch? Ließe sich machen.

> Beim Ladepunkt der vorgegebene max. Ladestrom sowie die Anzahl der verwendeten Phasen

Phasendisplay würde ich parken für Lastmanagement, kommt. Bisher gibts nur summarische Betrachtung.

> Beim Ladepunkt eine Unterscheidung zwischen „verbunden“ und „lädt“.

Ist aktuell über Schraffur gelöst. Auf Animation wollte ich verzichten :O

andig, 2020-06-19:
Danke für die Diskussion. Ich schließe bis die Implementierung fertig ist.

closed 2020-06-19 by andig