evcc-io/evcc #189: Public REST and/or mqtt for go-e Charger

ttjaden, 2020-06-08:
Now that everything works, I wanted to update the wiki but then i recognized that I'm unsure if and how the go-e Charger can be used additional to direct local IP. 

**Is your feature request related to a problem? Please describe.**
go-e Charger is directly used with IP address. If evcc (as in my case) runs on a web-server I have to open a port to expose the go-e Charger to the Internet and use a dyndns service (in my case myfritz) to have an URI for the yaml-Settings.  

**Describe the solution you'd like**
It would be great to use the go-e Charger alternatively over public REST or mqtt. I attached the API documentation.
[go-e_api.pdf](https://github.com/andig/evcc/files/4747223/go-e_api.pdf)

Cloud mqtt: wss://i8p7v0.messaging.internetofthings.ibmcloud.com
Cloud REST: https://api.go-e.co/

I would prefer the REST API because it uses the same token than the android app.
 
**Describe alternatives you've considered**
Leave it as it is. But the go-e charger now is exposed directly to the internet without token or auth.

andig, 2020-06-08:
Warum hängt Dein Charger im Internet nur um auf einen Port zuzugreifen? Das ist wirklich keine gute Idee und sollte auch nicht notwendig sein?

ttjaden, 2020-06-08:
go-e Charger ist im Wlan und evcc hoste ich auf einem öffentlichen Server. Nach dem Wiki, muss ich einen go-e Charger so konfigurieren, dass ich die direkte IP angebe. 
```
chargers:
- name: go
  type: go-e
  uri: 192.168.0.4
```
Das geht ja nicht, da ich evcc auf meinem Server (Standort Frankfurt, öffentliche IP) ja nicht die lokale IP meines go-Charger erreichen kann. Daher bei mir wie folgt:
```
chargers:
- name: go
  type: go-e
  uri: http://abcdefghifjklmop.myfritz.net:3000 
```
Anders rum gefragt: Wie kann die Konfiguration denn aussehen, wenn ich den go-e Charger über die offizielle REST API einbinde? Da müsste ja noch der Token dazu:
```
chargers:
- name: go
  type: go-e
  token: abc1234567 #10-stelliger Token
  uri: https://api.go-e.co/ 
```

andig, 2020-06-08:
Ok, Szenario verstanden. Im Prinzip kein grosses Problem.

ttjaden, 2020-06-08:
> Ok, Szenario verstanden. Im Prinzip kein grosses Problem. Leider ist ausser dem Token auch die Struktur des Apis in Teilen anders.

Alternativ gibt es ja noch das Cloud mqtt, was von der Antwortstrukur (zumindest bei status) ja gleich ist wie bei lokaler Nutzung. Gegen Cloud mqtt spricht aber, dass man als Nutzer sich explizit als Einzelanfrage an den Hersteller werden muss, um einen "MQTT_AUTH" Passwort zu bekommen.

Daher ist die Umsetzung der REST API wohl wirklich die bessere Lösung. Falls man dir dafür mal einen Kaffee oder ein Bier spendieren kann, dann lass mich gerne wissen wie ;-)

PR #190 (2020-06-18): Support go-e cloud api
Fix #189
closed 2020-06-18 by andig