evcc-io/evcc #192: Regelverhalten im PV Mode

schenlap, 2020-06-13:
**Describe the bug**
Im PV Mode hat heute die Regelung nicht ganz sauber funktioniert. Siehe Screenshots
Da die Einspeiseleistung über dem Bezug ist (mein Smart Meter missst das getrennt wenn es auf unterschiedlichen Phasen stattfindet) wird effektiv Leistung eingespeist. Dennnoch regelt evcc zurück, was nicht hätte sein müssen. Im nächsten Schritt wird der Ladestrom wieder erhöht. Der Ladestrom schwingt mit jedem Regelschritt hoch und dann wieder runter-
Die angezeigt Einspeiseleistung verringert sich bei erhöhen der Ladeleistung da die Wallbox und die PV Anlage auf der selben Phase hängen. Ausschlaggebend ist die Differenz der beiden.

**To Reproduce**
Steps to reproduce the behavior:
Leider nicht bekannt.

**Expected behavior**
Da sich der Bezug nicht ändert hätte die Ladeleistung konstant bleiben sollen.

**EVCC details:**
[image]
[image]

schenlap, 2020-06-13:
Meine Version (Branch feature/multiple-loadpoints-core) 29d43789715659d8bafd50800ef591056ad4ac30

andig, 2020-06-13:
Wie ist Deine enable/disable Konfiguration? Es kann schnell passieren dass man da bei Fehlkonfiguration ein Schwingungsverhalten einbaut.

schenlap, 2020-06-13:
```
loadpoints:
- name: main # name for logging
  vehicle: meins
  charger: evsewifi
  enable: # pv mode enable behavior
    delay: 2m # threshold must be exceeded for this long, 2m
    threshold: -1400 # minimum export power (W). If zero, export must exceeds minimum charge power to enable
  disable: # pv mode disable behavior
    delay: 2m # threshold must be exceeded for this long, 2m
    threshold: 300 # maximum import power
  sensitivity: 1 # current raise/lower steps size (default 1A)
  guardduration: 20s # switch charger contactor not more often than this (default 10m)
  maxcurrent: 16 # maximum charge current (default 16A)
  phases: 1 # ev phases (default 3)
```

schenlap, 2020-06-13:
und ähnliches ist auch im MIN+PV Modus aufgetreten. Es könnte auch mit +1A geladen werden.
[image]
[image]
[image]

andig, 2020-06-13:
Kannst Du nachvollziehen ob irgendwo die Berechnung im Logfile nicht stimmt? Falls diese korrekt wäre muss der Fehler irgendwo in den Messwerten liegen?

premultiply, 2020-06-13:
Ich sage hier immer vorsicht, denn hier werden in der Berechnung eigentlich zwei Dinge miteinander vermischt. Die CP-Vorgabe ist die maximale Scheinleistung (VA) die das Auto ziehen darf, die Meter liefern aber Standardmäßig die Wirkleistung (W). Je nach Ladesystem und Leistung gibt es da massive Diskrepanzen. Theoretisch müsste man aus den Metern auch immer die Scheinleistung auslesen um die Regelung genauer zu machen.

andig, 2020-06-13:
Das erklärt nicht warum wir hier ein schwingen sehen würden? Auslesen der Scheinleistung hilft m.e. Nicht, da mit zum EVU auch nur die Wirkleistung interessiert?

premultiply, 2020-06-13:
Mit dem schwingen hat das nichts zu tun, richtig. Schwingen kommt eigentlich immer von irgendeiner nicht berücksichtigten Zeitkonstante  im Regelkreis oder einer falschen Messung

schenlap, 2020-06-14:
`Kannst Du nachvollziehen ob irgendwo die Berechnung im Logfile nicht stimmt? Falls diese korrekt wäre muss der Fehler irgendwo in den Messwerten liegen?`

Die angezeigten Messwerte und die aus dem Log stimmen meiner Meinung nach zusammen.

Ich glaube ich habe eine mögliche Ursache gefunden. _Main Charge Power_ hat eine Abweichung zur erwarteten Ladeleistung (welche aus Stromvorgabe * Phasen * Spannung) berechnet wird. evcc geht davon aus dass die Ladeleistung exakt dieser Berechnung folgt. Es wird da aber immer kleine Abweichungen geben. Ich kann auch nicht sagen wie genau die Ladegeräte im Auto die Leistung einstellen, kann auch über die PWM-Signal-Erzeugung ein kleiner Fehler passieren, wird eigentlich Scheinleistung vorgegeben (laut premultiply)

target power = charge power - grid power (laut log). Die Target Power wird dann geteilt durch (Spannung * Phasen) und dass gibt die Stromvorgabe. In diesem Fall wäre es besser wenn einfach nur die Grid-Power genommen wird und diese geteilt durch (Spannung * Phase) und das zur aktuellen Stromvorgabe addiert wird. Es würde also nur die Differnez berechnet. Dadurch wäre man nicht davon abhängig wie genau die Ladeleistung eingestellt wird, da die sowieso gemessen wird.

Ich bin mir aber nicht sicher ob da nicht in meiner Wallbox was suboptimal läuft. z.B bei 8 A ist charge power = 1670, was 7.3A entspricht. Laut @premultiply ist CP eine Scheinleistungsvorgabe, für die Regelung interessant ist aber die Wirkleistung. Das würde auch für meine vorgeschlagene Berechnung sprechen. Ich werde die nächsten Tage noch meine Wallbox genau testen. Bis ich da ein Ergebnis habe würde ich Vorschlagen noch nichts umzusetzen.

Dass die Site Power lange < -230W ist und evcc nicht nachregelt ist sicher ein Bug. Ich gehe davon aus dass mit einer Korrektur auch das Schwingen aus dem ersten Bild behoben wird.

Habt ihr auch schon beobachtet dass eine Regelabweichung von > 230W stehen bleibt, oder ist das nur bei mir?

schenlap, 2020-06-15:
Ich habe jetzt die Wirkleistung vs. Ladestrom gemessen. Die Ladeleistung aus evse habe ich mit den Daten vom Smartmeter verglichen und das paßt sehr gut zusammen, denke daher dass die Messung richtig ist.

[image]

Ergibt einen cos phi = 0.9, das wirkt sich natürlich auf die derzeitige Berechnung in evcc aus.

andig, 2020-06-15:
Die Idee das Delta zu berechnen statt des neuen Gesamtstromes und dabei nur auf Grid, aber nicht auf Charge Power zu setzen gefällt mir gut. Ich muss nochmal überlegen wie sich das darstellen würde wenn nur ein PV Zähler verbaut ist, aber der Ansatz kommt auf die Liste!

andig, 2020-06-19:
> target power = charge power - grid power (laut log). Die Target Power wird dann geteilt durch (Spannung * Phasen) und dass gibt die Stromvorgabe. In diesem Fall wäre es besser wenn einfach nur die [...]

Ich musste eine Weile nachdenken, aber: das dürfte die Ursache sein. Zwar wird die Charge Power gemessen, sie wird aktuell für den Zielstrom aber ignoriert. Bis zur verfügbaren Leistung ist noch alles richtig (da wird die gemessene Ladeleistung angewandt), danach aber aus der Gesamt verfügbaren Leistung auf den Gesamtstrom gerechnet und damit steigt die Fehlerquote bei höherem Blindleistungsanteil.

Ich hatte schon ein paarmal gesehen, dass die Regelung es nicht geschafft hat, die verfügbare Leistung am Netzpunkt auszuschöpfen- das könnte die gleiche Ursache haben. Lösung dafür sollte mit v2 kommen!

schenlap, 2020-07-19:
Ich verlinke es hier auch noch.
#223

PR #223 (2020-07-19): Ignore current charge power when calculating delta current to reduce …
…impact of apparent vs active power differences

Fix #192
closed 2020-07-19 by andig