evcc-io/evcc #193: weiteres Regelziel: maximale Netznutzung

tmeidinger, 2020-06-13:
Um eine Anschluß-Überlastung oder ggf. Strafzahlung zu verhindern wäre eine Ausregelung auf den gesamten Netzbezug schön. D.h. wenn z.B. die Wärmepumpe/Geschirrspülmaschine/etc. anläuft wird automatisch die Ladeleistung reduziert und ggf. später wieder nach oben gesetzt.

andig, 2020-06-13:
@tmeidinger das ist Kern der Funktion von evcc ("PV" Modus)- wo siehst Du das aktuell nicht erfüllt?

tmeidinger, 2020-06-13:
Ich meine maximale Netznutzung (selbst wenn es keine PV gibt). Der monatliche Anschlusspreis wird bei immer mehr EVU (AT und CH) über die maximale Monatsbezugsleistung festgelegt. Da wäre es schön wenn man das ausregeln könnte. Technisch könnte man es über PV-Modus lösen und ein Meter definieren bei dem man von einer gewünschten max. Leistung (z.B. 15kW) den gesamten Netzbezug abzieht und darüber die Ladeleistung ausregelt.

andig, 2020-06-13:
Hast Du mal überlegt, ob sich das nicht mit enable/disable Einstellungen lösen liesse? Falls nein- was fehlt? V.a. wie sollte sich das darstellen? Als neuer Modus? Oder Teil von „pv“?

tmeidinger, 2020-06-13:
Ich hab mal folgendes Meter angelegt:

meters:
- name: shelly3em
  type: default
  power:
    type: calc
    add:
    - type: script
      cmd: /bin/sh -c "echo 15000"
      timeout: 5s
    - type: http
      uri: http://shelly/emeter/0
      method: GET
      headers:
      - content-type: application/json
      jq: .power
      scale: -1
    - type: http
      uri: http://shelly/emeter/1
      method: GET
      headers:
      - content-type: application/json
      jq: .power
      scale: -1
    - type: http
      uri: http://shelly/emeter/2
      method: GET
      headers:
      - content-type: application/json
      jq: .power
      scale: -1

Dazu folgenden Loadpoint:

loadpoints:
- name: home 
  vehicle: bmw
  charger: go-e
  meters:
    pv: shelly3em
  enable:
    delay: 1m
    threshold: 4200
  disable:
    delay: 1m
    threshold: 1
  guardduration: 10m
  mincurrent: 0
  maxcurrent: 16
  sensitivity: 1
  phases: 3

Das sollte von der Logik her hinkommen, das Shelly erfasst alle Verbraucher bis auf die Wallbox. Allerdings bekomme ich im GUI nur Stop/Sofortladen angezeigt.

andig, 2020-06-13:
Netzausregelung geht (logisch?!) nur mit Netzzähler- Du müsstest den Shelly also also grid eintragen, nicht als pv.

andig, 2020-06-13:
Mincurrent 0 ist vmtl falsch- oder kann Deine Wallbox/Auto mit <6A laden?

premultiply, 2020-06-13:
Was hier am Ende fehlt ist quasi eine Grundeinstellung für das Gridmeter "maximaler Netzbezug" bzw. eigentlich "Anschlussleistung" als Obergrenze (hach ist das nicht ein schöner Begriff...).

tmeidinger, 2020-06-13:
Ich denke auch, das würde es verständlicher machen.

andig, 2020-06-13:
Sollte es nicht reichen den disable threshold auf diesen Wert zu setzen? Mir ist einfach nicht klar was ihr wirklich erreichen wollt und v.a. wie das mit dem pv mode zusammen spielen soll.

premultiply, 2020-06-13:
Es hat überhaupt nichts mit dem PV- oder einem anderen Mode zu tun.
EVCC soll zusätzlich immer einen maximalen Netzbezug berücksichtigen.

Angenommen man hätte nur einen 20 kVA-Hausanschluss. Dann soll EVCC im Rahmen seiner Regelmöglichkeiten immer Sicherstellen, dass diese Schwelle nicht überschritten wird obwohl vielleicht auch temporär mal andere Verbraucher im Haus laufen. Das bedeutet also z. B. auch eine Reduzierung im Sofortladen-Modus wenn erforderlich.

andig, 2020-06-13:
D.h. dieser Request wäre egtl. „Maximalen Netzbezug in allen Modi auf einstellbaren Wert begrenzen“? Dann hab ichs verstanden :)

tmeidinger, 2020-06-13:
This!

premultiply, 2020-06-13:
Ja, ich hab es auch schon zu #150 hinzugefügt.

andig, 2020-06-13:
Perfekt, dann machen wir hier zu- danke für den Request und die Erläuterung!

closed 2020-06-13 by andig