evcc-io/evcc #194: Add CH341 driver support to read Tesla Wallbox via RS485 adapter

hopfi2k, 2020-06-15:
I'd like seeing support added for the CH341 serial RS-485 adapter that is commonly used to control the Tesla Wallbox (Tesla Wall Connector) set to 'slave' mode.

There is a quite successful project called [TWCManager](https://github.com/dracoventions/TWCManager) that allows to control a Tesla Wall Connector programmatically through a cheap (< $5) RS-485 USB serial adapter (using the CH341 driver). 
A [fork](https://github.com/ngardiner/TWCManager) of this project has added some new features but the whole thing is a bit dated...

While I have mangled through the code of the TWCManager to add support for Kostal Plenticore Inverters successfully, I'm unfortunately not familiar with programming Go, otherwise I would have contributed a charger module. Such module should enable to run **evcc** from a raspberry pi zero connected (or even put inside a Tesla Wall Connector like I did) and "convert" the original Tesla Wall Connector to an intelligent charger.

If nobody is willing to add such driver, I might need to find time to teach myself with Go and write it on my own. Please consider that this would take me endless hours while an experienced Go programmer might hack it together in a fraction of time...

Thanks!

andig, 2020-06-16:
@hopfi2k seems this will oly work for Gen2 TWC, no longer for Gen3 so it seems usefulness is limited in future. As for CH341 I don‘t think there is any support needed. EVCC will see a linux device regardless of which adapter you use, so this comes down to supporting TWC Gen2.

hopfi2k, 2020-06-16:
Correct it comes down to accessing a serial device. TWC Gen 2 have been sold for the past years and it's safe to assume that hundred thousands are installed world wide. Even newer Gen 3's have an RS-485 port, but since it supports WLAN connectivity it might not be necessary to use a RS-485 adapter to control charge limits (depending on future Tesla software upgrades). Nevertheless even Gen 3 support slave mode.
From my point of view, TWC's most probably have the broadest installation base of all chargers currently supported by EVCC. And adding the RS_485 adapter (internally or externally) and putting the rotary switch to position 'F' (slave mode) is extremely simple. Giving the right software support I'm pretty much sure that more people would be willing to 'modify' their TWC to benefit from using it with EVCC. Worth considering?

andig, 2020-06-16:
I‘ve had a look at the protocol nd it seems pretty straightforward. Would be willing to tackle this but would need access to a twc2 for testing. If you want to grant access please pm me at cpuidle@gmx.de

andig, 2020-06-19:
For all concerned: a Protocol 2 version is necessary. Basic communication with the charger is already possible (pending debugging). We need to understand if it is possible to start/stop the charge process using the charger instead of the car API (as TWC does).

Any additional insights welcome.

closed 2020-10-24 by stale[bot]

normanu, 2025-02-04:
I would love this option aswell!
I have 3x TWC gen 2's and nowadays all different kind of cars are charging on it. (not only Tesla's)
So if they could be throttled much preffered instead of using the API's of the cars.

I also found this project, https://github.com/zagor/wallconnector