evcc-io/evcc #196: go-e charger: unsupported protocol scheme

Tscherno, 2020-06-18:
Ich versuche gerade evcc über Docker auf Synology zum Laufen zu bringen:

Dabei wird ein Fehler beim goE geworfen:
Config:
chargers:
- name: go
  type: go-e
  uri: 192.168.1.183

Fehler:
main charge controller error: All attempts fail: #1: Get 192.168.1.183/status: unsupported protocol scheme "" #2: Get 192.168.1.183/status: unsupported protocol scheme "" #3: Get 192.168.1.183/status: unsupported protocol scheme "" 

Ein Aufruf der URL im Browser bringt das erwartete JSON zurück. In Home-Assistant läuft die Einbindung auch wunderbar.

andig, 2020-06-18:
Stell bittemal ein http:// davor ;)

Tscherno, 2020-06-18:
Damn... Feature-Request: Please make configuration idiot proof :P

closed 2020-06-18 by Tscherno

andig, 2020-06-18:
fyi: siehe https://github.com/andig/evcc/pull/190 für Unterstützung des Cloud API

>  Feature-Request: Please make configuration idiot proof :P

Mach gerne neues Issue auf ;)