evcc-io/evcc #197: Kostal Plenticore, BYD, Kostal Smart Meter

Tscherno, 2020-06-18:
Hi,

ich habe die oben genannte Kombination. Mit der Verbindung zum Wechselrichter alleine klappt es. Allerdings werden ja die Verbrauchswerte offensichtlich nicht vom WR per Modbus weitergegeben, sondern es muss noch separat das Smart Meter angezapft werden. Da scheitere ich leider mit dem Versuch das anhand der Beispiele nachzubilden:

meters:
- name: pv-dc-kostal
  type: default
  power:
    type: calc
    add: # The add plugin sums up both string values
    - type: modbus
      model: sunspec
      value: 160:1:DCW # string 1
      uri: wechselrichter.fritz.box:1502 
      id: 71 # Configured Modbus Device ID 
    - type: modbus  
      value: 160:2:DCW # string 2
      uri: wechselrichter.fritz.box:1502 
      id: 71 # Configured Modbus Device ID 
- name: grid-kostal
    type: modbus # <=== line 22
    model: ksem
    uri: 192.168.1.11:502 
    id: 1 # Configured Modbus Device ID 
    power: Power
    energy: Export

===> 
While parsing config: yaml: line 22: mapping values are not allowed in this context

andig, 2020-06-18:
Ich habe mir erlaubt, Deine Config zu formatieren- da siehst Du, dass ab Line 22 die Einrückung kaputt ist.

Tscherno, 2020-06-18:
Grml, yaml... Sorry, dass ich hier so nerve. Sobald es läuft, schreib ich die vollständige Config dafür ins Wiki um es dem nächsten etwas zu erleichtern.

Tscherno, 2020-06-18:
Muss nochmal was fragen. Welche Auswirkung hat die Battery in der Konstellation? Auf der Weboberfläche scheint es keinen Unterschied zu machen. Ich hätte jetzt erwartet, dass der SOC angezeigt wird und diese irgendwie mit in die Ladelogik einbezogen wird (bsp. Lade erst BYD auf 100%, dann Überschussladung...).

Config:

meters:
- name: pv-dc-kostal
  type: default
  power:
    type: calc
    add: # The add plugin sums up both string values
    - type: modbus
      model: sunspec
      value: 160:1:DCW # string 1
      uri: wechselrichter.fritz.box:1502 
      id: 71 # Configured Modbus Device ID 
    - type: modbus  
      value: 160:2:DCW # string 2
      uri: wechselrichter.fritz.box:1502 
      id: 71 # Configured Modbus Device ID 
- name: grid-kostal
  type: modbus
  model: ksem
  uri: 192.168.1.11:502 
  id: 1 # Configured Modbus Device ID 
  power: Power
  energy: Export
- name: battery-kostal
  type: modbus
  power: 802:W
  uri: wechselrichter.fritz.box:1502 
  id: 71 # Configured Modbus Device ID ("Gerät")

loadpoints:
- name: main # name for logging
  vehicle: bmw
  charger: go # charger
  meters:
    grid: grid-kostal # grid meter
    pv: pv-dc-kostal # pv meter
    battery: battery-kostal # battery meter
    #charge: charge # charge meter
  enable: # pv mode enable behavior
    delay: 1m # threshold must be exceeded for this long
    threshold: 0 # minimum export power (W). If zero, export must exceeds minimum charge power to enable
  disable: # pv mode disable behavior
    delay: 5m # threshold must be exceeded for this long
    threshold: 200 # maximum import power (W)
  guardduration: 10m # switch charger contactor not more often than this (default 10m)
  mincurrent: 6 # minimum charge current (default 6A)
  maxcurrent: 16 # maximum charge current (default 16A)
  phases: 3 # ev phases (default 3)
  sensitivity: 1 # current raise/lower step size (default 10A)

andig, 2020-06-18:
Kannst dus bitte formatieren (markdown)?

Tscherno, 2020-06-18:
Ich hoffe das klappt so:

```
chargers:
- name: go
  type: go-e
  uri: http://192.168.1.183

meters:
- name: pv-dc-kostal
  type: default
  power:
    type: calc
    add: # The add plugin sums up both string values
    - type: modbus
      model: sunspec
      value: 160:1:DCW # string 1
      uri: wechselrichter.fritz.box:1502 
      id: 71 # Configured Modbus Device ID 
    - type: modbus  
      value: 160:2:DCW # string 2
      uri: wechselrichter.fritz.box:1502 
      id: 71 # Configured Modbus Device ID 
- name: grid-kostal
  type: modbus
  model: ksem
  uri: 192.168.1.11:502 
  id: 1 # Configured Modbus Device ID 
  power: Power
  energy: Export
- name: battery-kostal
  type: modbus
  power: 802:W
  uri: wechselrichter.fritz.box:1502 
  id: 71 # Configured Modbus Device ID ("Gerät")

vehicles:
- name: bmw
  type: bmw
  title: i3
  capacity: 65 # kWh
  user: ...
  password: ...
  vin: WBY...
  cache: 5m
      
loadpoints:
- name: main # name for logging
  vehicle: bmw
  charger: go # charger
  meters:
    grid: grid-kostal # grid meter
    pv: pv-dc-kostal # pv meter
    battery: battery-kostal # battery meter
    #charge: charge # charge meter
  enable: # pv mode enable behavior
    delay: 1m # threshold must be exceeded for this long
    threshold: 0 # minimum export power (W). If zero, export must exceeds minimum charge power to enable
  disable: # pv mode disable behavior
    delay: 5m # threshold must be exceeded for this long
    threshold: 200 # maximum import power (W)
  guardduration: 10m # switch charger contactor not more often than this (default 10m)
  mincurrent: 6 # minimum charge current (default 6A)
  maxcurrent: 16 # maximum charge current (default 16A)
  phases: 3 # ev phases (default 3)
  sensitivity: 1 # current raise/lower step size (default 10A)

mfueser, 2020-06-19:
Moin, moin! Ich habe extakt die gleiche Konfiguration. Meine Konfig ist auch der Ursprung für die Daten im Wiki :).
Man könnte es als Beispielconfig natürlich auch am Stück eintragen. Vielleicht machen wir dafür dann eine eigene Rubrik a la "Komplette Setups", wo jeder seinen Aufbau beschreibt und die Konfiguration dazu bereitstellt.

mfueser, 2020-06-19:
> 
 
> Muss nochmal was fragen. Welche Auswirkung hat die Battery in der Konstellation? Auf der >Weboberfläche scheint es keinen Unterschied zu machen. Ich hätte jetzt erwartet, dass der SOC >angezeigt [...]

Die Hausbatterie wird aktuell noch nicht angezeigt. Ein PR mit einer neuen Oberfläche sollte das dann irgendwann mitbringen, wenn ich das richtig mitbekommen habe.
Die Rechnung sollte wie folgt aussehen (Auszug Doku):
Charge Power = Current Charge Power - Grid Meter Power + Battery Meter Power - Residual Power

Wenn also die Batterie geladen wird, so wird das als "freie Ladeleistung" interpretiert und entsprechend die Wallbox höher geregelt. Ich muss das bei mir mal debuggen, da ich den Eindruck habe, dass das nicht immer funktioniert.

andig, 2020-06-19:
Wie @mfueser schreibt. Die Batterie „siehst“ Du aktuell nur im Log (log: debug), sie wird aber berücksichtigt. Konkret bekommt sie in den PV Modi Berücksichtigung so dass statt der Hausbatterie bevorzugt das Auto geladen wird.

Tscherno, 2020-06-19:
Ok, das ist nicht wie das ganze fahren will. Bei mir hat die Batterie Priorität, solange ein gewisser SoC vom Auto nicht erreicht ist. Ich schreibe dazu mal einen Feature Request. Gibt es für die Batterie-Visualisierung schon einen Request? Falls nicht, würde ich einen separaten schreiben.

andig, 2020-06-19:
Batterievisualisierung bekommst Du automatisch. Batterie als Prio vor Auto macht m.E. keinen Sinn da Du damit schlimmstenfalls 2x Ladeverluste hast- erst in die Batterie danach aus der Batterie ins Auto. Das Du mit Auto laden ab einem bestimmten SoC aufhören möchtest wäre ein neuer Request, das wäre auch unabhängig von Batterie umsetzbar und sinnvoll.

andig, 2020-06-19:
Ich mache hier mal zu, Featurediskussion in #198

closed 2020-06-19 by andig