evcc-io/evcc #198: Ladelogik SoC Auto > SoC Batterie > Überschuss + Minmal SoC Auto

Tscherno, 2020-06-19:
Ich würde mir (für mein Nutzungsverhalten sinnvolles) Lademanagement vorstellen:
Ein Modus "Eco Min Charge Auto"
- Höchste Priortät soll ein Minimalwert vom SoC Auto haben. 
- Dieser (für mich beispielsweise 30%) soll sinnvollerweise über die Oberfläche (Schieberegler?) konfigurierbar sein.
- Bis zu diesem SoC vom Auto wird auf jeden Fall direkt geladen, damit ist sichergestellt, dass ich in die Arbeit bzw. zum Einkaufen fahren kann wo idealerweise kostenlos geladen werden kann
- Über diesen Wert, soll die Solar Batterie Prio haben. Dessen Minimum SoC soll konfigurierbar sein (reicht über Config). Erst wenn der SoC überschritten hat, wird auf Überschuss geprüft.
- Ist der SoC Batterie > Grenzwert, wird geprüft ob ein Ertragsüberschuss besteht. Dieser soll im Idealfall so verwendet werden, dass nur der Überschuss, aber nicht aus der Batterie geladen wird. 

==> Mein Ziel ist, dass ich wirklich nur den Überschuss zum Laden nutze. Überschuss ist für mich nicht, wenn die Solarbatterie geladen wird.

premultiply, 2020-06-19:
Den Minimal-SoC kannst du glaube ich in der Konfig schon einstellen bis zu dem immer direkt aufgefüllt wird. Oder verwechsle ich das gerade mit openWB? 🤔

Wenn du die Priorisierung der Fahrzeugladung nicht möchtest entferne einfach die Hausbatterie aus der Konfig dann wird nur der tatsächliche Netzüberschuss beachtet.

Ansonsten ist vielleicht das Featuremonster openWB für dich vielleicht besser geeignet.

Tscherno, 2020-06-19:
Naja mir schien das hier als frisches Projekt, das durchaus noch interessante Features einbauen will. Der Minimal-SoC würde ja hart in der Config konfiguriert. Die Idee ist, das über die Oberfläche anzubieten.

andig, 2020-06-19:
Ich denke wir reden jetzt vom PV Modus? Dann möchtest Du
- Auto bis MinSoc laden (aber nicht die Batterie)
- Dann Batterie bis deren MinSoc laden (aber nicht das Auto)
- Dann wieder das Auto voll laden
- Erst dann wieder die Batterie?

Das scheint mir nicht unbedingt ein massentauglicher Prozess zu sein? Welchem Zweck soll das dienen? Wenn Deine Hausbatterie MinSoC erreicht hat kannst Du Dein Auto immer noch woanders kostenlos laden- also wieso den Modus wieder wechseln?

Tscherno, 2020-06-19:
Ich würde bei der Hausbatterie auch 100% einstellen.

andig, 2020-06-19:
> Ich würde bei der Hausbatterie auch 100% einstellen.

Die Frage bleibt dennoch die gleiche. Warum anschließend wieder das Auto laden wenn Du das anderswo umsonst machen kannst? Dann lieber die Einspeisevergütung mitnehmen (?).

Die Motivation für MinSoC beim EV ist m.E. eher der Wunsch der Akkuschonung, also das EV nicht immer „voll“ rumstehen zu haben.

Tscherno, 2020-06-19:
Die kostenlose Ladung ist halt nicht sichergestellt. Wenn es klappt gut, wenn nicht dann halt nicht. Daher lasse ich meinen i3 bis auf ein Minimum daheim, ausser ich habe gerade PV Überschuss. Und Überschuss ist für mich, dass die Solarbatterie geladen ist. Aus der Solarbatterie ins Auto zu laden macht natürlich keinen Sinn. Aber der (finanzielle) Verlust aus dem Netz zu laden wäre IMO grösser.

andig, 2020-06-19:
Ok, verstanden. Liesse sich mittels MinSoC Parameter bei EV und Batterie lösen:
- bis MinSoC EV erreicht ist hat EV Vorrang
- bis MinSoC bei Battery erreicht ist hat Batterie Vorrang
- ab MinSoc Battery hat wieder EV Vorrang

Default Battery MinSoC: 0 bedeutet also, Auto hat immer Vorrang. 

Das Alles gilt ausschließlich im PV Modus. Für den MinPV Modus wäre separat zu überlegen, wie MinSoC fürs EV dort berücksichtigt werden sollte.

Generell müssten solche Optionen vmtl. Ladepunkt-spezifisch konfiguriert werden. Hierfür braucht es noch ein Gesamtkonzept bevor es an die Umsetzung gehen kann...

Tscherno, 2020-06-19:
MinSoC EV wäre tatsächlich eher Fahrzeug, als Ladepunktspezifisch. Man kann ja davon ausgehen, dass das Nutzungsverhalten am Auto, nicht am Ladepunkt hängt. Das wäre jetzt für mich aber kein Anwendungsfall, da nur ein E-Mobil und nur ein Ladepunkt. Wichtig wäre für mich, dass der MinSoC EV über die Oberfläche regelbar ist.

premultiply, 2020-07-08:
Das könnte ggf. auch via "Zielladen" realisiert werden, oder?

GiK90, 2020-07-17:
Ich denke die Funktion grundsätzlich wäre schon sinnvoll, nur wäre mein Einsatzzweck ein anderer. 

Der Haus-Akku sollte wenn er unter einem gewissen SoC ist bis dahin (minHomeSoC) geladen werden. 

Das Elektroauto sollte mit einer Mindestladung (minVehicleSoc) an PV Strom geladen in der Garage stehen, um am nächsten Tag den Weg in die Arbeit und wieder zurück zu schaffen. 

Die Priorität soll steuerbar sein (priority: vehicle) oder (priority: home) um zu regeln, wwas zuerst auf min gebracht wird und wenn beide auf min sind danach zuerst vollgeladen bzw. bis maxVehicleSoC im Falle des Autos geladen wird. 

Zusätzlich ist ein maximaler SoC (maxVehicleSoc) fürs Elektroauto denkbar um den Akku zu schonen. 

Beispiel für die Config (n. Rs mit andig):

minVehicleSoc: 50
maxVehicleSoc: 90
minHomeSoC: 80
priority: vehicle

Tscherno, 2020-07-17:
Was mir momentan am meisten Fehlt ist MinSoc Car <= Bis zu dem Punkt soll die Ladung einfach unabhängig von PV automatisch mit maximaler Leistung starten.

GiK90, 2020-07-17:
Das sollte aber als Einstellung machbar sein womit bis zum Minimalwert geladen werden soll.  Aus meiner Sicht auch vorrangig mit PV! und ab z. B. 20 Uhr (keine schöne Methode) dann erst aus dem Netz.

premultiply, 2020-07-17:
Wirklich? Möchtest du nicht in Wahrheit haben dass dein Fahrzeug zu einem bestimmten Zeitpunkt eine bestimmte Mindestreichweite hat?

GiK90, 2020-07-17:
Oder man löst das dann über "Min+PV", denn im "PV" soll ja eigentlich mit PV geladen werden...
Vielleicht sollte man das dann so machen, dass ab einer gewissen PV-Leistung (einstellbar z. B. 300 Watt) mit maximaler Leistung bis MinSoC vollgeaden wird.
Ich denke grad mal laut :D

premultiply, 2020-07-17:
Vielleicht wäre es eher hilfreich mal ein reale Nutzungsszenario beschreiben und sich dann erst gemeinsam zu überlegen wie dies abgebildet werden kann.

Bis jetzt sind diese ganzen furchtbar komplizierten Spezialkonstrukte nämlich meist ein Kartenhaus.

GiK90, 2020-07-17:
Bei meinem beschriebenen Verfahren wäre die tägliche Logik die, das Auto so vollzuladen, dass man auf jeden Fall auf Arbeit und wieder Heim kommt. Rest erstmal Haus, weil das über Nacht auf jeden Fall wieder leer wird. 

Und das Autp könnte auch am nächsten Tag gar vollgeladen werden ... hätte ja dann genu drin. 
Ich mal ein Ablaufdiagramm... mal sehn...

premultiply, 2020-07-17:
Dann übersetze ich mal:

Du möchtest sicherstellen dass zum einem definierten Zeitpunkt mindestens eine vorgegebene Energiemenge im Fahrzeugakku gespeichert ist.

GiK90, 2020-07-17:
Quasi. Aber erst übernacht! Ansonsten kann man ja auch einfach die Einstellung "Sofortladen" nutzen, wenn man weiß dass man es am gleichen Tag sofort nochmal braucht.

premultiply, 2020-07-17:
Das könnte man in Anlehnung an vergleichbare Projekte wie folgt implementieren:
**Zielladen** stellt sicher, dass zu einem vom Nutzer vorgegebenen Zeitpunkt die vorgegebene Energiemenge mindestens im Fahrakku bereitsteht.

Dies wird so implementiert, dass erst zum spätmöglichsten Zeitpunkt (inkl. Reservern für spezielle Ladecharakteristik) bis zur vorgegebenen Mindestmenge mit maximaler Leistung nachgeladen wird. Im Zeitraum davor wird natürlich anfallender Überschuss auch schon geladen.
- Im Idealfall wird also überhaupt nicht mehr gesondert (nach)geladen.
- Im Normalfall wahrscheinlich eine relativ kleine verbleibende Differenzenergiemenge.
- Im Worst-Case wird ausschließlich mit voller Leistung und Netzbezug geladen.

Der Unterfall Sofortladen setzt den Zielzeitpunkt auf JETZT und das Energieziel auf 100%.

Übrigens: Ob der ggf. zugekaufte Strom in den Fahrzeugakku geladen wird oder vom Haus verbraucht wird spielt überhaupt keine Rolle. Daher kann man eine Hausbatterie vollkommen aus dieser Berechnung herauslassen. Eine Hausbatterie wird immer nachgelagert nur mit PV-Überschuss geladen der nicht von der Fahrzeugladung gebraucht wird.
In einem Akku gespeicherte Energie ist nie verloren. Ob man die Energie verfährt oder im Haus verheizt spielt billanziell keine Rolle.

EVCC kümmert sich ohnehin nur um den Aspekt der Fahrzeugladung.

GiK90, 2020-07-17:
Die Idee finde ich Klasse!

Tscherno, 2020-07-17:
Die Idee mit "Ich hätte gerne zum Zeitpunkt X mindestens Y Prozent" finde ich auch super.

premultiply, 2020-07-19:
Hier der passende FR dazu: #225

premultiply, 2020-07-20:
Kann hier dann zu oder fehlt noch was?

closed 2020-09-25 by stale[bot]