evcc-io/evcc #201: Monitoring the SOC even when car is not connected

ttjaden, 2020-06-26:
**Is your feature request related to a problem? Please describe.**
There is no problem. It's only, at least for me, a very nice feature

**Describe the solution you'd like**
Deliver the SOC the the monitoring plugins, like influxdb, even if the car is not connected to the charger. Regarding the front-end of evcc the SOC should further on only be shown, if the car is connected.

**Additional context**
Regarding Smart-Home-Monitoring it would be really great to have ca consistent timeseries for all measured meters and chargers. One thin that is missing: the SOC of the car.

andig, 2020-06-26:
Agreed, should be addedß

closed 2020-07-21 by andig

ttjaden, 2020-08-07:
Is this feature already in the recent release version 0.21? 
--> I push the data to an influxDB and there is only an SOC when vehicle is connected.

andig, 2020-08-07:
https://github.com/andig/evcc/pull/228/files#diff-82287afc024e8be115336bfb52980986R222

ttjaden, 2020-08-07:
Danke!

PR #228 (2020-07-21): Allow updating soc when vehicle disconnected
Fix #201