evcc-io/evcc #205: error during installation docker on Synology

Zyro16, 2020-07-08:
Hi Andig,

thanks a lot for your work!
I want to run your docker image on my Synology.

During the installation following your wiki guid [1.2.](https://github.com/andig/evcc/wiki/1.2.-Docker-installation)

It seems that your evcc.yaml now is evcc.dist.yaml.
After execute your commands I got following error:

[image]

Can you please help, how I can run your docker image on my Synology!?

Thanks a lot and keep going! :)

Best regards from Landsberg am Lech,
Daniel

andig, 2020-07-08:
See error message- seems you have enabled messaging using pushover but not configured an app name? To get started, you could disable push messaging.

andig, 2020-07-10:
ping @Zyro16 issue resolved?

Zyro16, 2020-07-13:
Hi @andig ,

thank you but not yet.

I downloaded the code and copied the files to /EVCC folder.
After that, I use your docker command: docker run -v $(pwd)/evcc.dist.yaml:/etc/evcc.yaml -p 7070:7070 andig/evcc -h
But it seems there is an error 
[image]

Can you please help?

Thank you, best regards!
Daniel

[evcc.dist.yaml.txt](https://github.com/andig/evcc/files/4912501/evcc.dist.yaml.txt)

andig, 2020-07-13:
> But it seems there is an error

I don't see an error. You've basically run evcc -h and that worked?

> The file at /etc/evcc.yaml does not exist.

You've mounted the file into the container- I'm sure it's available there.

It seems that you really don't have any docker experience. I'd suggest to make yourself familiar with docker and- while doing so- start using evcc with the binary files provided instead of starting with docker right away ;). 

Unfortunately, providing support for basic docker usage is beyond my capacity, I'm therefore closing this issue. Please feel free to reopen when you're experiencing problems with evcc.

closed 2020-07-13 by andig