evcc-io/evcc #206: Ist das Wiki aktuell?

GiK90, 2020-07-09:
Erstmal Klasse Projekt!!! Freue mich riesig darüber! Würde auch gerne bei Tests helfen!

Habe einen 

SMA SHM 2.0 fürs Grid
Energy Meter für PV
Sunny Island Ladereglee
Go-E Charger 22 kw
Bald i3s

Im Docker auf dem NUC läuft alles Prima. Es kommen nur keine Werte an. 

Ich habe eine Frage bezüglich der Config für den SMA Sunny Home Manager 2.0. 

Ist die Beschreibung im Wiki aktuell?
Denn in der Config-Beispiel-Datei sind andere Infos enthalten, wie z. B. Der Type etc was im Wiki nicht aufgeführt ist. 

Danke und schöne Grüße

GiK90, 2020-07-09:
👎

premultiply, 2020-07-09:
Bei der SMA-Vernetzung muss du beachten, dass die Geräte per Multicast miteinander sprechen.
Insbesondere die Zähler SHM / EM senden ihre Daten per Multicast ("SMA Speedwire") aus.
Dies muss ggf. auch im Docker-Container ankommen bzw. von diesem abonniert werden können.
Ich hab das Ganze ohne Docker auf einem Raspi mit SHM 2.0 laufen und da geht es einwandfrei.

GiK90, 2020-07-09:
Super vielen Dank! Der Hinweis war gut. Habe in meinem Docker-Aufruf noch --net=host hinzugefügt...

sudo docker run --net=host -v $(pwd)/evcc.dist.yaml:/etc/evcc.yaml -p 7070:7070 andig/evcc

andig, 2020-07-10:
Willkommen bei evcc!

> Habe in meinem Docker-Aufruf noch --net=host hinzugefügt...

Das sollte eigentlich nicht notwendig sein. Du müsstest aber den Multicast Port auf jeden Fall durchleiten, also 9522: https://github.com/andig/evcc/blob/master/meter/sma/listener.go#L17

andig, 2020-07-16:
Scheint sich erledigt zu haben...

closed 2020-07-16 by andig