evcc-io/evcc #208: Support WhatsApp

andig, 2020-07-12:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

Tscherno, 2020-07-16:
Gibt es denn noch irgendeine funktionierende Whatsapp API?

andig, 2020-07-16:
Zumindest gibt es ein WA Business API für das sich Bibliotheken finden. "Funktionierend" ist eine schwierige Vokabel: https://github.com/Rhymen/go-whatsapp/issues/410

closed 2020-09-30 by stale[bot]