evcc-io/evcc #209: Modbus/TCP Verbindungen Timeout

premultiply, 2020-07-12:
Für via Modbus/TCP angebundene Geräte (z. B. Wechselrichter) wird offensichtlich die Verbindung derzeit immer nach einem fixem Timeout von 5 Sekunden getrennt (laut Log) und eine neue Verbindung aufgebaut. Dies könnte bei schwachbrüstigen Embedded-Geräten zu einer höheren CPU-Last und damit zu einer erhöhten Latenz führen. Im blödesten Fall werden dadurch beschränkte Verbindungsressourcen blockiert.

Hier wäre es günstiger bestehende Verbindungen beizubehalten bzw. weiter zu nutzen und ggf. zu diesem Zweck das Timeout deutlich zu erhöhen oder ggf. an die Regelzykluszeit anzupassen (z. B. Zykluszeit + 30s)

andig, 2020-07-13:
Ich sehe da keinen Vorteil. Wenn die Verbindung nicht abgebaut wird können gerade schwächere Geräte nicht von anderen clients abgefragt werden. Falls das keine konkreten Probleme verursacht würde ich es so lassen.

premultiply, 2020-07-13:
Ich glaube da gibt es ein Missverständnis.

Derzeit baut wohl jede Modbus-Abfrage eine neue Verbindung auf. Die Alt-Verbindungen laufen dann in ein Timeout.

Das ist eine völlige Ressourcenverschwendung und genau das Gegenteil von dem was eigentlich beabsichtigt ist. Wozu bleiben denn die Verbindungen bis zum Timeout bestehen wenn sie ohnehin nicht mehr benutzt werden?

Um das beabsichtigte Ziel zu erreichen gibt es zwei Möglichkeiten:
1) Eine bestehende TCP-Verbindung wird aktiv nach jeder Abfrage geschlossen und jeweils neu aufgebaut.
2) Verbindungen werden über ein konfigurierbares Timeout geschlossen wenn diese nicht mehr verwendet werden. Gibt es innerhalb des Timeouts eine neue Anfrage wird die bestehende Verbindung wiederverwertet.

Die definitiv schnellere Variante ist b, da dann der ganze TCP-Handshake bei jeder Abfrage entfällt. Gerade bei Embedded-Geräten macht dies mehrere ms pro Abfrage aus.
Im schlimmsten Fall wird auch noch jede eingehende Modbus-Verbindung irgendwo geloggt.

andig, 2020-07-13:
> Derzeit baut wohl jede Modbus-Abfrage eine neue Verbindung auf. Die Alt-Verbindungen laufen dann in ein Timeout.

Das wäre in der Tat doof und sollte nicht der Fall sein. Innerhalb eines "update" Zyklus sollten sich alle Abfragen eine Verbindung teilen. Lässt sich die Vermutung per Log verifizieren?

premultiply, 2020-07-13:
Nein im Tracelog steht dazu glaub ich nix. Da sieht man nur die Timouts nach fixen 5 Sekunden.
Müsste man ggf. per Wireshark ran.

andig, 2020-07-13:
> Nein im Tracelog steht dazu glaub ich nix. Da sieht man nur die Timouts nach fixen 5 Sekunden.

Dann sehe ich das Problem nicht. *Jede* Verbindung innerhalb der 5s wird diese Verbindung nutzen.

> Derzeit baut wohl jede Modbus-Abfrage eine neue Verbindung auf.

Ist m.E. einfach falsch.

andig, 2020-07-16:
Ich mache mal zu- es gibt keine Indikation dass hier ein Problem bzgl. mehrerer Verbindungen im Abfragezeitraum besteht.

closed 2020-07-16 by andig