evcc-io/evcc #211: Zeit- und Energieanzeige

premultiply, 2020-07-13:
Hinsichtlich der Zeit- und Energieanzeige in der GUI sowie auch bei den Push-Benachrichtigungen würde ich mir eine andere Rechnung wünschen. Statt nur die Zeit und Energie einer einzelnen Ladephase (also zwischen Schütz an und Schütz aus) wären die Summen dieser einzelnen Ladephasen interessanter sprich die Summe der Ladephasen zwischen Stecker einstecken und Stecker abziehen (Nettoladezeit).

Für manch anderen mag vielleicht auch noch die verstrichene Zeit seit "Stecker rein" interessant sein (Bruttoladezeit).

Aus beiden Varianten könnte man auch noch eine Durchschnittsladeleistung ausweisen. Für die Statisikliebhaber unter uns ;)
Könnte aber helfen zu zeigen wie wenig Ladeleistung man eigentlich wirklich braucht.

premultiply, 2020-07-20:
Zumindest (Teil?)lösung #237, oder?

andig, 2020-07-25:
Sollte mit #238 gelöst sein

closed 2020-07-25 by andig