evcc-io/evcc #212: Does EVCC work with "new" 2020 BMW-API ?

ak68-hub, 2020-07-14:
I´ve managed to run EVCC on QNAP-Docker:

[image]

Unfortunately, I dn´t get any data from my BMW i3:

[image]

I´m not shure, if it depends on the new BMW-API (changed 2020).

Who one is using EVCC with a BMW i3 and does it work correctly in 2020 ?

andig, 2020-07-14:
/cc @premultiply

premultiply, 2020-07-14:
Warum? Hätte ich BMW-Aktien würde ich sie verkaufen ;)

andig, 2020-07-14:
Gibts denn irgendeinen Fehler? Ladezustand wird nur angezeigt wenn das Auto verbunden ist.

ak68-hub, 2020-07-14:
Wenn das Auto per App oder per Kabel verbunden ist ?

andig, 2020-07-14:
Per Kabel. EVCC weiss ncihts von der App.

ak68-hub, 2020-07-14:
Vielen Dank für Deine Hilfe,

aber die Connecteddrive-Cloud kennt doch den aktuellen Batteriestand des Autos- oder ?

Und von EVCC wird doch die Cloud abgefragt, oder habe ich einen Denkfehler ?

premultiply, 2020-07-14:
EVCC fragt den Ladezustand des Fahrzeugs nur ab während es mit der Ladestation verbunden ist.
Daher wird niemals ein SOC angezeigt wenn kein Fahrzeug angeschlossen ist.

ak68-hub, 2020-07-15:
> EVCC fragt den Ladezustand des Fahrzeugs nur ab während es mit der Ladestation verbunden ist.
> Daher wird niemals ein SOC angezeigt wenn kein Fahrzeug angeschlossen ist.

Dann bin  ich von falschen Vorstellungen ausgegangen, vielen Dank für die Klarstellung !

andig, 2020-07-15:
Bestens, siehe auch #201

ak68-hub, 2020-07-15:
Leider funktioniert der Abruf auch WÄHREND des Ladevorgangs NICHT:

[image]

[image]

Wer hat Rat und wer hat einen i3 mit korrekter Abfrage (ich will nur sichergehen, dass es NICHT an der neuen Connecteddrive-API liegt !)

andig, 2020-07-15:
Ich könnte testen falls Du mir Deine Zugangsdaten (per Mail) teilen möchtest. Gibt es eine andere App mit der der Zugriff funktioniert wo ich abgucken könnte?

ak68-hub, 2020-07-15:
> Antwort an: 	andig/evcc
> <reply@reply.github.com>
> Kopie (CC): 	ak68-hub <karrer-a@web.de>, Author <author@noreply.github.com>

Hallo "andig";

lt. Aussage von Sergej scheint sein php-scruipt auch nach der Umstellung
von BMW noch zu Laufen:

https://github.com/sergejmueller/battery.ebiene.de

Von diesem Projakt abgeschaut ist:

https://www.elektrifiziert.net/blog/index.php?entry/56-webapp-fuer-bmw-i3-8-und-plugins/

auch die Abfrage für Symcom scheint zu Laufen:

https://github.com/Wolbolar/IPSymconBMWConnectedDrive

Gerne kann ich Dir auch zum Testen die Zugangsadten zukommen lassen

Vielen Dank für deine Bemühungen und schöne Grüße
Andreas

-------- Weitergeleitete Nachricht --------
Betreff: 	Re: [andig/evcc] Does EVCC work with "new" 2020 BMW-API ? (#212)
Datum: 	Wed, 15 Jul 2020 13:35:24 +0000 (UTC)

<https://github.com/andig/evcc/issues/212#issuecomment-658771467>, or
unsubscribe

andig, 2020-07-16:
> Gerne kann ich Dir auch zum Testen die Zugangsdaten zukommen lassen

Gerne, meld Dich einfach (cpuidle@gmx.de). Bis dahin könntest Du schonmal mit `log: trace` ein Logfile von Start bis Fehler erstellen, bitte ebenfalls per Mail oder persönliche Daten hier schwärzen.

Tscherno, 2020-07-16:
Ich kann nur von meiner Seite aus sagen, dass es mit meinem i3 einwandfrei läuft (bis auf ein paar fehlende Features ;)

[image]

ak68-hub, 2020-07-16:
> Ich kann nur von meiner Seite aus sagen, dass es mit meinem i3 einwandfrei läuft (bis auf ein paar fehlende Features ;)

Hast Du identische Einstellungen in der Konfig-Datei ?

[image]

[image]

Tscherno, 2020-07-16:
User, PW und VIN17 sind sicher anders ;)

############# Charger #############
chargers:
- name: go
  type: go-e
  uri: http://192.168.1.183 #don't forget the http:// in front!
############# /Charger #############

############# Meters #############
meters:
- name: pv-dc-kostal
  type: default
  power:
    type: calc
    add: # The add plugin sums up both string values
    - type: modbus
      model: sunspec
      value: 160:1:DCW # string 1
      uri: wechselrichter.fritz.box:1502 # Hostname / IP of Kostal Plenticore - Port 1502 Default for Modbus
      id: 71 # Configured Modbus Device ID 
    - type: modbus  
      value: 160:2:DCW # string 2
      uri: wechselrichter.fritz.box:1502  # Hostname / IP of Kostal Plenticore - Port 1502 Default for Modbus
      id: 71 # Configured Modbus Device ID 
- name: grid-kostal
  type: modbus
  model: ksem
  uri: 192.168.1.11:502  # Hostname / IP of Kostal Smart Meter - Port 502 Default for Modbus
  id: 1 # Configured Modbus Device ID 
  power: Power
  energy: Export
- name: battery-kostal # (BYD) Battery connected to Kostal Plenticore
  type: modbus
  power: 802:W
  uri: wechselrichter.fritz.box:1502 
  id: 71 # Configured Modbus Device ID ("Gerät")
############# /Meters #############

############# Car #############
vehicles:
- name: bmw
  type: bmw
  title: i3
  capacity: 65 # kWh
  user: @
  password: ...
  vin: WBY8P2101234567
  cache: 5m
############# /Car #############

loadpoints:
- name: main # name for logging
  vehicle: bmw
  charger: go # charger
  meters:
    grid: grid-kostal # grid meter
    pv: pv-dc-kostal # pv meter
    battery: battery-kostal # battery meter
    #charge: charge # charge meter
  enable: # pv mode enable behavior
    delay: 1m # threshold must be exceeded for this long
    threshold: 0 # minimum export power (W). If zero, export must exceeds minimum charge power to enable
  disable: # pv mode disable behavior
    delay: 1m # threshold must be exceeded for this long
    threshold: 200 # maximum import power (W)
  guardduration: 2m # switch charger contactor not more often than this (default 10m)
  mincurrent: 6 # minimum charge current (default 6A)
  maxcurrent: 16 # maximum charge current (default 16A)
  phases: 3 # ev phases (default 3)
  sensitivity: 1 # current raise/lower step size (default 10A)

Tscherno, 2020-07-16:
PS: Die capacity stimmt weder bei dir noch bei mir. Vielleicht hats aber dennoch eine Auswirkung?

andig, 2020-07-16:
Wie ich schrieb: mit log und zugangsdaten könnte ich debuggen...

andig, 2020-07-17:
Gelöst- falsche VIN. mit #218 kann die VIN jetzt auch entfallen solange es nur 1 Auto gibt. Alle identifizierten VINs werden im Log ausgegeben.

closed 2020-07-17 by andig