evcc-io/evcc #213: Hilfe: Meine Konfiguration funktioniert nicht

andig, 2020-07-16:
Mit #199 hat sich die Konfiguration geändert:

- loadpoint config is split into `site` and `loadpoints`
- grid/ pv and battery meters are moved to `site`
- push message templates are updated- more values available and missing values shown in logfile
- now requires InfluxDB 1.8/2.0 - InfluxDB 1.6/1.7 are no longer supported

Beispiele für geänderte Konfiguration sind in `evcc.dist.yaml` zu finden: https://github.com/andig/evcc/blob/master/evcc.dist.yaml

ak68-hub, 2020-07-16:
1. Wie ist die korrekte Syntax für "site" ?

[image]

[image]

2. Wie ist die korrekte Systax für Loadpoints/ meter ?
meters:
    charge: keba

ttjaden, 2020-07-16:
Bin den Infos / der Syntax aus der verlinkten yaml (https://github.com/andig/evcc/blob/master/evcc.dist.yaml) klappt es bei mir.

Aber ich bekomme wieder eine Fehlermeldung, die mit einem älteren Repo-Stand eigentlich weg war: 
[image]

@andig : Ich habe irgendwann beiläufig gesehen, dass ein anderer User eine Änderung am go-e.go vorgenommen hat. Hat sich hier wieder ein Fehler eingeschlichen?

`Jul 16 20:36:21 oeko-t evcc[19408]: [cache] TRACE 2020/07/16 20:36:21 chargedEnergy: 0
Jul 16 20:36:21 oeko-t evcc[19408]: [cache] TRACE 2020/07/16 20:36:21 chargeDuration: 0s
Jul 16 20:36:21 oeko-t evcc[19408]: [cache] TRACE 2020/07/16 20:36:21 socCharge: -1
Jul 16 20:36:21 oeko-t evcc[19408]: [cache] TRACE 2020/07/16 20:36:21 chargeEstimate: -1
Jul 16 20:36:21 oeko-t evcc[19408]: [iflx] TRACE 2020/07/16 20:36:21 write chargedEnergy=0 (map[loadpoint:go-e Charger])
Jul 16 20:36:21 oeko-t evcc[19408]: [go-e] TRACE 2020/07/16 20:36:21 https://api.go-e.co/api_status?token=xxxxxxxxxxxx
Jul 16 20:36:21 oeko-t evcc[19408]: {"success":true,"age":3730,"data":{"version":"B","tme":"1607202036","rbc":"26","rbt":"341442642","car":"1","amp":"9","err":"0","ast":"0","alw":"1","stp":"0","cbl":"32","pha":"56","tmp":"22","tma":[22,21.5,21.75,21.37],"amt":"32","dws":"0","dwo":"0","adi":"0","uby":"0","eto":"4170","wst":"3","txi":"1","nrg":[225,224,227,1,0,0,0,0,0,0,0,0,0,0,0,0],"fwv":"033","sse":"010486","wss":"G-Hof","wke":"","wen":"1","cdi":"0","tof":"101","tds":"1","lbr":"148","aho":"3","afi":"7","azo":"0","ama":"32","al1":"10","al2":"16","al3":"20","al4":"24","al5":"32","cid":"255","cch":"16748800","cfi":"65280","lse":"1","ust":"2","wak":"","r1x":"2","dto":"0","nmo":"0","sch":"AAAAAAAAAAAAAAAA","sdp":"0","eca":"0","ecr":"40","ecd":"0","ec4":"0","ec5":"0","ec6":"0","ec7":"0","ec8":"0","ec9":"0","ec1":"0","rca":"6414","rcr":"1719","rcd":"","rc4":"","rc5":"","rc6":"","rc7":"","rc8":"","rc9":"","rc1":"","rna":"G\\u00e4ste","rnm":"Tjarko","rne":"ZOE","rn4":"","rn5":"","rn6":"","rn7":"","rn8":"","rn9":"","rn1":"","loe":0,"lot":0,"lom":0,"lop":0,"log":"","lon":0,"lof":0,"loa":0,"lch":28650,"upd":"0"}}
Jul 16 20:36:21 oeko-t evcc[19408]: [lp-1] DEBUG 2020/07/16 20:36:21 charger status: A
Jul 16 20:36:21 oeko-t evcc[19408]: [lp-1] DEBUG 2020/07/16 20:36:21 set charge current: 9A
Jul 16 20:36:21 oeko-t evcc[19408]: [cache] TRACE 2020/07/16 20:36:21 connected: false
Jul 16 20:36:21 oeko-t evcc[19408]: [cache] TRACE 2020/07/16 20:36:21 charging: false
Jul 16 20:36:21 oeko-t evcc[19408]: [go-e] TRACE 2020/07/16 20:36:21 https://api.go-e.co/api?token=xxxxxxxxxxx&payload=amp=9
Jul 16 20:36:21 oeko-t evcc[19408]: {"success":true,"payload":"amp=9"}
Jul 16 20:36:21 oeko-t evcc[19408]: [lp-1] ERROR 2020/07/16 20:36:21 charge controller error: amp update failed: 0
Jul 16 20:36:21 oeko-t evcc[19408]: [core] DEBUG 2020/07/16 20:36:21 lp-1 remaining power: 691W = 691W - 0W
Jul 16 20:36:21 oeko-t evcc[19408]: [cache] TRACE 2020/07/16 20:36:21 error: [lp-1] ERROR 2020/07/16 20:36:21 charge controller error: amp update failed: 0\n`

andig, 2020-07-16:
Bitte einzelne Issues aufmachen für die Probleme.

andig, 2020-07-16:
@ttjaden funzt https://github.com/andig/evcc/pull/215 für Dich?

closed 2020-08-24 by andig