evcc-io/evcc #214: GUI wird auf Smartphone unvollständig angezeigt

ReRo456, 2020-07-16:
**Describe the bug**
Nach Update auf die neue Version e34167c wird bei mir die GUI auf dem Smartphone unvollständig dargestellt: außer dem Titel "evcc" und dem Menü ist nichts zu sehen (siehe Screenshot). Auch, wenn ich im Menü auf "Laden" gehe, erscheint nichts. Auf dem PC ist alles da, es werden die Daten zur "site" und zum "loadpoint" dargestellt und ich kann den Ladevorgang starten (siehe Screenshot). 
Smartphone: Chrome-Browser Version 84.0.4147.89 unter Android 9
PC: Firefox 78.0.1 unter Windows 10

**To Reproduce**
Steps to reproduce the behavior:
1. Go to http://[hostname]:7070

**Expected behavior**
Die GUI wird auch im Smartphone-Browser vollständig dargestellt und EVCC kann bedient werden (Lademodus ändern).

**EVCC details:**
**If using Docker:**
Show output of `docker run andig/evcc -v`: evcc version e34167c (e34167c)
Show evcc log output with `docker logs <container id>`: [evcc_no_gui.log](https://github.com/andig/evcc/files/4931771/evcc_no_gui.log)
Konfiguration: [evcc.yaml](https://github.com/andig/evcc/files/4931772/evcc.yaml.txt)

[image]

[image]

andig, 2020-07-16:
Bitte auf iOS die Websitedaten löschen. Ich kann nur vermuten, dass es ein Cacheproblem ist.

ReRo456, 2020-07-16:
OK, es geht wiegesagt um Chrome unter Android, nicht um iOS. Der Tipp hat aber trotzdem geholfen. Ich habe im Chrome den Cache gelöscht und danach wurde alles wieder angezeigt. Danke für den Tipp!
Hast du auch noch eine Idee, warum unter Ladedauer jetzt 2562047:47h angezeigt wird? Das ist unsinnig und sieht nach einer nicht initialisierten Variablen aus...

andig, 2020-07-16:
> Hast du auch noch eine Idee, warum unter Ladedauer jetzt 2562047:47h angezeigt wird? Das ist unsinnig und sieht nach einer nicht initialisierten Variablen aus...

Ist soeben behoben.

closed 2020-07-16 by andig