evcc-io/evcc #216: Fehlermeldung bei go-e Charger

ttjaden, 2020-07-16:
**Describe the bug**

1. Ich bekomme wieder eine Fehlermeldung, die mit einem älteren Repo-Stand eigentlich weg war. --> ERROR 2020/07/16 20:36:21 charge controller error: amp update failed
2. Der SOC wird nicht mehr angezeigt und auch keine Ladeleistung
3. Egal welchen Knopf ich drücke (Sofort laden / Min+PV), es wird immer mit min current = 9A geladen (Infos am Smartmeter Portal)

**To Reproduce**
evcc mit Stand vom 16.07.2020 18:00 Uhr
kompilliert und in Betrieb auf Ubuntu Server

**Expected behavior**
Wahl der Ladesteuerung sollte wieder gehen und
Anzeige des SOC und der Ladeleistung sollte wieder gehen
--> Hinweis: SOC mittels APP wird angezeigt und "./evcc vehicle" liefert auch den SOC

**EVCC details:**
Show output of `evcc -v`
`evcc version 0f02f6f (0f02f6f)`
Show evcc log output with `--log trace`
`Jul 16 20:36:21 oeko-t evcc[19408]: [cache] TRACE 2020/07/16 20:36:21 chargedEnergy: 0 
Jul 16 20:36:21 oeko-t evcc[19408]: [cache] TRACE 2020/07/16 20:36:21 chargeDuration: 0s 
Jul 16 20:36:21 oeko-t evcc[19408]: [cache] TRACE 2020/07/16 20:36:21 socCharge: -1 
Jul 16 20:36:21 oeko-t evcc[19408]: [cache] TRACE 2020/07/16 20:36:21 chargeEstimate: -1 
Jul 16 20:36:21 oeko-t evcc[19408]: [iflx] TRACE 2020/07/16 20:36:21 write chargedEnergy=0 (map[loadpoint:go-e Charger]) 
Jul 16 20:36:21 oeko-t evcc[19408]: [go-e] TRACE 2020/07/16 20:36:21 https://api.go-e.co/api_status?token=xxxxxxxxxxxx 
Jul 16 20:36:21 oeko-t evcc[19408]: {"success":true,"age":3730,"data":{"version":"B","tme":"1607202036","rbc":"26","rbt":"341442642","car":"1","amp":"9","err":"0","ast":"0","alw":"1","stp":"0","cbl":"32","pha":"56","tmp":"22","tma":[22,21.5,21.75,21.37],"amt":"32","dws":"0","dwo":"0","adi":"0","uby":"0","eto":"4170","wst":"3","txi":"1","nrg":[225,224,227,1,0,0,0,0,0,0,0,0,0,0,0,0],"fwv":"033","sse":"010486","wss":"G-Hof","wke":"","wen":"1","cdi":"0","tof":"101","tds":"1","lbr":"148","aho":"3","afi":"7","azo":"0","ama":"32","al1":"10","al2":"16","al3":"20","al4":"24","al5":"32","cid":"255","cch":"16748800","cfi":"65280","lse":"1","ust":"2","wak":"","r1x":"2","dto":"0","nmo":"0","sch":"AAAAAAAAAAAAAAAA","sdp":"0","eca":"0","ecr":"40","ecd":"0","ec4":"0","ec5":"0","ec6":"0","ec7":"0","ec8":"0","ec9":"0","ec1":"0","rca":"6414","rcr":"1719","rcd":"","rc4":"","rc5":"","rc6":"","rc7":"","rc8":"","rc9":"","rc1":"","rna":"G\\u00e4ste","rnm":"Tjarko","rne":"ZOE","rn4":"","rn5":"","rn6":"","rn7":"","rn8":"","rn9":"","rn1":"","loe":0,"lot":0,"lom":0,"lop":0,"log":"","lon":0,"lof":0,"loa":0,"lch":28650,"upd":"0"}} 
Jul 16 20:36:21 oeko-t evcc[19408]: [lp-1] DEBUG 2020/07/16 20:36:21 charger status: A 
Jul 16 20:36:21 oeko-t evcc[19408]: [lp-1] DEBUG 2020/07/16 20:36:21 set charge current: 9A 
Jul 16 20:36:21 oeko-t evcc[19408]: [cache] TRACE 2020/07/16 20:36:21 connected: false 
Jul 16 20:36:21 oeko-t evcc[19408]: [cache] TRACE 2020/07/16 20:36:21 charging: false 
Jul 16 20:36:21 oeko-t evcc[19408]: [go-e] TRACE 2020/07/16 20:36:21 https://api.go-e.co/api?token=xxxxxxxxxxx&payload=amp=9 Jul 16 20:36:21 oeko-t evcc[19408]: {"success":true,"payload":"amp=9"} 
Jul 16 20:36:21 oeko-t evcc[19408]: [lp-1] ERROR 2020/07/16 20:36:21 charge controller error: amp update failed: 0 
Jul 16 20:36:21 oeko-t evcc[19408]: [core] DEBUG 2020/07/16 20:36:21 lp-1 remaining power: 691W = 691W - 0W J
ul 16 20:36:21 oeko-t evcc[19408]: [cache] TRACE 2020/07/16 20:36:21 error: [lp-1] ERROR 2020/07/16 20:36:21 charge controller error: amp update failed: 0\n`

andig, 2020-07-16:
Mist, erst 2min nach deinem ticket...

ttjaden, 2020-07-16:
Sorry, da hatte sich unsere Kommunikation gerade überschnitten. 
Fazit: habe den fix/go-e branch kompilliert und getestet. Nun

[image]

sudo tail /var/log/syslog
Jul 16 22:11:57 oeko-t evcc[680]: [go-e] TRACE 2020/07/16 22:11:57 https://api.go-e.co/api_status?token=xxxxxxxx
Jul 16 22:11:57 oeko-t evcc[680]: {"success":true,"age":1790,"data":{"version":"B","tme":"1607202211","rbc":"26","rbt":"347180045","car":"1","amp":"9","err":"0","ast":"0","alw":"1","stp":"0","cbl":"32","pha":"56","tmp":"22","tma":[22.37,21.75,22.37,22.25],"amt":"32","dws":"0","dwo":"0","adi":"0","uby":"0","eto":"4190","wst":"3","txi":"1","nrg":[226,226,227,1,0,0,0,0,0,0,0,0,0,0,0,0],"fwv":"033","sse":"010486","wss":"G-Hof","wke":"","wen":"1","cdi":"0","tof":"101","tds":"1","lbr":"148","aho":"3","afi":"7","azo":"0","ama":"32","al1":"10","al2":"16","al3":"20","al4":"24","al5":"32","cid":"255","cch":"16748800","cfi":"65280","lse":"1","ust":"2","wak":"","r1x":"2","dto":"0","nmo":"0","sch":"AAAAAAAAAAAAAAAA","sdp":"0","eca":"0","ecr":"40","ecd":"0","ec4":"0","ec5":"0","ec6":"0","ec7":"0","ec8":"0","ec9":"0","ec1":"0","rca":"6414","rcr":"1719","rcd":"","rc4":"","rc5":"","rc6":"","rc7":"","rc8":"","rc9":"","rc1":"","rna":"G\\u00e4ste","rnm":"Tjarko","rne":"ZOE","rn4":"","rn5":"","rn6":"","rn7":"","rn8":"","rn9":"","rn1":"","loe":0,"lot":0,"lom":0,"lop":0,"log":"","lon":0,"lof":0,"loa":0,"lch":1485,"upd":"0"}}
Jul 16 22:11:57 oeko-t evcc[680]: [lp-1] ERROR 2020/07/16 22:11:57 charge controller error: json: invalid use of ,string struct tag, trying to unmarshal "033" into string
Jul 16 22:11:57 oeko-t evcc[680]: [core] DEBUG 2020/07/16 22:11:57 lp-1 remaining power: 1982W = 1982W - 0W
Jul 16 22:11:57 oeko-t evcc[680]: [cache] TRACE 2020/07/16 22:11:57 error: [lp-1] ERROR 2020/07/16 22:11:57 charge controller error: json: invalid use of ,string struct tag, trying to unmarshal \"033\" into string\n
`

andig, 2020-07-16:
behoben!

ttjaden, 2020-07-16:
Danke! SOC ist wieder da und Steuerung funtioniert auch. Aber es wurde keine Push-Nachricht beim beenden gesendet, was sicherlich mit folgender Meldung zusammenhängt:

[image]

Liegt der Fehler hier noch auf meiner Seite in der yaml?

andig, 2020-07-17:
Sie Meldung- die Variablen habne sich geändert. Z.b. Energy -> charegedEnergy. Siehe dist.yaml.

PR #215 (2020-07-17): Fix go-e cloud updates
Fix #216
closed 2020-07-17 by andig