evcc-io/evcc #219: Daten Senden an MQTT-Server

ak68-hub, 2020-07-17:
Hallo Andreas,

wie kann ich Daten (z.B. Ladezustzand) an einen MQTT-Servver (z.B. Mosquitto) im eigenen Heimnetz senden ?
Alternativ würde auch HTTP-Get in Frage kommen

Danke
Andreas

andig, 2020-07-17:
Bisher gibt es nur `/api/state`, ist aber eine gute Idee!

ak68-hub, 2020-07-17:
Hast Du das nicht schon eingebunden ?

[image]

andig, 2020-07-17:
> Hast Du das nicht schon eingebunden ?

Jaein. Aktuell können wir Werte für Zähler etc einlesen und Controller über MQTT steuern (=schreiben). Es gibt aber keinen Mechnismus *beliebige* Werte wenn sie Anfallen über MQTT oder andere Mittel "nach draußen" zu schreiben.

PR #243 (2020-07-26): Add mqtt interface
Fix #219
closed 2020-07-26 by andig