evcc-io/evcc #222: Kia Bluelink

andig, 2020-07-18:
Ich suche Anwender mit Kia zwecks Test der Apiintegration. Bitte hier melden.

mclane, 2020-07-27:
Hallo andig,
ich bin seit einigen Tagen Besitzer des Kia e-Soul. Wir haben eine PV Anlage, eine Eigenbau-Wallbox (mit Phoenix EM-CP-PP-ETH Steuerung, SDM72 Meter via MQTT). Die Netzleistung wird mit einem SDM630 via MQTT gemessen. Das Ganze ist in eine ioBroker Automatisierung eingebunden. Ich hatte auch schon eine PV Überschusssteuerung im ioBroker gebaut. Diese hat auch mit meinem vorherigen Fahrzeug (Ioniq plug-in) funktioniert. Der e Soul verträgt aber die Wallbox-Ladeabbrüche nicht und startet die Ladung nicht mehr, wenn wieder genügend PV Leistung erreicht sind. Man müsste das also über UVO-Connect vom Fahrzeug aus steuern - das habe ich bisher nicht hinbekommen.
Ich bin also gern Tester für die UVO Connect Integration.
Grüsse, mclane

andig, 2020-07-27:
Cool! Im Moment ist es ein bisschen viel- ich würde aber gerne wieder drauf zurück kommen!

andig, 2020-08-16:
@mclane hab Dir bei Slack geantwortet

mclane, 2020-08-20:
Da gibt es ein schönes Dokument [hier](https://github.com/plips-nl/Kuvork/blob/master/from%20bluelink%20to%20uvo%202.rtf) Danach müssten nur einige Konstanten in den headern uund die urls geändert werden.

closed 2020-08-24 by andig